f4pga/prjxray #456: Different bit names on the same bit position in CLBLL

tmichalak, 2019-01-08:
While working on fixing the htmlgen.py (#173 ) I stumbled on the situation in which I had different bit names on the same bit position.
On tile_clbll_l.html I have:
CLBLL_L.SLICEL_X1.CARRY4.ACY0 | 31_04 
and at the same time
CLBLL_L.SLICEL_X1.AFF.ZINI | 31_04

I also have a second case in the same file where :
CLBLL_L.SLICEL_X1.CARRY4.ACY0 | 31_15
and 
CLBLL_L.SLICEL_X1.AFF.ZRST | 31_15

mcmasterg, 2019-01-09:
This is a regression from ROI expansion. Originally we only allowed the roi module to be placed in the ROI. However, as the ROI grew larger, we removed this constraint as it was causing unroutable designs as the roi included critical IO and related resources. Looks like this allowed the packer to place FFs in these LUTs, which just happen to correlate with our usage

Probably the easiest is to put dummy FFs in these CLBs to exclude placement. Alternatively you might be able to write some constraints to formally exclude placement

tmichalak, 2019-01-10:
I indeed verified that the bit name is correct when we set the EXCLUDE_PLACMENT property on the roi pblock. 
So far all my attempts to LOC dummy FFs inside the CLB failed. Actually, it makes the result even worse by returning more candidates for the targeted tag.

closed 2019-01-28 by litghost