f4pga/prjxray #465: Fuzzer 072-ordered-wires RAM explosion

natsfr, 2019-01-08:
When running the 072 Fuzzer the process stop outputting message quickly on that last one for long time:

**INFO: [Common 17-14] Message 'Vivado 12-2683' appears 100 times and further instances of the messages will be disabled. Use the Tcl command set_msg_config to change the current settings.**

After a long time the RAM use is at about 16GB on my machine and the process get killed:
**/home/nats/Xilinx/Vivado/2017.2/bin/loader: line 179: 14890 Killed                  "$RDI_PROG" "$@"
Makefile:13: recipe for target 'specimen_001/OK' failed
make[1]: *** [specimen_001/OK] Error 137
make[1]: Leaving directory '/home/nats/project/symbiflow/prjxray/fuzzers/072-ordered_wires'
Makefile:17: recipe for target 'run' failed
make: *** [run] Error 2**

Maybe my computer is too RAM limited but I would expect 16GB to be enough.

mithro, 2019-01-08:
I believe 128gb RAM is needed for some at the moment..

natsfr, 2019-01-09:
Working on a solution and may have found a way to mitigate the issue.

natsfr, 2019-01-11:
I found a way to mitigate the problem. Let's try to explain here what is going on from my point of view.
1) in some way the vivado **get_nodes** tcl wrapper is leaky. It could be a bad implementation of their tcl interpreter or like litghost pointed out it could be something in the binary behind. To prove that I tried a smaller code:
```
set pips [get_pips]
 
for { set i 0 } { $i < [llength $pips] } { incr i } {
    set dwn_nodes [get_nodes -downhill -of_object [lindex $pips $i]]
    for { set j 0 } { $j < [llength $dwn_nodes] } { incr j } {
        #puts [lindex $dwn_nodes $j]
    }
    unset dwn_nodes
}
```
Even with unsetting the returned list and making sure it doesn't stay linked to global namespace vars the leak is here (about 9MB per seconds on my computer).

2) I tried something more elaborate using slave interpreter because the spec of TCL says that a slave interpreter must be flushed away with all his own variable after delete (not necessarily right after, the GC is free to choose when) but it doesn't happens in the following code, it still eats RAM until being killed by my Linux.
```
proc process_block { start stop } {
    puts "inside interpreter $start $stop"
    set downhill_fp [open downhill_wires.txt a]
    set uphill_fp [open uphill_wires.txt a]
   
    set pips [get_pips]
   
    for { set i $start } { $i < $stop } { incr i } {
        set pip [lindex $pips $i]
        foreach downhill_node [get_nodes -downhill -of_object $pip] {
            set ordered_downhill_wires [get_wires -from $pip -of_object $downhill_node]
            puts $downhill_fp "$pip $downhill_node $ordered_downhill_wires"
        }
        foreach uphill_node [get_nodes -uphill -of_object $pip] {
            set ordered_uphill_wires [get_wires -to $pip -of_object $uphill_node]
            puts $uphill_fp "$pip $uphill_node $ordered_uphill_wires"
        }
    }
   
    close $downhill_fp
    close $uphill_fp
}
 
close [open downhill_wires.txt w]
close [open uphill_wires.txt w]
 
set pips [get_pips]
set pipslen [llength $pips]
 
set nbblocks 64
set blocksize [expr $pipslen / $nbblocks]
 
unset pips
 
for { set i 0 } { $i < $nbblocks } { incr i } {
    puts "Block: $i"
    set startI [expr $i * $blocksize]
    set stopI  [expr ($i + 1) * $blocksize]
    puts "StartI: $startI - StopI: $stopI"
    set tmpinterp [interp create]
    interp alias $tmpinterp process_block {} process_block
    interp eval $tmpinterp "process_block $startI $stopI"
    interp delete $tmpinterp
}
```
Since Xilinx didn't compile their TCL interpreter with **memory** module it's hard to go deeper for me, my conclusion is that in some way you have to get rid of all the vivado instance and start a fresh one. That's what i did in my PR #489 and it works, allowing also to parallelize things with python thread.

PS: while diving into that I found a keyword I didn't know **set_param messaging.disableStorage 1** we have should check if it's still useful in post 2014 release.

closed 2019-01-14 by natsfr