f4pga/prjxray #477: BRAM interconnect muxes are missing

litghost, 2019-01-10:
There are pips that switch between the RAM18 and RAM36 that are not documented.  Example:
BRAM_L.BRAM_LOGIC_OUTS_B10_3.BRAM_RAMB18_DOADO3

Example pip junction: BRAM_L_X6Y110/BRAM_LOGIC_OUTS_B10_3

Which has the following pips:
BRAM_L_X6Y110/BRAM_L.BRAM_RAMB18_DOADO3->BRAM_LOGIC_OUTS_B10_3
BRAM_L_X6Y110/BRAM_L.BRAM_FIFO36_DOADOU3->BRAM_LOGIC_OUTS_B10_3

litghost, 2019-01-10:
This is the complete list of missing bits from the test:

```
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B10_3.BRAM_RAMB18_DOADO3 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B10_3.BRAM_RAMB18_DOADO3'
Segment DB BRAM_INT_INTERFACE_L, key BRAM_INT_INTERFACE_L.INT_INTERFACE_LOGIC_OUTS_L10.INT_INTERFACE_LOGIC_OUTS_L_B10 not found from line 'BRAM_INT_INTERFACE_L_X6Y123.INT_INTERFACE_LOGIC_OUTS_L10.INT_INTERFACE_LOGIC_OUTS_L_B10'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRBWRADDRU8.BRAM_IMUX_ADDRBWRADDRU8 not found from line 'BRAM_L_X6Y120.BRAM_ADDRBWRADDRU8.BRAM_IMUX_ADDRBWRADDRU8'
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B7_2.BRAM_RAMB18_DOADO8 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B7_2.BRAM_RAMB18_DOADO8'
Segment DB BRAM_INT_INTERFACE_L, key BRAM_INT_INTERFACE_L.INT_INTERFACE_LOGIC_OUTS_L7.INT_INTERFACE_LOGIC_OUTS_L_B7 not found from line 'BRAM_INT_INTERFACE_L_X6Y122.INT_INTERFACE_LOGIC_OUTS_L7.INT_INTERFACE_LOGIC_OUTS_L_B7'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRBWRADDRU14.BRAM_IMUX_ADDRBWRADDRU14 not found from line 'BRAM_L_X6Y120.BRAM_ADDRBWRADDRU14.BRAM_IMUX_ADDRBWRADDRU14'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU2.BRAM_IMUX_ADDRARDADDRU2 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU2.BRAM_IMUX_ADDRARDADDRU2'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU1.BRAM_IMUX_ADDRARDADDRU1 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU1.BRAM_IMUX_ADDRARDADDRU1'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU3.BRAM_IMUX_ADDRARDADDRU3 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU3.BRAM_IMUX_ADDRARDADDRU3'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU4.BRAM_IMUX_ADDRARDADDRU4 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU4.BRAM_IMUX_ADDRARDADDRU4'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRBWRADDRU1.BRAM_IMUX_ADDRBWRADDRU1 not found from line 'BRAM_L_X6Y120.BRAM_ADDRBWRADDRU1.BRAM_IMUX_ADDRBWRADDRU1'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRBWRADDRU2.BRAM_IMUX_ADDRBWRADDRU2 not found from line 'BRAM_L_X6Y120.BRAM_ADDRBWRADDRU2.BRAM_IMUX_ADDRBWRADDRU2'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRBWRADDRU3.BRAM_IMUX_ADDRBWRADDRU3 not found from line 'BRAM_L_X6Y120.BRAM_ADDRBWRADDRU3.BRAM_IMUX_ADDRBWRADDRU3'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRBWRADDRU4.BRAM_IMUX_ADDRBWRADDRU4 not found from line 'BRAM_L_X6Y120.BRAM_ADDRBWRADDRU4.BRAM_IMUX_ADDRBWRADDRU4'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRBWRADDRU13.BRAM_IMUX_ADDRBWRADDRU13 not found from line 'BRAM_L_X6Y120.BRAM_ADDRBWRADDRU13.BRAM_IMUX_ADDRBWRADDRU13'
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B15_3.BRAM_RAMB18_DOPADOP0 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B15_3.BRAM_RAMB18_DOPADOP0'
Segment DB BRAM_INT_INTERFACE_L, key BRAM_INT_INTERFACE_L.INT_INTERFACE_LOGIC_OUTS_L15.INT_INTERFACE_LOGIC_OUTS_L_B15 not found from line 'BRAM_INT_INTERFACE_L_X6Y123.INT_INTERFACE_LOGIC_OUTS_L15.INT_INTERFACE_LOGIC_OUTS_L_B15'
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B2_4.BRAM_RAMB18_DOADO14 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B2_4.BRAM_RAMB18_DOADO14'
Segment DB BRAM_INT_INTERFACE_L, key BRAM_INT_INTERFACE_L.INT_INTERFACE_LOGIC_OUTS_L2.INT_INTERFACE_LOGIC_OUTS_L_B2 not found from line 'BRAM_INT_INTERFACE_L_X6Y124.INT_INTERFACE_LOGIC_OUTS_L2.INT_INTERFACE_LOGIC_OUTS_L_B2'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU8.BRAM_IMUX_ADDRARDADDRU8 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU8.BRAM_IMUX_ADDRARDADDRU8'
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B5_4.BRAM_RAMB18_DOADO13 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B5_4.BRAM_RAMB18_DOADO13'
Segment DB BRAM_INT_INTERFACE_L, key BRAM_INT_INTERFACE_L.INT_INTERFACE_LOGIC_OUTS_L5.INT_INTERFACE_LOGIC_OUTS_L_B5 not found from line 'BRAM_INT_INTERFACE_L_X6Y124.INT_INTERFACE_LOGIC_OUTS_L5.INT_INTERFACE_LOGIC_OUTS_L_B5'
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B0_4.BRAM_RAMB18_DOADO12 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B0_4.BRAM_RAMB18_DOADO12'
Segment DB BRAM_INT_INTERFACE_L, key BRAM_INT_INTERFACE_L.INT_INTERFACE_LOGIC_OUTS_L0.INT_INTERFACE_LOGIC_OUTS_L_B0 not found from line 'BRAM_INT_INTERFACE_L_X6Y124.INT_INTERFACE_LOGIC_OUTS_L0.INT_INTERFACE_LOGIC_OUTS_L_B0'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU10.BRAM_IMUX_ADDRARDADDRU10 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU10.BRAM_IMUX_ADDRARDADDRU10'
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B2_3.BRAM_RAMB18_DOADO11 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B2_3.BRAM_RAMB18_DOADO11'
Segment DB BRAM_INT_INTERFACE_L, key BRAM_INT_INTERFACE_L.INT_INTERFACE_LOGIC_OUTS_L2.INT_INTERFACE_LOGIC_OUTS_L_B2 not found from line 'BRAM_INT_INTERFACE_L_X6Y123.INT_INTERFACE_LOGIC_OUTS_L2.INT_INTERFACE_LOGIC_OUTS_L_B2'
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B13_4.BRAM_RAMB18_DOADO5 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B13_4.BRAM_RAMB18_DOADO5'
Segment DB BRAM_INT_INTERFACE_L, key BRAM_INT_INTERFACE_L.INT_INTERFACE_LOGIC_OUTS_L13.INT_INTERFACE_LOGIC_OUTS_L_B13 not found from line 'BRAM_INT_INTERFACE_L_X6Y124.INT_INTERFACE_LOGIC_OUTS_L13.INT_INTERFACE_LOGIC_OUTS_L_B13'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRBWRADDRU10.BRAM_IMUX_ADDRBWRADDRU10 not found from line 'BRAM_L_X6Y120.BRAM_ADDRBWRADDRU10.BRAM_IMUX_ADDRBWRADDRU10'
Segment DB BRAM_L, key BRAM_L.BRAM_LOGIC_OUTS_B15_4.BRAM_RAMB18_DOADO7 not found from line 'BRAM_L_X6Y120.BRAM_LOGIC_OUTS_B15_4.BRAM_RAMB18_DOADO7'
[... cut ...]
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU6.BRAM_IMUX_ADDRARDADDRU6 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU6.BRAM_IMUX_ADDRARDADDRU6'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU5.BRAM_IMUX_ADDRARDADDRU5 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU5.BRAM_IMUX_ADDRARDADDRU5'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU14.BRAM_IMUX_ADDRARDADDRU14 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU14.BRAM_IMUX_ADDRARDADDRU14'
Segment DB BRAM_L, key BRAM_L.BRAM_ADDRARDADDRU13.BRAM_IMUX_ADDRARDADDRU13 not found from line 'BRAM_L_X6Y120.BRAM_ADDRARDADDRU13.BRAM_IMUX_ADDRARDADDRU13'
```

mcmasterg, 2019-01-10:
This isn't unexpected. The idea was to check this after we got a stable db to test against, but the db never stabalized. My plan was to add some to 050-intpips and see if that caused them to get solved

litghost, 2019-01-18:
Database is stable (barring #554). After I finish investigating #554, I'll try enabling the BRAM pips to 050 and see what happens.  I don't expect much success, as the BRAM pips tend to be very correlated (e.g. all or nothing).

PR #581 (2019-01-31): BRAM Cascade pip fuzzer.
PIP fuzzer that solves BRAM cascade junction box.  Fixes #477
closed 2019-01-31 by litghost