f4pga/prjxray #498: Understand effects of faulty bitstreams

JohnDMcMaster, 2019-01-11:
It would be good to understand what the worst case effect of a faulty bitstream is on real hardware. For example, some of the pips are essentially one hot. What happens if you two hot them?

I think we also noted there might be two places to inject into global clock network? If this is the case, what happens if you drive both of them?

Propose something like measure power consumption, and see if there is an unusual increase. If there is, we should take extra precautions when publishing databases to make sure we don't cause this sort of hardware fault.

mithro, 2019-02-08:
@azonenberg - This seems like the type of thing you are interested in.

azonenberg, 2019-02-08:
I can confirm that (at least) with CoolRunner-II, the one-hot muxes are literally pass transistors on a shared wire. I have experimental data to show that yes, you can cause a significant (100x or more) increase in device power consumption if you create an internal bus fight.

In the case of the XC2C32A, shorting all internal routing muxes with a worst-case pattern (50% 0, 50% 1) resulted in power consumption climbing from <1 mA  to 100+ mA. This was fatal to the device (completely unresponsive including to JTAG) after about 20 minutes.

I strongly recommend creation of some kind of DRC pass that checks for these conditions in routing as a final step before bitstream generation.

mithro, 2019-02-09:
The cool runner is a very old device. I would suggest we look actually look at new devices.