f4pga/prjxray #501: generate piplist once

mcmasterg, 2019-01-11:
This is shared and regenerated by several 05x series fuzzers

mcmasterg, 2019-01-11:
Specifically running https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/piplist.tcl

litghost, 2019-01-14:
Worth noting that the data piplist outputs is a subset of the data that 074-dump_all outputs into tile_X.json files as part of its processing.  If 074 can be made to run quickly and with limited memory, one way to resolve this bug might be to move 072-074 to before the 05x fuzzers, and have the 05x fuzzers use the tile database files that 074 generates.  @natsfr has already improved the runtime of 072 from ~90 minutes to ~30 minutes in #489, hopefully their work on 074 might make 072-074 fast enough to insert them earlier in the flow.  072-074 don't have prerequisites, so they are eligible to move earlier.

kgugala, 2019-01-17:
#533 fixes this one

closed 2019-01-17 by kgugala