f4pga/prjxray #515: fuzzer 074 fails on generate_after_dump

kgugala, 2019-01-15:
```
+ cd ..
+ ./generate_after_dump.sh
generate.sh: line 7: ./generate_after_dump.sh: No such file or directory 
```
it seems commit 19a93121a6b45a3236eeb2d0445b8b86b771fc85 introduced this bug
I've tested this with zynq7 settings

natsfr, 2019-01-15:
I'll correct that this evening, I couldn't test this case completely but I'm surprised because this part appeared to work.

natsfr, 2019-01-15:
Ok could you keep all the file including CSV generated by the 074 ? I would like to compare them to the new version I'm working on. (by the way did you see inconsistencies in 072 run ?)

kgugala, 2019-01-15:
haven't seen so far. I'm testing it with zynq7 settings so I may hit some other problems. If I find something I'll report that.

natsfr, 2019-01-15:
Did you notice the speed increase ?I would like to optimize that, but maybe we can talk about that on IRC.

PR #516 (2019-01-15): fuzzers: 074: fix path to generate_after_dump
fixes #515
closed 2019-01-15 by litghost