f4pga/prjxray #529: Fuzzer pre-run sanity check

natsfr, 2019-01-15:
We could add a check of all needed files before running slow fuzzer to avoid failing late in the process.

kgugala, 2019-01-16:
global fuzzer makefile defines the required dependencies. If you run make for a selected fuzzer from there all the deps should be checked and built if needed

kgugala, 2019-01-16:
to be more precise you need to run `make fuzzer/run.ok` e.g. `make 074-dump_all/run.ok` will build 74th fuzzer with all the dependencies (072)

natsfr, 2019-01-16:
Ok I'll try and check but I think there were some miss yesterday.

kgugala, 2019-01-16:
of course, if a dependency is built for a different chip, the makefile will not detect it

natsfr, 2019-01-16:
For that point I opened an other Feature request #528 it could be linked :)

natsfr, 2019-01-28:
See #494

closed 2019-01-28 by natsfr