f4pga/prjxray #534: fuzzer 018-clbram fails with duplicate bits

kgugala, 2019-01-16:
```
AssertionError: strict: got duplicate bits frozenset({'31_16', '31_17'}): CLB.SLICE_X0.BLUT.RAM CLB.SLICE_X0.ALUT.RAM
...
make[2]: *** [pushdb] Error 1
make[1]: *** [run] Error 2
make: *** [018-clb-ram/run.ok] Error 2
```
tested with zynq7 settings

acomodi, 2019-01-16:
tested with artix7 settings and there are no errors

kgugala, 2019-01-16:
I rerun this fuzzer alone on zynq also with no errors. It may be interference with some other fuzzer, or some kind of a race condition.

litghost, 2019-01-16:
This could be a side effect of using randomness for CLB RAM.  One way to solve this is to ensure that all 16 combination (or so) 64X1S's are generated.  So the permutation starts with 4 LUT6's in a SLICE, and the permutation ends with 4 64X1S.  There are more than enough CLB's to do this for both Y0 and Y1.

litghost, 2019-01-16:
Note that the DLUT must be a 64X1S if any 64X1S are present.

litghost, 2019-01-17:
#543 fixes this.

closed 2019-01-17 by litghost