f4pga/prjxray #567: INT FAN_ALT[0-9].GFAN0 solutions are still unstable

litghost, 2019-01-28:
At some probability, the INT 056-rem fuzzer is generating back solutions.  I've seen the following error at least once with the latest setting:

-INT_R.FAN_ALT4.GFAN0 !22_08 !23_08 !24_08 21_08 25_08
+INT_R.FAN_ALT4.GFAN0 !00_10 !00_11 !01_09 !01_10 !22_08 !23_08 !24_08 01_14 21_08 25_08

01_14 is a bit for the GFAN0 input PIPs, not the FAN_ALT4 input.  Because 01_14 is very correlated to INT_R.FAN_ALT4.GFAN0, a dedicated fuzzer (rather than 056-rem) is a good idea.

litghost, 2019-01-29:
Should test to see if reverting https://github.com/SymbiFlow/prjxray/pull/518/files solves this?

kgugala, 2019-01-30:
054 was using the same algorithm for creating the test cases. Restoring it may help in a way that there will be more fuzzers trying to solve the same problem, so they may be more lucky. This, IMHO,  will not be a bulletproof solution.

litghost, 2019-01-30:
Agreed, a more targetted fuzzer makes sense given how correlated these bits are.

mithro, 2019-03-13:
This is still happening when running both Artix 7 and Zynq 7 fuzzers;
```patch
-INT_L.BYP_ALT1.GFAN0 20_15 !22_15 !23_15 24_15 !25_15
+INT_L.BYP_ALT1.GFAN0 !00_10 !00_11 !01_09 !01_10 01_14 20_15 !22_15 !23_15 24_15 !25_15
```

tmichalak, 2019-03-29:
Commit https://github.com/SymbiFlow/prjxray/commit/3c764a65e14d34aca0821a75d64ef401fe5e8f71  reverts the MAKETODO_FLAGS so that the 054-pip-fan-alt fuzzer only solves the FAN_ALT.GFANs. Before the revert it solved BYP_ALT.GFANs as well. Maybe this is a regression that is causing conflicts @mithro spotted ?
I am testing it in #728

mithro, 2019-05-16:
What was the solution here?

tmichalak, 2019-05-16:
The solution was added in PR #789. It moved solving of the BYP_ALT.GFAN PIPs to the 054-pip-fan-alt fuzzer. Now the fuzzer solves the BYP_ALT.GFAN along with FAN_ALT.GFAN PIPs.
I haven't seen the error from this ticket for a very long time now.

mithro, 2019-05-16:
Great! Just wanted to make sure that information was captured here.

closed 2019-05-16 by mithro