f4pga/prjxray #568: Parallel build does not build all the targets

kgugala, 2019-01-29:
calling `make -jX` ends successfully but not all the targets are built e.g:

```
./001-part-yaml/run.ok
./005-tilegrid/run.ok
./010-clb-lutinit/run.ok
./011-clb-ffconfig/run.ok
./012-clb-n5ffmux/run.ok
./013-clb-ncy0/run.ok
./014-clb-ffsrcemux/run.ok
./015-clb-nffmux/run.ok
./016-clb-noutmux/run.ok
./017-clb-precyinit/run.ok
./018-clb-ram/run.ok
./019-clb-ndi1mux/run.ok
./025-bram-config/run.ok
./026-bram-data/run.ok
./050-pip-seed/run.ok
./072-ordered_wires/run.ok
./073-get_counts/run.ok
```

while it should be:

```
./001-part-yaml/run.ok
./005-tilegrid/run.ok
./010-clb-lutinit/run.ok
./011-clb-ffconfig/run.ok
./012-clb-n5ffmux/run.ok
./013-clb-ncy0/run.ok
./014-clb-ffsrcemux/run.ok
./015-clb-nffmux/run.ok
./016-clb-noutmux/run.ok
./017-clb-precyinit/run.ok
./018-clb-ram/run.ok
./019-clb-ndi1mux/run.ok
./025-bram-config/run.ok
./026-bram-data/run.ok
./050-pip-seed/run.ok
./051-pip-imuxlout-bypalts/run.ok
./052-pip-clkin/run.ok
./053-pip-ctrlin/run.ok
./055-pip-gnd/run.ok
./056-pip-rem/run.ok
./057-pip-bi/run.ok
./071-ppips/run.ok
./072-ordered_wires/run.ok
./073-get_counts/run.ok
./074-dump_all/run.ok
```

running make finishes the build

kgugala, 2019-01-30:
I'm reopening this one. #573 did not fix this completely. My latest clean build ended with:

```
./001-part-yaml/run.ok
./005-tilegrid/run.ok
./010-clb-lutinit/run.ok
./011-clb-ffconfig/run.ok
./012-clb-n5ffmux/run.ok
./013-clb-ncy0/run.ok
./014-clb-ffsrcemux/run.ok
./015-clb-nffmux/run.ok
./016-clb-noutmux/run.ok
./017-clb-precyinit/run.ok
./018-clb-ram/run.ok
./025-bram-config/run.ok
./026-bram-data/run.ok
./050-pip-seed/run.ok
./072-ordered_wires/run.ok
./073-get_counts/run.ok
./100-dsp-mskpat/run.ok
```
And `make` ended without errors.

mithro, 2019-01-31:
This is definitely still happening. It seems none of the ppips fuzzers are running. On a clean run I got the following with the database;
```
        deleted:    ppips_bram_int_interface_l.db
        deleted:    ppips_bram_int_interface_r.db
        deleted:    ppips_bram_l.db
        deleted:    ppips_bram_r.db
        deleted:    ppips_clbll_l.db
        deleted:    ppips_clbll_r.db
        deleted:    ppips_clblm_l.db
        deleted:    ppips_clblm_r.db
        modified:   ppips_hclk_l.db
        modified:   ppips_hclk_r.db
        deleted:    ppips_int_l.db
        deleted:    ppips_int_r.db
```

litghost, 2019-01-31:
Is this fixed with #584?

kgugala, 2019-01-31:
I think so, but I'd wait with closing this until @mithro confirm it works on a build server

mithro, 2019-02-01:
I'm still seeing issues here with bram / hclk output seeming to be missing after running `make -j80 MAX_VIVADO_PROCESS=80 && make`

As well going into the fuzzer directory and typing `make` causes more stuff to run again....

kgugala, 2019-02-01:
hmm, my latests clean run built all the fuzzers and running make again says `Nothing to be done`.

I build it with simple `make -j20`

mithro, 2019-02-05:
This is still happening it seems;
```
========================================
Running Database build
----------------------------------------
make --dry-run
.//run_fuzzer.py 000-init-db
.//run_fuzzer.py 001-part-yaml
.//run_fuzzer.py 005-tilegrid
.//run_fuzzer.py 010-clb-lutinit
.//run_fuzzer.py 011-clb-ffconfig
.//run_fuzzer.py 012-clb-n5ffmux
.//run_fuzzer.py 013-clb-ncy0
.//run_fuzzer.py 014-clb-ffsrcemux
.//run_fuzzer.py 015-clb-nffmux
.//run_fuzzer.py 016-clb-noutmux
.//run_fuzzer.py 017-clb-precyinit
.//run_fuzzer.py 018-clb-ram
.//run_fuzzer.py 019-clb-ndi1mux
.//run_fuzzer.py 025-bram-config
.//run_fuzzer.py 026-bram-data
.//run_fuzzer.py 027-bram36-config
.//run_fuzzer.py 028-fifo-config
.//run_fuzzer.py 050-pip-seed
.//run_fuzzer.py 051-pip-imuxlout-bypalts
.//run_fuzzer.py 052-pip-clkin
.//run_fuzzer.py 053-pip-ctrlin
.//run_fuzzer.py 055-pip-gnd
.//run_fuzzer.py 056-pip-rem
.//run_fuzzer.py 057-pip-bi
.//run_fuzzer.py 058-pip-hclk
.//run_fuzzer.py 060-bram-cascades
.//run_fuzzer.py 071-ppips
.//run_fuzzer.py 072-ordered_wires
.//run_fuzzer.py 073-get_counts
.//run_fuzzer.py 074-dump_all
.//run_fuzzer.py 100-dsp-mskpat
----------------------------------------
+ script -q -c 'make -j 96 MAX_VIVADO_PROCESS=96' -
000-init-db              -    0s: Starting @ 2019-02-05T00:13:38.622968
072-ordered_wires        -    0s: Starting @ 2019-02-05T00:13:38.626222
073-get_counts           -    0s: Starting @ 2019-02-05T00:13:38.627482
<snip>
050-pip-seed             - 1h12m: make[1]: Leaving directory '/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/050-pip-seed'
050-pip-seed             - 1h12m: 
050-pip-seed             - 1h12m: --------------------------------------------------------------------------
+ set +x
----------------------------------------
make --dry-run
.//run_fuzzer.py 051-pip-imuxlout-bypalts
.//run_fuzzer.py 052-pip-clkin
.//run_fuzzer.py 053-pip-ctrlin
.//run_fuzzer.py 055-pip-gnd
.//run_fuzzer.py 056-pip-rem
.//run_fuzzer.py 057-pip-bi
.//run_fuzzer.py 058-pip-hclk
.//run_fuzzer.py 071-ppips
.//run_fuzzer.py 100-dsp-mskpat
----------------------------------------

mithro, 2019-02-05:
It's not just "anything after `050-pip-seed`" as `060-bram-cascades` seems to be successfully run.

Is it something to do with depending on `050-pip-seed`?

kgugala, 2019-02-05:
did you try running the `make all` explicitly?

mithro, 2019-02-08:
This seems to be working now.

PR #573 (2019-01-29): Fuzzers: Makefile: use single colon target for run.ok targets
Fixes #568 . At least for a clean build.
closed 2019-02-08 by mithro