f4pga/prjxray #569: Blinky on Artix does not work with the latest DB

kgugala, 2019-01-29:
DB generated from 44ce9a46e84b8d3a354e1c3c7626465742598b15 fails when building counter demo with:

```
Traceback (most recent call last): fasm_assembler.py", line 133, in enable_feature
    for bit in segbits.feature_to_bits(db_k, address): File "tile_segbits.py", line 141, in feature_to_bits
    for bit in self.segbits[self.feature_addresses[feature][address]]:
KeyError: 'CLBLL_R.CLBLL_LOGIC_OUTS7.CLBLL_LL_DQ'
```

acomodi, 2019-01-29:
This should be solved by fuzzer 071, and before that all the pips ones.
I had a similar issue with the zynq and by running the 071 this step passed:

```
Traceback (most recent call last):
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/prjxray/fasm_assembler.py", line 133, in enable_feature
    for bit in segbits.feature_to_bits(db_k, address):
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/prjxray/tile_segbits.py", line 140, in feature_to_bits
    for bit in self.segbits[self.feature_addresses[feature][address]]:
KeyError: 'CLBLL_R.CLBLL_LOGIC_OUTS12.CLBLL_LL_A'
 ```

kgugala, 2019-01-29:
indeed the issue was caused by missing 071 results. I'm closing this one and we should focus on #568

closed 2019-01-29 by kgugala