f4pga/prjxray #572: RAM_TEST in arch-defs fails on Artix7

acomodi, 2019-01-29:
`ram_test` in symbiflow-arch-defs gives the following error with the most recent db (commit 44ce9a46e84b8d3a354e1c3c7626465742598b15):

```
Traceback (most recent call last):
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 102, in <module>
    main()
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 98, in main
    debug=args.debug)
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/utils/fasm2frames.py", line 56, in run
    assembler.parse_fasm_filename(filename_in)
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/prjxray/fasm_assembler.py", line 163, in parse_fasm_filename
    self.enable_feature(tile, feature, address, line_str)
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/prjxray/fasm_assembler.py", line 134, in enable_feature
    update_segbit(bit)
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/prjxray/fasm_assembler.py", line 119, in update_segbit
    self.frame_clear(frame_addr, word_addr, bit_index, line)
  File "/home/build/symbiflow-arch-defs/third_party/prjxray/prjxray/fasm_assembler.py", line 92, in frame_clear
    self.frames_line[key],
prjxray.fasm_assembler.FasmInconsistentBits: FASM line "INT_R_X15Y137.FAN_ALT4.GFAN0" wanted to clear bit (132992, 75, 11) but was set by FASM line "INT_R_X15Y137.GFAN0.NR1END1"
xc7/tests/ram_test/CMakeFiles/ram_test_bin.dir/build.make:66: recipe for target 'xc7/tests/ram_test/ram_test/artix7-xc7a50t-roi-virt-xc7a50t-test/top.frames' failed
make[3]: *** [xc7/tests/ram_test/ram_test/artix7-xc7a50t-roi-virt-xc7a50t-test/top.frames] Error 1
make[3]: *** Deleting file 'xc7/tests/ram_test/ram_test/artix7-xc7a50t-roi-virt-xc7a50t-test/top.frames'
CMakeFiles/Makefile2:36914: recipe for target 'xc7/tests/ram_test/CMakeFiles/ram_test_bin.dir/all' failed
make[2]: *** [xc7/tests/ram_test/CMakeFiles/ram_test_bin.dir/all] Error 2
CMakeFiles/Makefile2:36921: recipe for target 'xc7/tests/ram_test/CMakeFiles/ram_test_bin.dir/rule' failed
make[1]: *** [xc7/tests/ram_test/CMakeFiles/ram_test_bin.dir/rule] Error 2
Makefile:11613: recipe for target 'ram_test_bin' failed
make: *** [ram_test_bin] Error 2

```

Seems like an inconsistency in the DB

litghost, 2019-01-29:
Known issue, #567.  A dedicated fuzzer for the FAN_ALT[0-9].GFAN0 might be required.  I've found clearing database/${XRAY_PART}/*int*, re-pushing 050-055, and 058 and then re-running 056-rem sometimes resolves the issue.

litghost, 2019-01-29:
In the short term, use the old line FAN_ALT4.GFAN0 from #567

closed 2019-01-29 by litghost