f4pga/prjxray #579: fuzzer 019 produces duplicate tags

kgugala, 2019-01-30:
two consecutive runs gave:

```
WARNING: got duplicate tag CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI                                              
Orig line: CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI 00_00 24_00                                                
New line : CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI 00_00 
```
tested on artix

litghost, 2019-01-30:
That fuzzer is still on N = 1 (https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/019-clb-ndi1mux/Makefile#L1)

Try increasing N to 3 first.  I've found that lowers the probability of a bad bit significantly.

litghost, 2019-01-30:
There is logic in 019 "checkdb", but that logic isn't wired into the root makefile.  Maybe that is the bug?

kgugala, 2019-01-30:
with N=3 I still get the same error

litghost, 2019-01-30:
Wait, I may have misunderstood.  This issue is backwards.  The "Orig line" is wrong, so the database in your tree is corrupt.  Fuzzer 019 is returning the correct answer ("00_00" is correct, "00_00 24_00" is wrong).

https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/segbits_clblm_l.db#L352

kgugala, 2019-01-30:
ok so the first run with N=1 must have generated incorrect DB entry. I'll set up a test where I'll build the 019 over and over for a few hour. If that one passes with N=3, we can assume this solves the problem.

kgugala, 2019-01-31:
with N=3 It's been running for 10h without fail. I'll open a PR

PR #585 (2019-01-31): Fuzzers: 019: set N=3
fixes #579
closed 2019-01-31 by litghost