f4pga/prjxray #582: mask_dsp_l.db file seems to be invalid

mithro, 2019-01-30:
The `mask_XXX.db` files should be in the format at;
```
::::::::::::::
database/artix7/mask_bram_l.block_ram.db
::::::::::::::
bit 00_00
bit 00_01
bit 00_02
bit 00_03
bit 00_04
bit 00_05
bit 00_06
bit 00_07
bit 00_08
bit 00_09
bit 00_10
bit 00_11
bit 00_12
bit 00_13
bit 00_14
bit 00_15
bit 00_16
bit 00_17
bit 00_18
bit 00_19
bit 00_20
bit 00_21
bit 00_22
bit 00_23
bit 00_24
...
::::::::::::::
database/artix7/mask_clbll_l.db
::::::::::::::
bit 00_00
bit 00_01
bit 00_02
bit 00_03
bit 00_04
bit 00_05
bit 00_06
bit 00_07
bit 00_08
bit 00_09
bit 00_10
...
```

The `dsp_mask_l.db` file seems to have output like the following;
```
DSP.DSP48.DSP_0.MASK[0] 27_01
DSP.DSP48.DSP_0.MASK[10] 27_26
DSP.DSP48.DSP_0.MASK[11] 26_28
DSP.DSP48.DSP_0.MASK[12] 26_41
```

litghost, 2019-01-31:
dsp_mask_l is a segbits file, not a mask file.

litghost, 2019-01-31:
Are you talking about https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/segbits_dsp_r.db?

mithro, 2019-01-31:
dsp_mask_l should be a mask file? It looks like the segbits output is ending up in the mask files?

litghost, 2019-01-31:
Why database output are you referring too?  In prjxray-db I only see segbits_dsp_r, which looks fine.

mithro, 2019-01-31:
When I'm generating databases I'm getting a `artix7/mask_dsp_l.db` file as shown above.

mithro, 2019-01-31:
I'm not sure were it is coming from....

litghost, 2019-01-31:
Presumable 100-dsp-mskpat

mithro, 2019-01-31:
After doing a clean build, when running `make formatdb` you get the following;
```
Sorting DB   file mask_dsp_l.db                            Traceback (most recent call last):
  File "../../utils/sort_db.py", line 363, in <module>
    sys.exit(main(sys.argv))
  File "../../utils/sort_db.py", line 347, in main
    x = sort_db(n)
  File "../../utils/sort_db.py", line 303, in sort_db
    tosort.append(sortable_line_from_dbfile(l))
  File "../../utils/sort_db.py", line 233, in sortable_line_from_mask
    assert tag == 'bit', tag
AssertionError: DSP.DSP48.DSP_0.MASK[0]
Makefile:75: recipe for target 'formatdb' failed
make: *** [formatdb] Error 1
```

mithro, 2019-01-31:
Looking in the `prjxray/fuzzers/100-dsp-mskpat/build` directory, there is a `mask_dsp_l.db` file which looks correct;
```
bit 00_103
bit 00_106
bit 00_11
bit 00_139
bit 00_163
bit 00_167
bit 00_170
bit 00_203
bit 00_209
bit 00_217
...
```

There is also a `segbits_dsp_l.db` file which looks like;
```
DSP.DSP48.DSP_0.MASK[0] 27_01
DSP.DSP48.DSP_0.MASK[10] 27_26
DSP.DSP48.DSP_0.MASK[11] 26_28
DSP.DSP48.DSP_0.MASK[12] 26_41
...
````

It very much seems like the `segbits_dsp_l.db` output is incorrectly getting copied over the `mask_dsp_l.db` file?

mithro, 2019-02-01:
I rerun the fuzzers at commit 931576a770a6dec5ca91c8d3368be59a98c19817 and am still getting this behaviour. 
```
(env) [UPDATE PRJXRAY DATABASE] tansell@instance-1:~/github/SymbiFlow/prjxray$ make formatdb

Formatting database/artix7                                   
============================
Ignoring    file element_counts.csv                       .. success.
Sorting DB   file mask_bram_l.block_ram.db                 .. success.
Sorting DB   file mask_bram_l.db                           .. success.
Sorting DB   file mask_bram_r.block_ram.db                 .. success.
Sorting DB   file mask_bram_r.db                           .. success.
Sorting DB   file mask_clbll_l.db                          .. success.
Sorting DB   file mask_clbll_r.db                          .. success.
Sorting DB   file mask_clblm_l.db                          .. success.
Sorting DB   file mask_clblm_r.db                          .. success.     
Sorting DB   file mask_dsp_l.db                            .. success.
Sorting DB   file mask_dsp_r.db                            Traceback (most recent call last):
  File "../../utils/sort_db.py", line 363, in <module>          
    sys.exit(main(sys.argv))          
  File "../../utils/sort_db.py", line 347, in main
    x = sort_db(n)                                                                   
  File "../../utils/sort_db.py", line 303, in sort_db
    tosort.append(sortable_line_from_dbfile(l))                            
  File "../../utils/sort_db.py", line 233, in sortable_line_from_mask
    assert tag == 'bit', tag                   
AssertionError: DSP.DSP48.DSP_0.MASK[0]
Makefile:75: recipe for target 'formatdb' failed           
make: *** [formatdb] Error 1       
```

`mask_dsp_r.db` seems to have both `bit XXX` lines **and** `DSP.DSP48.DSP_0.PATTERN[0] 26_01` lines.

mithro, 2019-02-01:
@litghost Yes, removed all the files under the directory.

closed 2019-02-04 by mithro

mithro, 2019-02-05:
Looks like there is still some issue going on with the dsp fuzzer;
```
100-dsp-mskpat           - 3m04s: Finishing (User:772s System:27s).
100-dsp-mskpat           - 3m04s: Failed @ 2019-02-05T02:41:47.869730 with exit code: 2
100-dsp-mskpat           - 3m04s: --------------------------------------------------------------------------
100-dsp-mskpat           - 3m04s: ++ cd build/specimen_008
100-dsp-mskpat           - 3m04s: ++ cd build/specimen_001
100-dsp-mskpat           - 3m04s: ++ export SEED=7ee7ec67
100-dsp-mskpat           - 3m04s: ++ export SEED=fad8c932
100-dsp-mskpat           - 3m04s: ++ SEED=7ee7ec67
100-dsp-mskpat           - 3m04s: ++ SEED=fad8c932
100-dsp-mskpat           - 3m04s: +++ echo build/specimen_001
100-dsp-mskpat           - 3m04s: +++ md5sum
100-dsp-mskpat           - 3m04s: +++ md5sum
100-dsp-mskpat           - 3m04s: +++ cut -c1-8
100-dsp-mskpat           - 3m04s: +++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ export SEEDN=6
100-dsp-mskpat           - 3m04s: ++ SEEDN=6
100-dsp-mskpat           - 3m04s: ++ export SEEDN=4
100-dsp-mskpat           - 3m04s: ++ SEEDN=4
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: +++ cut -c1-8
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_007
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_002
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_006
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_004
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ export SEED=ddcf076c
100-dsp-mskpat           - 3m04s: ++ SEED=ddcf076c
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ export SEED=22b443a6
100-dsp-mskpat           - 3m04s: ++ SEED=22b443a6
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++ export SEEDN=7
100-dsp-mskpat           - 3m04s: ++ SEEDN=7
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_001
100-dsp-mskpat           - 3m04s: ++ export SEEDN=2
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h27bbb5c2'
100-dsp-mskpat           - 3m04s: ++ SEEDN=2
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_007
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_002
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h152d8583'
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_005
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ export SEEDN=1
100-dsp-mskpat           - 3m04s: ++ SEEDN=1
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h7ee7ec67'
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_001
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ export SEEDN=5
100-dsp-mskpat           - 3m04s: ++ SEEDN=5
100-dsp-mskpat           - 3m04s: ++ cd build/specimen_003
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''hfad8c932'
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ echo build/specimen_003
[... 17 lines cut ...]
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_003
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ export SEED=258dce4a
100-dsp-mskpat           - 3m04s: ++ SEED=258dce4a
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h131b315d'
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_008
100-dsp-mskpat           - 3m04s: ++ export SEEDN=8
100-dsp-mskpat           - 3m04s: ++ SEEDN=8
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_008
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h258dce4a'
[... cut ...] biflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: make[1]: *** [run] Error 2
100-dsp-mskpat           - 3m04s: 
100-dsp-mskpat           - 3m04s: --------------------------------------------------------------------------
100-dsp-mskpat           - 3m04s: Failed @ 2019-02-05T02:41:47.869730 with exit code: 2
Makefile:77: recipe for target '100-dsp-mskpat/run.ok' failed
make: *** [100-dsp-mskpat/run.ok] Error 2
make: *** Waiting for unfinished jobs....
```

PR #609 (2019-02-04): Revert Makefile changes, Fixes #582
Fixes #582
commit 9b697254 (2019-02-04): Revert Makefile changes, Fixes #582
commit a4c92576 (2019-02-04): Merge pull request #609 from litghost/unbreak_dsp_makefile

Revert Makefile changes, Fixes #582