f4pga/prjxray #588: INT_R.IMUX43.LOGIC_OUTS23 solution is unstable

litghost, 2019-01-31:
-INT_R.IMUX43.LOGIC_OUTS23 21_30 !22_30 !23_30 24_30 !25_30
+INT_R.IMUX43.LOGIC_OUTS23 20_00 21_30 !22_30 !23_30 24_30 !25_30

mithro, 2019-02-08:
Just did a fresh build and it appears to still be happening?

```patch
@@ -1916,7 +1916,7 @@ INT_R.IMUX43.FAN_BOUNCE3 20_30 !22_30 23_30 24_30 25_30
 INT_R.IMUX43.FAN_BOUNCE5 20_30 22_30 !23_30 24_30 25_30
 INT_R.IMUX43.LOGIC_OUTS1 21_30 22_30 !23_30 24_30 25_30
 INT_R.IMUX43.LOGIC_OUTS13 21_30 !22_30 23_30 24_30 25_30
-INT_R.IMUX43.LOGIC_OUTS23 20_00 21_30 !22_30 !23_30 24_30 !25_30
+INT_R.IMUX43.LOGIC_OUTS23 21_30 !22_30 !23_30 24_30 !25_30
 INT_R.IMUX43.EE2END1 19_31 !22_30 !23_30 !24_30 25_30
 INT_R.IMUX43.EL1END2 19_31 !22_30 23_30 24_30 25_30
 INT_R.IMUX43.ER1END1 18_31 22_30 !23_30 24_30 25_30
```

```
commit 2830c4ab88edd4984acea9ac1d53f3f9be89dc8e
Merge: 7c7f446 98cf09b
Author: litghost <537074+litghost@users.noreply.github.com>
Date:   Thu Feb 7 15:43:08 2019 -0800

    Merge pull request #624 from mithro/kmaster

    infra: Improve the diff output.
```

tmichalak, 2019-03-04:
I saw that we now have ``INT_R.IMUX43.LOGIC_OUTS23 21_30 !22_30 !23_30 24_30 `!25_30`` in our database. I tried running the fuzzer several times and didn't see the bit ``20_00`` show up for this feature, so I wonder whether or when we can consider this solution stable.

tmichalak, 2019-03-06:
Didn't see the problem in any of the runs I did lately. I guess we could close this issue?

closed 2019-03-06 by litghost