f4pga/prjxray #604: 060-bram-cascades missing dependency?

mithro, 2019-02-03:
```
060-bram-cascades        -    0s: Starting @ 2019-02-03T14:56:03.409587
060-bram-cascades        -    0s: Running make -C /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/060-bram-cascades run (with MAKEFLAGS=' -j --jobserver-fds=3,4 -- MAX_VIVADO_PROCESS=96')
060-bram-cascades        -   10s: Still running (1m:124.81%, 5m:150.46%, 15m:85.99%).
060-bram-cascades        -   10s: make---sh---bash---make---sh---vivado.sh---vivado---loader---vivado---2*[{vivado}]
060-bram-cascades        -   16s: Finishing (User:15s System:1s).
060-bram-cascades        -   16s: Failed @ 2019-02-03T14:56:19.990161 with exit code: 2
060-bram-cascades        -   16s: --------------------------------------------------------------------------
060-bram-cascades        -   16s: + MAKE=make
060-bram-cascades        -   16s: + echo make
060-bram-cascades        -   16s: + i=1
060-bram-cascades        -   16s: + true
060-bram-cascades        -   16s: + make ITER=1 cleaniter
060-bram-cascades        -   16s: + make ITER=1 build/todo.txt
060-bram-cascades        -   16s: cp: cannot stat '/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/database/artix7/segbits*.db': No such file or directory
060-bram-cascades        -   16s: make[2]: *** [build/database/seeded] Error 1
060-bram-cascades        -   16s: make[2]: *** Waiting for unfinished jobs....
060-bram-cascades        -   16s: make[1]: *** [run] Error 2
060-bram-cascades        -   16s: 
060-bram-cascades        -   16s: --------------------------------------------------------------------------
060-bram-cascades        -   16s: Failed @ 2019-02-03T14:56:19.990161 with exit code: 2
```

litghost, 2019-02-03:
You cleaned the database then failed to re-source the settings.sh.  All of the PIP fuzzers behave this way.

mithro, 2019-02-04:
Send a pull request which should fixed this here -> https://github.com/SymbiFlow/prjxray/pull/605

PR #605 (2019-02-04): fuzzers: Add 000-init-db fuzzer.
Creates the .db files via make rather than via sourcing settings.sh
allowing pip fuzzers to interact correctly with `make clean`.

Fixes #604.
commit 2eb1bf3b (2019-02-04): fuzzers: Add 000-init-db fuzzer.

Creates the .db files via make rather than via sourcing settings.sh
allowing pip fuzzers to interact correctly with `make clean`.

Fixes #604.
closed 2019-02-04 by litghost