f4pga/prjxray #613: DSP Fuzzer fails when parallel execution information is passed down

mithro, 2019-02-05:
With https://github.com/SymbiFlow/prjxray/pull/596 the parallel make information is successfully passed down to submake files and it seems that the `100-dsp-mskpat` fuzzer seems to get confused in some way;
```
100-dsp-mskpat           - 3m04s: Finishing (User:772s System:27s).
100-dsp-mskpat           - 3m04s: Failed @ 2019-02-05T02:41:47.869730 with exit code: 2
100-dsp-mskpat           - 3m04s: --------------------------------------------------------------------------
100-dsp-mskpat           - 3m04s: ++ cd build/specimen_008
100-dsp-mskpat           - 3m04s: ++ cd build/specimen_001
100-dsp-mskpat           - 3m04s: ++ export SEED=7ee7ec67
100-dsp-mskpat           - 3m04s: ++ export SEED=fad8c932
100-dsp-mskpat           - 3m04s: ++ SEED=7ee7ec67
100-dsp-mskpat           - 3m04s: ++ SEED=fad8c932
100-dsp-mskpat           - 3m04s: +++ echo build/specimen_001
100-dsp-mskpat           - 3m04s: +++ md5sum
100-dsp-mskpat           - 3m04s: +++ md5sum
100-dsp-mskpat           - 3m04s: +++ cut -c1-8
100-dsp-mskpat           - 3m04s: +++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ export SEEDN=6
100-dsp-mskpat           - 3m04s: ++ SEEDN=6
100-dsp-mskpat           - 3m04s: ++ export SEEDN=4
100-dsp-mskpat           - 3m04s: ++ SEEDN=4
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: +++ cut -c1-8
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_007
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_002
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_006
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_004
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ export SEED=ddcf076c
100-dsp-mskpat           - 3m04s: ++ SEED=ddcf076c
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ export SEED=22b443a6
100-dsp-mskpat           - 3m04s: ++ SEED=22b443a6
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++ export SEEDN=7
100-dsp-mskpat           - 3m04s: ++ SEEDN=7
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_001
100-dsp-mskpat           - 3m04s: ++ export SEEDN=2
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h27bbb5c2'
100-dsp-mskpat           - 3m04s: ++ SEEDN=2
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_007
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_002
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h152d8583'
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_005
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ export SEEDN=1
100-dsp-mskpat           - 3m04s: ++ SEEDN=1
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h7ee7ec67'
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_001
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ export SEEDN=5
100-dsp-mskpat           - 3m04s: ++ SEEDN=5
100-dsp-mskpat           - 3m04s: ++ cd build/specimen_003
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''hfad8c932'
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ echo build/specimen_003
100-dsp-mskpat           - 3m04s: +++ md5sum
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_005
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''hddcf076c'
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ cut -c1-8
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h22b443a6'
100-dsp-mskpat           - 3m04s: ++ export SEED=131b315d
100-dsp-mskpat           - 3m04s: ++ SEED=131b315d
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_003
100-dsp-mskpat           - 3m04s: ++ export SEEDN=3
100-dsp-mskpat           - 3m04s: ++ SEEDN=3
100-dsp-mskpat           - 3m04s: +++ echo build/specimen_008
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_003
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: ++ export SEED=258dce4a
100-dsp-mskpat           - 3m04s: ++ SEED=258dce4a
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h131b315d'
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: +++ sed 's/specimen_0*//'
100-dsp-mskpat           - 3m04s: ++++ pwd
100-dsp-mskpat           - 3m04s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/100-dsp-mskpat/build/specimen_008
100-dsp-mskpat           - 3m04s: ++ export SEEDN=8
100-dsp-mskpat           - 3m04s: ++ SEEDN=8
100-dsp-mskpat           - 3m04s: ++ echo build/specimen_008
100-dsp-mskpat           - 3m04s: ++ md5sum
100-dsp-mskpat           - 3m04s: ++ cut -c1-8
100-dsp-mskpat           - 3m04s: + echo '`define SEED 32'\''h258dce4a'
100-dsp-mskpat           - 3m04s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
100-dsp-mskpat           - 3m04s: make[1]: *** [run] Error 2
100-dsp-mskpat           - 3m04s: 
100-dsp-mskpat           - 3m04s: --------------------------------------------------------------------------
100-dsp-mskpat           - 3m04s: Failed @ 2019-02-05T02:41:47.869730 with exit code: 2
Makefile:77: recipe for target '100-dsp-mskpat/run.ok' failed
make: *** [100-dsp-mskpat/run.ok] Error 2
make: *** Waiting for unfinished jobs....
```

mithro, 2019-02-05:
It is unclear to me why this isn't affecting the other fuzzers?

mithro, 2019-02-05:
Looks like it might affect other fuzzers too;
```
052-pip-clkin            - 1m40s: Still running (1m:91.97%, 5m:55.85%, 15m:96.92%).
052-pip-clkin            - 1m40s: make---sh---bash---make---20*[bash---make---vivado.sh---vivado---loader---vivado]
053-pip-ctrlin           - 1m40s: Still running (1m:91.97%, 5m:55.85%, 15m:96.92%).
053-pip-ctrlin           - 1m40s: make---sh---bash---make-+-10*[bash---make---sh---python3]
053-pip-ctrlin           - 1m40s:                         |-6*[bash---make---vivado.sh---vivado---loader---vivado]
053-pip-ctrlin           - 1m40s:                         `-bash---make
053-pip-ctrlin           - 1m50s: Finishing (User:2124s System:67s).
053-pip-ctrlin           - 1m50s: Failed @ 2019-02-05T14:12:38.089913 with exit code: 2
053-pip-ctrlin           - 1m50s: --------------------------------------------------------------------------
053-pip-ctrlin           - 1m50s: ++ export SEED=9aa8f41f
053-pip-ctrlin           - 1m50s: ++ SEED=9aa8f41f
053-pip-ctrlin           - 1m50s: +++ echo build/1/specimen_019
053-pip-ctrlin           - 1m50s: +++ cut -c1-8
053-pip-ctrlin           - 1m50s: +++ sed 's/specimen_0*//'
053-pip-ctrlin           - 1m50s: ++++ pwd
053-pip-ctrlin           - 1m50s: ++++ pwd
053-pip-ctrlin           - 1m50s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin/build/1/specimen_017
053-pip-ctrlin           - 1m50s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin/build/1/specimen_016
053-pip-ctrlin           - 1m50s: + export FUZDIR=/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin
053-pip-ctrlin           - 1m50s: + FUZDIR=/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin
053-pip-ctrlin           - 1m50s: + source /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/genheader.sh
053-pip-ctrlin           - 1m50s: ++ '[' -z artix7 ']'
053-pip-ctrlin           - 1m50s: ++ set -ex
053-pip-ctrlin           - 1m50s: ++ export FUZDIR=/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin
053-pip-ctrlin           - 1m50s: ++ FUZDIR=/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin
053-pip-ctrlin           - 1m50s: ++ test 1 -ge 1
053-pip-ctrlin           - 1m50s: ++ test '!' -e ''
053-pip-ctrlin           - 1m50s: ++ export SPECDIR=build/1/specimen_020
053-pip-ctrlin           - 1m50s: ++ SPECDIR=build/1/specimen_020
053-pip-ctrlin           - 1m50s: ++ mkdir -p build/1/specimen_020
053-pip-ctrlin           - 1m50s: +++ md5sum
053-pip-ctrlin           - 1m50s: ++ export SEED=c045d7c3
053-pip-ctrlin           - 1m50s: +++ cut -c1-8
053-pip-ctrlin           - 1m50s: ++ SEED=c045d7c3
053-pip-ctrlin           - 1m50s: + echo '`define SEED 32'\''h5bb070fb'
053-pip-ctrlin           - 1m50s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
053-pip-ctrlin           - 1m50s: ++ export SEEDN=16
053-pip-ctrlin           - 1m50s: ++ SEEDN=16
053-pip-ctrlin           - 1m50s: ++ export SEEDN=15
053-pip-ctrlin           - 1m50s: ++ SEEDN=15
053-pip-ctrlin           - 1m50s: ++ echo build/1/specimen_016
053-pip-ctrlin           - 1m50s: ++ md5sum
053-pip-ctrlin           - 1m50s: ++ cut -c1-8
053-pip-ctrlin           - 1m50s: +++ sed 's/specimen_0*//'
053-pip-ctrlin           - 1m50s: +++ sed 's/specimen_0*//'
053-pip-ctrlin           - 1m50s: ++ echo build/1/specimen_015
053-pip-ctrlin           - 1m50s: make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
053-pip-ctrlin           - 1m50s: ++++ pwd
053-pip-ctrlin           - 1m50s: ++ md5sum
053-pip-ctrlin           - 1m50s: ++ export SEED=9fd99230
053-pip-ctrlin           - 1m50s: ++ SEED=9fd99230
053-pip-ctrlin           - 1m50s: ++ cut -c1-8
053-pip-ctrlin           - 1m50s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin/build/1/specimen_018
053-pip-ctrlin           - 1m50s: ++ export SEEDN=17
053-pip-ctrlin           - 1m50s: ++ SEEDN=17
053-pip-ctrlin           - 1m50s: +++ sed 's/specimen_0*//'
053-pip-ctrlin           - 1m50s: ++ cd build/1/specimen_020
053-pip-ctrlin           - 1m50s: + echo '`define SEED 32'\''hd19e3576'
053-pip-ctrlin           - 1m50s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
053-pip-ctrlin           - 1m50s: ++++ pwd
053-pip-ctrlin           - 1m50s: + echo '`define SEED 32'\''h60c717f3'
053-pip-ctrlin           - 1m50s: ++ export SEEDN=18
053-pip-ctrlin           - 1m50s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
053-pip-ctrlin           - 1m50s: ++ SEEDN=18
053-pip-ctrlin           - 1m50s: +++ echo build/1/specimen_020
053-pip-ctrlin           - 1m50s: +++ md5sum
053-pip-ctrlin           - 1m50s: ++ echo build/1/specimen_018
053-pip-ctrlin           - 1m50s: ++ echo build/1/specimen_017
[... 23 lines cut ...]
053-pip-ctrlin           - 1m50s: +++ sed 's/specimen_0*//'
053-pip-ctrlin           - 1m50s: ++++ pwd
053-pip-ctrlin           - 1m50s: +++ basename /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin/build/1/specimen_020
053-pip-ctrlin           - 1m50s: make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
053-pip-ctrlin           - 1m50s: ++ export SEEDN=20
053-pip-ctrlin           - 1m50s: ++ SEEDN=20
[... cut ...]
053-pip-ctrlin           - 1m50s: make[2]: *** [build/1/specimen_019/OK] Error 2
053-pip-ctrlin           - 1m50s: make[1]: *** [run] Error 2
053-pip-ctrlin           - 1m50s: 
053-pip-ctrlin           - 1m50s: --------------------------------------------------------------------------
053-pip-ctrlin           - 1m50s: Failed @ 2019-02-05T14:12:38.089913 with exit code: 2
Makefile:75: recipe for target '053-pip-ctrlin/run.ok' failed
make: *** [053-pip-ctrlin/run.ok] Error 2
make: *** Waiting for unfinished jobs....
```

mithro, 2019-02-06:
```
------------------------------------------ 
  Failed to route net mynet_36, status UNROUTED, route: CLBLM_L_X20Y112/CLBLM_L_A CLBLM_L_X20Y112/CLBLM_LOGIC_OUTS8 INT_L_X20Y112/NN6BEG0 INT_L_X20Y136/LV_L18 INT_L_X20Y136/WW4BEG3 INT_L_X16Y136/SW2BEG2 INT_R_X15Y135/WW4BEG3 INT_R_X11Y135/LVB12 INT_R_X11Y135/EE4BEG2 INT_R_X15Y135/CTRL1 CLBLL_R_X15Y135/CLBLL_LL_SR
Failed to route net
    while executing
"error "Failed to route net""
    invoked from within
"if { $assert } {
            error "Failed to route net"
        }"
    invoked from within
"if { $status != "ROUTED" } {
        puts "  Failed to route net $net, status $status, route: $fixed_route"
        if { $assert } {
            error..."
    (procedure "route_via" line 43)
    invoked from within
"route_via $mynet "$tile/$src_wire $tile/$dst_wire""
    ("for" body line 25)
    invoked from within
"for {set idx 0} {$idx < [llength $todo_lines]} {incr idx} {
    set line [lindex $todo_lines $idx]
    puts "== $idx: $line"

    set tile_type [linde..."
    (file "/tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/fuzzers/053-pip-ctrlin/generate.tcl" line 35)
Routing net mynet_29:
  set route [find_routing_path -quiet -from CLBLL_R_X17Y125/CLBLL_L_A -to INT_L_X24Y111/SW6BEG1]
  set route [find_routing_path -quiet -from INT_L_X14Y104/GFAN0 -to INT_L_X14Y104/CTRL_L0]
  INT_L_X14Y104/GFAN0 -> INT_L_X14Y104/CTRL_L0: no route found - assuming direct PIP
Routing net mynet_49:
INFO: [Common 17-206] Exiting Vivado at Wed Feb  6 00:12:03 2019...
  set route [find_routing_path -quiet -from CLBLL_L_X16Y118/CLBLL_L_A -to INT_R_X9Y121/WW4BEG2]
  set route [find_routing_path -quiet -from INT_L_X14Y104/CTRL_L0 -to CLBLL_L_X14Y104/CLBLL_L_SR]
  INT_L_X14Y104/CTRL_L0 -> CLBLL_L_X14Y104/CLBLL_L_SR: no route found - assuming direct PIP
  CLBLL_L_X2Y138/CLBLL_L_A -> INT_R_X21Y139/WW4BEG2: CLBLL_L_X2Y138/CLBLL_L_A CLBLL_L_X2Y138/CLBLL_LOGIC_OUTS8 INT_L_X2Y138/EE4BEG0 INT_L_X6Y138/EE4BEG0 INT_L_X10Y138/EE4BEG0 INT_L_X14Y138/EE4BEG0 INT_L_X18Y138/ER1BEG1 INT_R_X19Y138/ER1BEG2 INT_L_X20Y138/NE2BEG2 INT_R_X21Y139/WW4BEG2
```

litghost, 2019-02-06:
The failed to route net is a probabilistic failure we have seen in other fuzzers.  Solution was to wrap the "route_via" in a retry loop.  Let me see if I can refactor the loop for generic usage.

litghost, 2019-02-06:
#615 will help some, but I think #460 is also cropping up.  In general, we probably need to add advisory locks for reading and writing of database files.

PR #615 (2019-02-06): Add retries to 053 and 055 INT fuzzers.
Fixes #613 

051 doesn't have retries, but I don't think we've seen a routing failure yet.
closed 2019-02-06 by litghost