f4pga/prjxray #618: AssertionError: Tile INT_L_X24Y90 does not have bitstream info but 4483 tags

mithro, 2019-02-07:
```
2019-02-07T03:42:43.913561 - 050-pip-seed             - 14m27s: + python3 /tmpfs/src/github/symbiflow-prjxray-continuous-db-kintex7/fuzzers/050-pip-seed/generate.py
2019-02-07T03:42:43.913561 - 050-pip-seed             - 14m27s: Traceback (most recent call last):
2019-02-07T03:42:43.913561 - 050-pip-seed             - 14m27s:   File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-kintex7/fuzzers/050-pip-seed/generate.py", line 68, in <module>
2019-02-07T03:42:43.913561 - 050-pip-seed             - 14m27s:     segmk.compile(bitfilter=get_bitfilter(os.getenv('XRAY_DATABASE'), 'INT'))
2019-02-07T03:42:43.913561 - 050-pip-seed             - 14m27s:   File "/tmpfs/src/github/symbiflow-prjxray-continuous-db-kintex7/prjxray/segmaker.py", line 348, in compile
2019-02-07T03:42:43.913561 - 050-pip-seed             - 14m27s:     tilename, this_tile_tags)
2019-02-07T03:42:43.913561 - 050-pip-seed             - 14m27s: AssertionError: Tile INT_L_X24Y90 does not have bitstream info but 4483 tags
```

litghost, 2019-02-07:
Which part is this?

litghost, 2019-02-07:
If this is Kintex7, this could be due to #566

mithro, 2019-02-07:
Give https://source.cloud.google.com/results/invocations/19862a66-f746-45ad-97e8-b3f9140f291d/targets a look...

litghost, 2019-02-07:
Ya, that's a Kintex build,

litghost, 2019-02-07:
Looks like I'll need to completely solve for INT tiles (or filter them):

2019-02-07T19:49:19.566035 - 051-pip-imuxlout-bypalts - 14m14s: AssertionError: Tile INT_R_X25Y24 does not have bitstream info but 33 tags
2019-02-07T19:49:19.566035 - 051-pip-imuxlout-bypalts - 14m14s: make[3]: *** [generate.ok] Error 1

Working on it.

PR #621 (2019-02-07): Add XADC INT and ICAP INT fuzzers to solve 2 of 4 missing INT columns.
Fixes #618, and solves 2 of the 4 missing INT columns from #566 .