f4pga/prjxray #626: -j option not getting passed down through some fuzzer scripts

mithro, 2019-02-08:
Make automatically detects commands that start with `$(MAKE)` and passes the jobserver information down to the submake process. If make is calling another tool which then calls make, it needs to be prefixed with a `+` symbol.

For example;
```Makefile

target1:
	# Make doesn't know to pass jobserver / -j information down to this script.
	shell_script_which_runs_make.sh
```

```Makefile

target1:
	# Make now knows that this command will launch submake processes.
	+shell_script_which_runs_make.sh
```

We need to audit all our fuzzers are doing the right thing here. You will get the following error message otherwise;
```
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
```

mithro, 2019-04-04:
@tmichalak Is this done now?

tmichalak, 2019-04-05:
@mithro yes, all necessary changes have been merged a while back and I didn't see any of the warning messages for quite some time.

mithro, 2019-04-05:
Great!

closed 2019-04-05 by mithro