f4pga/prjxray #643: Documentation License

SadAlpaca, 2019-02-13:
This git repo for documenting the Xilinx 7-series bit-stream format says in two places (in readme and contributing) that "all _code_ in the Project X-Ray repository is licensed under the very permissive ISC Licence." (emphasis added.) The license itself also says permission is being granted for the _software_ (emphasis also added.) Does that not include the documentation of the Xilinx 7-series bit-stream format that's being made? Why _only_ the software and not everything?

mithro, 2019-02-13:
The [database itself is available under CC0-1-Universal here](https://github.com/SymbiFlow/prjxray-db).

SadAlpaca, 2019-02-13:
Thanks but I mean the non-code/non-software stuff in _this_ one. An example being the docs directory. Since it's not code I take it that it doesn't fall into the ""all code in the Project X-Ray repository is licensed under the very permissive ISC Licence." Why _only_ the software and not everything in this git repository? The vtr-verilog-to-routing repo makes it quite clear by comparison that it includes everything and not _just_ the code/software itself.

PR #679 (2019-02-27): Small fix to README in license section.
Fixes #643.
commit ac8f228a (2019-02-27): Small fix to README in license section.

Fixes #643.
closed 2019-02-27 by litghost