f4pga/prjxray #657: Collision in latest Kintex7 CI

litghost, 2019-02-14:
ERROR: 1 collisions
  00400c87_096_22: had INT_R_X25Y97.INT_R.NR1BEG3.EL1END3, got CLK_BUFG_BOT_R_X67Y100.CLK_BUFG_BOT_R.BUFGCTRL.BUFGCTRL_X0Y15.PRESELECT_I1

litghost, 2019-02-14:
```CLK_BUFG_BOT_R.BUFGCTRL.BUFGCTRL_X0Y15.PRESELECT_I1 07_118 26_252```

Segmatch should be using "-c 1"

litghost, 2019-02-14:
Given that the same count is already 50, this fuzzer likely needs additional stimulus or a bitfilter.

litghost, 2019-02-19:
This fuzzer has been persistently unstable, disabling for now.

tmichalak, 2019-03-04:
I reran the fuzzer several time and I was getting the following result consistently:
<pre>
WARNING: got duplicate tag CLK_BUFG_TOP_R.BUFGCTRL.BUFGCTRL_X0Y6.PRESELECT_I1
  Orig line: CLK_BUFG_TOP_R.BUFGCTRL.BUFGCTRL_X0Y6.PRESELECT_I1 26_108
  New line : CLK_BUFG_TOP_R.BUFGCTRL.BUFGCTRL_X0Y6.PRESELECT_I1 18_192 26_108
</pre>
However, I noticed that the last change to the Makfile (commit d9251986 by @litghost ) reduced the specimen count to 40. 
When I changed it back to 50 I started to consistently get solely bit ``26_108`` for ``CLK_BUFG_TOP_R.BUFGCTRL.BUFGCTRL_X0Y6.PRESELECT_I1`` and pushdb finished without any conflicts.
Maybe we should restore the count to 50 and enable the fuzzer in master?

litghost, 2019-03-04:
Try a PR at 50, but I believe it was still unstable.

litghost, 2019-03-06:
In theory https://github.com/SymbiFlow/prjxray/pull/695 fixed this.

PR #664 (2019-02-18): Simplify routing noise in 042
Fixes #657
closed 2019-03-06 by litghost