f4pga/prjxray #661: 057-pip-bi: Fuzzer failed CI

acomodi, 2019-02-15:
This happens with both kintex7 and zynq7 and is unstable from one build to another.

litghost, 2019-02-18:
I've also seen 052 fail, and maybe 053 fail.  It does seem to affect kintex7 and zynq7 more so than artix7.

litghost, 2019-02-20:
Latest from last run:

zync:
2019-02-20T08:22:23.782547 - 053-pip-ctrlin           - 1h25m: WARNING: Fuzzer failed!
kintex:
2019-02-20T07:00:51.396308 - 052-pip-clkin            - 31m35s: WARNING: Fuzzer failed!
2019-02-20T10:39:33.129469 - 056-pip-rem              - 19m17s: WARNING: Fuzzer failed!
2019-02-20T16:00:09.852344 - 057-pip-bi               - 1m40s: WARNING: Fuzzer failed!
2019-02-20T16:15:12.263067 - 057-pip-bi               - 16m42s: WARNING: Fuzzer failed!
2019-02-20T16:25:30.745754 - 057-pip-bi               - 27m01s: WARNING: Fuzzer failed!
artix:
<no failures>

acomodi, 2019-02-20:
One possibility is to increase the number of tries, but it will still be non-deterministic and could end up in failure

litghost, 2019-02-20:
I think it's worth cataloging where the faliures are occuring (e.g. which PIPs).

kgugala, 2019-03-04:
it happened here https://github.com/SymbiFlow/prjxray/pull/691 for Kintex on fuzzer 057:

```
2019-03-03T20:35:01.646102 - 057-pip-bi               - 2m48s: + make -f /tmpfs/src/github/symbiflow-prjxray-presubmit-db-kintex7/utils/top_generate.mk^M
2019-03-03T20:35:01.646102 - 057-pip-bi               - 2m48s: make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.^M
2019-03-03T20:35:01.646102 - 057-pip-bi               - 2m48s: make[3]: *** [vivado.ok] Error 1^M
2019-03-03T20:35:01.646102 - 057-pip-bi               - 2m48s: make[2]: *** [build/1/specimen_005/OK] Error 2^M
2019-03-03T20:35:01.646102 - 057-pip-bi               - 2m48s: make[2]: *** Waiting for unfinished jobs....^M
2019-03-03T20:35:01.646102 - 057-pip-bi               - 2m48s: make[1]: *** [run] Error 2^M
```

The log does not provide any additional info. The build passed for bothe Artix and Zynq.

litghost, 2019-03-06:
```
2019-03-06T02:10:14.955164 - 056-pip-rem              - 11m05s: ERROR: [DRC PDIL-1] Invalid Site Configuration: Invalid configuration for site SLICE_X21Y101. Reason: Cannot replace net mynet_32 with net mynet_9 on bel pin A/A
2019-03-06T02:10:14.955164 - 056-pip-rem              - 11m05s: , Cannot replace net mynet_32 with net mynet_9 on bel pin AUSED/0
2019-03-06T02:10:14.955164 - 056-pip-rem              - 11m05s: , Cannot replace net mynet_32 with net mynet_9 on bel pin AUSED/OUT
2019-03-06T02:10:14.955164 - 056-pip-rem              - 11m05s: , Site pin to site pin route-thru requires conflicting attribute string values for user logic element 'A6LUT in site 'SLICE_X21Y101'. Attribute 'EQN' is programmed to '2'h0' but needs a value of 'O6=A2' for the route-thru.
2019-03-06T02:10:14.955164 - 056-pip-rem              - 11m05s: .
2019-03-06T02:10:14.955164 - 056-pip-rem              - 11m05s: ERROR: [DRC RTSTAT-5] Partial antennas: 1 net(s) have a partial antenna. The problem bus(es) and/or net(s) are mynet_32.
2019-03-06T02:10:14.955164 - 056-pip-rem              - 11m05s: ERROR: [DRC RTSTAT-6] Partial route conflicts: 2 net(s) have a partial conflict. The problem bus(es) and/or net(s) are mynet_9, and mynet_32.
2019-03-06T02:10:14.955164 - 056-pip-rem              - 11m05s: ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net mynet_32. The following site(s) are invalid: SLICE_X21Y101.
```

tmichalak, 2019-03-06:
<pre>
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m: + make -f /tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/top_generate.mk
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m: make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m: Traceback (most recent call last):
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/dbfixup.py", line 426, in <module>
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:     main()
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/dbfixup.py", line 422, in main
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:     verbose=args.verbose)
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/dbfixup.py", line 395, in run
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:     verbose=verbose)
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/dbfixup.py", line 372, in update_segs
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:     fn_inouts, zero_db, clb_int, lazy=lazy, strict=strict, verbose=verbose)
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/dbfixup.py", line 290, in update_seg_fns
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:     fn_in, zero_db, clb_int=clb_int, strict=strict, verbose=verbose)
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/dbfixup.py", line 153, in add_zero_bits
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:     zero_range(tag, bits, 22, 25)
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/dbfixup.py", line 36, in zero_range
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:     raise ValueError("%s: inconsistent bit index" % tag)
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m: ValueError: INT.IMUX_L22.BYP_BOUNCE2: inconsistent bit index
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m: make[2]: *** [database] Error 1
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m: make[1]: *** [run] Error 2
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m:
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m: --------------------------------------------------------------------------
2019-03-06T04:22:40.471050 - 056-pip-rem              - 2h31m: Failed @ 2019-03-06T04:22:40.423857 with exit code: 2
2019-03-06T04:22:40.485522 - 056-pip-rem              - 2h31m: WARNING: Fuzzer failed!
Makefile:94: recipe for target '056-pip-rem/run.ok' failed
</pre>

tmichalak, 2019-05-31:
All issues with the symptoms described within this issue have been fixed.
We can close this ticket as there are separate issues being opened for all other stabilization problems

closed 2019-05-31 by tmichalak