f4pga/prjxray #684: Document all bits used in the basys3 SWBUT ROI

litghost, 2019-03-01:


litghost, 2019-03-01:
Unknown clocking bits:

```
# CLK_HROW or BUFG_REBUF
{ unknown_bit = "00400b9a_93_8", unknown_segment = "0x00400b80", unknown_segbit = "26_2984" }
{ unknown_bit = "00400b9a_42_25", unknown_segment = "0x00400b80", unknown_segbit = "26_1369" }
{ unknown_bit = "00400b9a_42_28", unknown_segment = "0x00400b80", unknown_segbit = "26_1372" }

# Likely CMT
{ unknown_bit = "0002009b_50_25", unknown_segment = "0x00020080", unknown_segbit = "27_1625" }
{ unknown_bit = "0040151b_49_25", unknown_segment = "0x00401500", unknown_segbit = "27_1593" }
{ unknown_bit = "0040151b_49_31", unknown_segment = "0x00401500", unknown_segbit = "27_1599" }
{ unknown_bit = "0040151b_49_29", unknown_segment = "0x00401500", unknown_segbit = "27_1597" }
{ unknown_bit = "0040151b_49_23", unknown_segment = "0x00401500", unknown_segbit = "27_1591" }

# BUFG
{ unknown_bit = "00400b9b_93_14", unknown_segment = "0x00400b80", unknown_segbit = "27_2990" }
```

litghost, 2019-03-01:
13 IOB bits appear to belong to a _SING tile, so likely solving #341 will solve those 13 bits.

litghost, 2019-03-01:
Confirmed, solving #341 solves all 13 of the IOB bits.

tmichalak, 2019-03-06:
I get the following list of missing bits generated by bit2fasm with the database from prjxray-db:
<pre>
# In frame 0x00400026 3 bits were not converted.
{ unknown_bit = "00400026_89_30", unknown_segment = "0x00400000", unknown_segbit = "38_2878" }
{ unknown_bit = "00400026_90_22", unknown_segment = "0x00400000", unknown_segbit = "38_2902" }
{ unknown_bit = "00400026_89_22", unknown_segment = "0x00400000", unknown_segbit = "38_2870" }

# In frame 0x00400027 6 bits were not converted.
{ unknown_bit = "00400027_90_29", unknown_segment = "0x00400000", unknown_segbit = "39_2909" }
{ unknown_bit = "00400027_89_23", unknown_segment = "0x00400000", unknown_segbit = "39_2871" }
{ unknown_bit = "00400027_90_17", unknown_segment = "0x00400000", unknown_segbit = "39_2897" }
{ unknown_bit = "00400027_89_21", unknown_segment = "0x00400000", unknown_segbit = "39_2869" }
{ unknown_bit = "00400027_90_23", unknown_segment = "0x00400000", unknown_segbit = "39_2903" }
{ unknown_bit = "00400027_89_29", unknown_segment = "0x00400000", unknown_segbit = "39_2877" }

# In frame 0x00000c9a 1 bits were not converted.
{ unknown_bit = "00000c9a_7_28", unknown_segment = "0x00000c80", unknown_segbit = "26_252" }
</pre>                                                                           
It's different from what @litghost posted few days ago and I am not sure if the presumptions about what these bits belong it still stand.
I guess my question here is to @litghos: how did you generate the previous list?

litghost, 2019-03-06:
That's likely just a different clock route.  Those bits match my expectations.

litghost, 2019-03-13:
I know what the bits that are located in the HROW are, they are the L/R to BUFG pips.  I'll update 041 to fuzz for those pips as well.

litghost, 2019-03-15:
> I know what the bits that are located in the CLK_HROW are, they are the L/R to BUFG pips. I'll update 041 to fuzz for those pips as well.

2 more bits were indeed part of the CLK_HROW switch box, leaving 5 pips left.  4 of the pips appear to belong to the HCLK_CMT switch box, I'll work on documenting that next.

litghost, 2019-03-16:
2 more bits were indeed part of the HCLK_CMT switch box, remaining bits are:

```
{ unknown_bit = "0040151b_49_29", unknown_segment = "0x00401500", unknown_segbit = "27_1597" }
{ unknown_bit = "0040151b_49_31", unknown_segment = "0x00401500", unknown_segbit = "27_1599" }
{ unknown_bit = "0002009b_50_25", unknown_segment = "0x00020080", unknown_segbit = "27_1625" }
```

litghost, 2019-03-16:
Correction, 4 of the bits were in the HCLK_CMT switch box, only remaining bits are:
```
{ unknown_bit = "0002009b_50_25", unknown_segment = "0x00020080", unknown_segbit = "27_1625" }
```

I'm guessing this is a bit set when the RIOI3_X43Y25/RIOI3.IOI_ILOGIC0_O->RIOI_I2GCLK_TOP0 PIP is set.  I'll test that theory in a bit.

litghost, 2019-03-16:
Hooray, all bits are solved!

PR #727 (2019-03-22): Solve remaining bits in the ROI
This PR add fuzzers for the BUFG and HCLK_CMT interconnect, and completes the interconnect fuzzing of the CLK_HROW switch box.

With these fuzzers, all remaining bits in the basys3 ROI are solved, fixes #684.  This PR superceeds #717, but the comments from #717 apply here.
closed 2019-03-22 by litghost