f4pga/prjxray #697: 005-tilegrid fails for Zynq 7020

herbertp, 2019-03-05:
```
# git clone https://github.com/SymbiFlow/prjxray.git
   follow steps 1,2,4,5.1
# sed -i 's/xc7z010clg400/xc7z020clg400/' settings/zynq7.sh
# source settings/zynq7.sh
# (cd fuzzers/001-part-yaml; make run)
# cd fuzzers/005-tilegrid/
# sed -i '/TILEGRID.*iob/ s/^/#/' Makefile
# sed -i 's/== "xc7z010clg400-1"/in ("xc7z010clg400-1", "xc7z020clg400-1")/' ps7_int/top.py
# make
```
... fails with ...

```
python3 add_tdb.py \
	--fn-in build/basicdb/tilegrid.json \
	--fn-out build/tilegrid_tdb.json
Traceback (most recent call last):
  File "add_tdb.py", line 135, in <module>
    main()
  File "add_tdb.py", line 131, in main
    run(args.fn_in, args.fn_out, verbose=args.verbose)
  File "add_tdb.py", line 108, in run
    tilej = database[tile]
KeyError: 'INT_L_X0Y50'
Makefile:108: recipe for target 'build/tilegrid_tdb.json' failed
make: *** [build/tilegrid_tdb.json] Error 1
```

herbertp, 2019-03-05:
```
$ grep -r INT_L_X0Y50 *
ps7_int/build/specimen_001/params.csv:INT_L_X0Y50,0
ps7_int/build/specimen_001/segdata_tilegrid.txt:tag INT_L_X0Y50.DWORD:0.DBIT:17.DFRAME:14 1
ps7_int/build/segbits_tilegrid.tdb:INT_L_X0Y50.DWORD:0.DBIT:17.DFRAME:14 00000914_000_17 00000918_000_17
ps7_int/build/specimen_002/params.csv:INT_L_X0Y50,1
ps7_int/build/specimen_002/segdata_tilegrid.txt:tag INT_L_X0Y50.DWORD:0.DBIT:17.DFRAME:14 0
ps7_int/top.py:        params['INT_L_X0Y50'] = isone

```

acomodi, 2019-03-05:
I think that `ps7_int` can be avoided for now. PS7 is directly connected to the first column INT tiles, and IMO these can be found even without the `ps7_int`. As it is now, `ps7_int` is most probably highly part-specific and will cause this failure.

mithro, 2019-06-12:
@tmichalak Do you know if this should work now?

tmichalak, 2019-06-13:
@mithro just like @acomodi wrote this problem is part specific.This means that the problem is still there and we need to adjust the ps7 tilegrid fuzzer for 7020.

tmichalak, 2021-02-02:
Zynq 7020 is currently the main fuzzing part for the zynq database. The fuzzers seem to run fine in CI. Hence, this issue is no longer valid.

closed 2021-02-02 by tmichalak