f4pga/prjxray #703: Building roi_harness for zybo fails

mithro, 2019-03-08:
```
WARNING: [Vivado 12-3286] Net, clk, could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, din[0], could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, din[1], could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, din[2], could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, din[3], could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, dout[0], could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, dout[1], could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, dout[2], could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, dout[3], could not be marked fixed.  The net is either fully contained within a site or is unrouted.
WARNING: [Vivado 12-3286] Net, <const1>, could not be marked fixed.  The net is either fully contained within a site or is unrouted.
# write_checkpoint -force design.dcp
Writing placer database...
Writing XDEF routing.
Writing XDEF routing logical nets.
Writing XDEF routing special nets.
Write XDEF Complete: Time (s): cpu = 00:00:00.02 ; elapsed = 00:00:00.01 . Memory (MB): peak = 2017.582 ; gain = 0.000 ; free physical = 124649 ; free virtual = 342994
INFO: [Common 17-1381] The checkpoint '/big/github/SymbiFlow/prjxray-zynq/minitests/roi_harness/build/design.dcp' has been generated.
# write_bitstream -force design.bit
Command: write_bitstream -force design.bit
Attempting to get a license for feature 'Implementation' and/or device 'xc7z010-clg400'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7z010-clg400'
Running DRC as a precondition to command write_bitstream
Command: report_drc (run_mandatory_drcs) for: bitstream_checks
INFO: [DRC 23-27] Running DRC with 8 threads
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port din[0] expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port din[1] expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port din[2] expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC RPBF-3] IO port buffering is incomplete: Device port din[3] expects both input and output buffering but the buffers are incomplete.
WARNING: [DRC ZPS7-1] PS7 block required: The PS7 cell must be used in this Zynq design in order to enable correct default configuration.
report_drc (run_mandatory_drcs) completed successfully
INFO: [Vivado 12-3199] DRC finished with 0 Errors, 5 Warnings
INFO: [Vivado 12-3200] Please refer to the DRC report (report_drc) for more information.
INFO: [Project 1-821] Please set project.enableDesignId to be 'true'.
INFO: [Designutils 20-2272] Running write_bitstream with 8 threads.
Loading data files...
Loading site data...
Loading route data...
Processing options...
Creating bitmap...
Creating bitstream...
Writing bitstream ./design.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-120] WebTalk data collection is mandatory when using a WebPACK part without a full Vivado license. To see the specific WebTalk data collected for your design, open the usage_statistics_webtalk.html or usage_statistics_webtalk.xml file in the implementation directory.
INFO: [Common 17-186] '/big/github/SymbiFlow/prjxray-zynq/minitests/roi_harness/build/design/usage_statistics_webtalk.xml' has been successfully sent to Xilinx on Fri Mar  8 00:34:01 2019. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2017.2/doc/webtalk_introduction.html.
9 Infos, 5 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:09 ; elapsed = 00:00:09 . Memory (MB): peak = 2351.000 ; gain = 333.418 ; free physical = 124630 ; free virtual = 342974
INFO: [Common 17-206] Exiting Vivado at Fri Mar  8 00:34:01 2019...
++ fgrep CRITICAL vivado.log
+ test -z ''
+ /big/github/SymbiFlow/prjxray-zynq/build/tools/bitread --part_file /big/github/SymbiFlow/prjxray-zynq/database/zynq7/xc7z010clg400-1.yaml -F 0x00000000:0xffffffff -o design.bits -z -y design.bit
Bitstream size: 2083839 bytes
Config size: 520922 words
Number of configuration frames: 5144
DONE
+ python3 /big/github/SymbiFlow/prjxray-zynq/utils/bit2fasm.py --verbose design.bit
+ python3 /big/github/SymbiFlow/prjxray-zynq/utils/fasm2frames.py design.fasm design.frm
+ PYTHONPATH=/big/github/SymbiFlow/prjxray-zynq:/big/github/SymbiFlow/prjxray-zynq/third_party/fasm:/big/github/SymbiFlow/prjxray-zynq:/big/github/SymbiFlow/prjxray-zynq/third_party/fasm:/big/github/SymbiFlow/prjxray-zynq:/big/github/SymbiFlow/prjxray-zynq/third_party/fasm::/big/github/SymbiFlow/prjxray-zynq/utils
+ python3 ../create_design_json.py --design_info_txt design_info.txt --design_txt design.txt --pad_wires design_pad_wires.txt --design_fasm design.fasm
Traceback (most recent call last):
  File "../create_design_json.py", line 119, in <module>
    main()
  File "../create_design_json.py", line 92, in main
    unknown_base_address)
AssertionError: Found unknown bit in base address 0x00001380
Makefile:14: recipe for target 'build/design.json' failed
make: *** [build/design.json] Error 1
```

mithro, 2019-03-08:
[build.zip](https://github.com/SymbiFlow/prjxray/files/2943963/build.zip)

litghost, 2019-03-08:
There are 5 unknown bits in that base address:
```
# In frame 0x0000139a 3 bits were not converted.
{ unknown_bit = "0000139a_0_8", unknown_segment = "0x00001380", unknown_segbit = "26_08" }
{ unknown_bit = "0000139a_42_25", unknown_segment = "0x00001380", unknown_segbit = "26_1369" }
{ unknown_bit = "0000139a_42_28", unknown_segment = "0x00001380", unknown_segbit = "26_1372" }

# In frame 0x0000139b 2 bits were not converted.
{ unknown_bit = "0000139b_24_12", unknown_segment = "0x00001380", unknown_segbit = "27_780" }
{ unknown_bit = "0000139b_0_14", unknown_segment = "0x00001380", unknown_segbit = "27_14" }
```

litghost, 2019-03-08:
So there is probably a problem with the CLK_HROW fuzzer on zync, because those bits are understood in artix7:
https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/segbits_clk_hrow_top_r.db#L874

mithro, 2019-03-08:
That exact same line appears in the zynq db?
https://github.com/SymbiFlow/prjxray-db/blob/master/zynq7/segbits_clk_hrow_top_r.db#L874

litghost, 2019-03-08:
In a7 it is:

```
CLK_HROW_TOP_R.CLK_HROW_CK_MUX_OUT_R0.CLK_HROW_CK_IN_L5 26_08 27_14
```

Which is not in z7.

mithro, 2019-03-08:
It appears I'm just not good at github. It seems something to do with the L[0-13] lines;
```
tansell@instance-1:~/github/SymbiFlow/prjxray/database$ diff -u artix7/segbits_clk_hrow_bot_r.db zynq7/segbits_clk_hrow_bot_r.db
--- artix7/segbits_clk_hrow_bot_r.db    2019-03-01 23:59:10.534119935 +0000
+++ zynq7/segbits_clk_hrow_bot_r.db     2019-03-02 00:01:13.952688062 +0000
@@ -21,7 +21,7 @@                                                     
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X0Y5.CE_TYPE.ASYNC 29_86                
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X0Y5.IN_USE 29_87                       
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X0Y5.INIT_OUT 28_83                        
-CLK_HROW_BOT_R.BUFHCE.BUFHCE_X0Y5.ZINV_CE 29_83                         
+CLK_HROW_BOT_R.BUFHCE.BUFHCE_X0Y5.ZINV_CE 27_199 29_83                  
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X0Y6.CE_TYPE.ASYNC 27_134               
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X0Y6.IN_USE 27_135                      
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X0Y6.INIT_OUT 26_131                    
@@ -94,20 +94,6 @@                                                     
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X1Y11.IN_USE 29_215                      
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X1Y11.INIT_OUT 28_211                    
 CLK_HROW_BOT_R.BUFHCE.BUFHCE_X1Y11.ZINV_CE 29_211                     
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L0_ACTIVE 26_114                       
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L1_ACTIVE 27_114                       
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L2_ACTIVE 26_115                       
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L3_ACTIVE 27_115                        
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L4_ACTIVE 26_116                          
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L5_ACTIVE 27_116                          
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L6_ACTIVE 26_117                          
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L7_ACTIVE 27_117                        
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L8_ACTIVE 26_118                        
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L9_ACTIVE 27_118                        
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L10_ACTIVE 26_80                        
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L11_ACTIVE 27_80                          
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L12_ACTIVE 26_81                          
-CLK_HROW_BOT_R.CLK_HROW_CK_IN_L13_ACTIVE 27_81                          
 CLK_HROW_BOT_R.CLK_HROW_CK_IN_R0_ACTIVE 28_114                        
 CLK_HROW_BOT_R.CLK_HROW_CK_IN_R1_ACTIVE 29_114                        
 CLK_HROW_BOT_R.CLK_HROW_CK_IN_R2_ACTIVE 28_115                        
@@ -122,20 +108,6 @@                                                    
 CLK_HROW_BOT_R.CLK_HROW_CK_IN_R11_ACTIVE 29_80                         
 CLK_HROW_BOT_R.CLK_HROW_CK_IN_R12_ACTIVE 28_81                         
 CLK_HROW_BOT_R.CLK_HROW_CK_IN_R13_ACTIVE 29_81                         
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L0 26_56 26_60   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L1 26_56 27_60   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L2 26_56 26_61   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L3 26_56 27_61   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L4 26_56 26_62      
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L5 26_56 27_62      
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L6 26_56 26_63      
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L7 26_56 27_63   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L8 26_60 27_56   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L9 27_56 27_60   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L10 26_61 27_56  
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L11 27_56 27_61    
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L12 26_62 27_56    
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_L13 27_56 27_62    
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_R0 26_57 26_60   
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_R1 26_57 27_60   
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_CK_IN_R2 26_57 26_61   
@@ -184,20 +156,6 @@                                                    
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_R_CK_GCLK29 26_58 27_62 
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_R_CK_GCLK30 26_58 26_63 
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L0.CLK_HROW_R_CK_GCLK31 26_58 27_63 
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L0 28_56 28_60   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L1 28_56 29_60   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L2 28_56 28_61   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L3 28_56 29_61    
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L4 28_56 28_62      
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L5 28_56 29_62      
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L6 28_56 28_63      
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L7 28_56 29_63   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L8 28_60 29_56   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L9 29_56 29_60   
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L10 28_61 29_56
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L11 29_56 29_61     
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L12 28_62 29_56     
-CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_L13 29_56 29_62     
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_R0 28_57 28_60   
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_R1 28_57 29_60   
 CLK_HROW_BOT_R.CLK_HROW_CK_MUX_OUT_L1.CLK_HROW_CK_IN_R2 28_57 28_61   
....
```

mithro, 2019-03-08:
https://github.com/SymbiFlow/symbiflow-arch-defs/issues/448

litghost, 2019-03-12:
Looks like the BUFG tile is in the Zynq harness, which explains the problem, in addition to potential bugs in the 041 fuzzer.

PR #711 (2019-03-12): zybo: Fix Zybq Zynq-7 roi_harness
fixes #703
closed 2019-03-12 by litghost