f4pga/prjxray #713: Test output on CI cleanup needed

mithro, 2019-03-12:
The following things about the test output on the CI system needs cleanup;
 - [ ] Remove `logs` from the level name in `fuzzers/0xx-xxx-xxx/logs`
 - [ ] [Target log](https://source.cloud.google.com/results/invocations/43addc3d-62f3-4b3d-9f54-c80f34b64cc3/targets;collapsed=/fuzzers%2F060-bram-cascades%2Flogs/log) output just says "Target built."
 - [ ] Duration is always listed as the same amount
 - [ ] Retries are not appearing in the log output 
 - [ ] Artefacts output is pretty useless

litghost, 2019-03-12:
> Remove logs from the level name in fuzzers/0xx-xxx-xxx/logs

We want to keep those logs.

mithro, 2019-03-12:
> Remove logs from the level **name** in `fuzzers/0xx-xxx-xxx/logs`

tmichalak, 2019-03-14:
> Remove logs from the level name in fuzzers/0xx-xxx-xxx/logs

So we basically want to keep the logs, but in the fuzzer directory?

> Target log output just says "Target built."

I didn't find "Target built." in the target log. What should the output be instead of "Target built." ?

> Artefacts output is pretty useless

What output do you mean exactly?  Do we want to change it to something more meaningful or get rid of it entirely?

tmichalak, 2019-03-14:
> Duration is always listed as the same amount
Duration is updated every 10 seconds. This is due to the timeout in the wait method of the subprocess we run the fuzzer on that we use for checking the status of its execution. Upon success we also display 100 lines of the stdout log. All of the lines are prefixed with the string holding the duration value from the moment when the log method was called.
What is the desired output?

mithro, 2019-03-14:
Have you looked at the CI output from Kokoro? See [this for example](https://source.cloud.google.com/results/invocations/43addc3d-62f3-4b3d-9f54-c80f34b64cc3/targets).

tmichalak, 2019-03-15:
Yes. I did. However, I still don't know what is the desired outcome in some of the points above.
Could you provide more details?

tmichalak, 2019-05-15:
@mithro: This issue is the last one on ToDo list in the "Stabilize prjxray output" project (https://github.com/orgs/SymbiFlow/projects/5).
I don't know how urgent this is on the priorities list, but if the points are still valid then I would like to address them.
However, I need some clarification in order to do that:

> Remove logs from the target name in fuzzers/0xx-xxx-xxx/logs

What do you mean by _target name_? Could you be more specific what you have in mind here?

> Target log output just says "Target built."

Currently the target log, e.g. ``fuzzers/030-iob/logs/`` has ``stderr`` and ``stdout`` logs with the entire execution. Maybe this is working fine now?

> Duration is always listed as the same amount

I analyzed this and as it turns out the duration is updated every 10 seconds. This is due to the timeout in the wait method of the subprocess we run the fuzzer on that we use for checking the status of its execution. Upon success we also display 100 lines of the stdout log. All of the lines are prefixed with the string holding the duration value from the moment when the log method was called.
What is the desired output then? Do we want to decrease the timeout on the wait method to get higher resolution?

> Retries are not appearing in the log output

Hmm...they are now:
<pre>
2019-05-15T11:28:18.601417 - 010-clb-lutinit          -    0s: Running fuzzer attempt: 0
2019-05-15T11:28:18.601679 - 010-clb-lutinit          -    0s: Starting @ 2019-05-15T11:28:18.601672
2019-05-15T11:28:18.601713 - 010-clb-lutinit          -    0s: Running make -C /home/tmichalak/Development/X-Ray/prjxray/fuzzers/010-clb-lutinit run (with MAKEFLAGS='')
2019-05-15T11:28:18.626828 - 010-clb-lutinit          -    0s: Finishing (User:0s System:0s).
2019-05-15T11:28:18.632825 - 010-clb-lutinit          -    0s: Failed @ 2019-05-15T11:28:18.626921 with exit code: 2
</pre>

> Artefacts output is pretty useless

I don't know which artifacts you are referring to. Could you be more specific?

mithro, 2019-05-16:
> @mithro: This issue is the last one on ToDo list in the "Stabilize prjxray output" project (https://github.com/orgs/SymbiFlow/projects/5).

They are still open.

> I don't know how urgent this is on the priorities list, but if the points are still valid then I would like to address them.

These are mostly clean up & usability improvements around the build result access. I'm talking about the data you see on output on [Build status
 @ source.cloud.google.com/results/invocations page](https://source.cloud.google.com/results/invocations/43addc3d-62f3-4b3d-9f54-c80f34b64cc3/targets;collapsed=). You get to this page by "details" link next to the checks. See the image below;

[image]

> However, I need some clarification in order to do that:
> 
> > Remove logs from the target name in fuzzers/0xx-xxx-xxx/logs
> 
> What do you mean by _target name_? Could you be more specific what you have in mind here?

See the `/logs` bit circled in red in [the following build output](https://source.cloud.google.com/results/invocations/43addc3d-62f3-4b3d-9f54-c80f34b64cc3/targets;collapsed=)

[image]

> Currently the target log, e.g. `fuzzers/030-iob/logs/` has `stderr` and `stdout` logs with the entire execution. Maybe this is working fine now?
>
>
> I analyzed this and as it turns out the duration is [...]

See the stuff marked with **red dots** and **red lines** in [the following build output](https://source.cloud.google.com/results/invocations/43addc3d-62f3-4b3d-9f54-c80f34b64cc3/targets;collapsed=)

[image]

> 
> > Retries are not appearing in the log output
> 
> Hmm...they are now:
> 
> 2019-05-15T11:28:18.601417 - 010-clb-lutinit          -    0s: Running fuzzer attempt: 0
> 2019-05-15T11:28:18.601679 - 010-clb-lutinit          -    0s: Starting @ 2019-05-15T11:28:18.601672
> 2019-05-15T11:28:18.601713 - 010-clb-lutinit          -    0s: Running make -C /home/tmichalak/Development/X-Ray/prjxray/fuzzers/010-clb-lutinit run (with MAKEFLAGS='')
> 2019-05-15T11:28:18.626828 - 010-clb-lutinit          -    0s: Finishing (User:0s System:0s).
> 2019-05-15T11:28:18.632825 - 010-clb-lutinit          -    0s: Failed @ 2019-05-15T11:28:18.626921 with exit code: 2

> > Artefacts output is pretty useless
> 
> I don't know which artifacts you are referring to. Could you be more specific?

See the tabs circled in red found at [the following build output](https://source.cloud.google.com/results/invocations/43addc3d-62f3-4b3d-9f54-c80f34b64cc3/targets;collapsed=)

[image]

[image]

mithro, 2019-06-12:
@tmichalak What is the status of this?

tmichalak, 2019-06-12:
@mithro: This still needs to be done.