f4pga/prjxray #714: Make the pip fuzzers able to handle DRC check failures

mithro, 2019-03-12:
Occasionally we try and place stuff at bad locations, like the following;
```
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: + echo '`define SEED 32'\''h5998d28a'
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: + make -f /tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/top_generate.mk
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ++ export SEEDN=24
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ++ SEEDN=24
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: + echo '`define SEED 32'\''h9dca8ab5'
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: + make -f /tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/top_generate.mk
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ++ echo build/6/specimen_024
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ++ cut -c1-8
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ++ md5sum
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: + echo '`define SEED 32'\''h2af6530a'
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: + make -f /tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/top_generate.mk
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ERROR: [DRC PDIL-1] Invalid Site Configuration: Invalid configuration for site SLICE_X27Y117. Reason: Cannot replace net mynet_14 with net mynet_22 on bel pin A/A
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: , Cannot replace net mynet_14 with net mynet_22 on bel pin AUSED/0
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: , Cannot replace net mynet_14 with net mynet_22 on bel pin AUSED/OUT
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: , Site pin to site pin route-thru requires conflicting attribute string values for user logic element 'A6LUT in site 'SLICE_X27Y117'. Attribute 'EQN' is programmed to '2'h0' but needs a value of 'O6=A1' for the route-thru.
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: .
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ERROR: [DRC RTSTAT-5] Partial antennas: 1 net(s) have a partial antenna. The problem bus(es) and/or net(s) are mynet_14.
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ERROR: [DRC RTSTAT-6] Partial route conflicts: 2 net(s) have a partial conflict. The problem bus(es) and/or net(s) are mynet_14, and mynet_22.
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ERROR: [DRC RTSTAT-11] Invalid Site Programming: Invalid site programming for net mynet_14. The following site(s) are invalid: SLICE_X27Y117.
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: ERROR: [Vivado 12-1345] Error(s) found during DRC. Bitgen not run.
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: make[3]: *** [vivado.ok] Error 1
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: make[2]: *** [build/6/specimen_006/OK] Error 2
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: make[2]: *** Waiting for unfinished jobs....
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: make[1]: *** [run] Error 2
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s:
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: --------------------------------------------------------------------------
2019-03-12T10:59:34.925313 - 056-pip-rem              - 47m41s: Failed @ 2019-03-12T10:59:34.892651 with exit code: 2
2019-03-12T10:59:34.935122 - 056-pip-rem              - 47m41s: WARNING: Fuzzer failed!
Makefile:94: recipe for target '056-pip-rem/run.ok' failed
make: *** [056-pip-rem/run.ok] Error 2
```
It seems like the fuzzer should check for this type of failure and just ignore them and try again?

mithro, 2019-03-12:
@kgugala Had an idea about how he fixed a similar issue for another fuzzer?

tmichalak, 2019-03-18:
I am still working on reproducing the DRC error locally, so that I can analyze the design and see how we could avoid this situation. This is the approach @kgugala took in another fuzzer. 
In the meantime we could just lower the PDIL-1 and RTSTAT drc checks' severities to Warning.

litghost, 2019-03-18:
You can't disable the RTSTAT DRC check because its a routing conflict.

tmichalak, 2019-03-19:
You are right, for now I just disabled the PDIL.
The related PR that is testing the stabilization attempts is #728

closed 2019-05-15 by tmichalak

tmichalak, 2019-05-15:
All DRC related problems we have had in the fuzzers so far were an indicator of something unexpected going which required fixing. Hence adding a capability to the fuzzers that would handle such DRCs could mask potential problems.

tmichalak, 2019-05-15:
@mithro I closed this ticket because of the reason I described above, but feel free to reopen the issue if you think we still need some kind of DRC waiver mechanism.