f4pga/prjxray #746: unknown bit in zynq7 harness

mithro, 2019-03-27:
Using database https://github.com/mithro/prjxray-db/commit/88d9a13ebdf431ec9bd94377817daf4d46f7a938
```
Writing bitstream ./design.bit...
INFO: [Vivado 12-1842] Bitgen Completed Successfully.
INFO: [Project 1-118] WebTalk data collection is enabled (User setting is ON. Install Setting is ON.).
INFO: [Common 17-186] '/usr/local/google/home/tansell/github/SymbiFlow/prjxray/minitests/roi_harness/build/design/usage_statistics_webtalk.xml' has been successfully sent to Xilinx on Tue Mar 26 18:26:29 2019. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2017.2/doc/webtalk_introduction.html.
9 Infos, 5 Warnings, 0 Critical Warnings and 0 Errors encountered.
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:07 ; elapsed = 00:00:06 . Memory (MB): peak = 2350.082 ; gain = 332.453 ; free physical = 17416 ; free virtual = 39424
INFO: [Common 17-206] Exiting Vivado at Tue Mar 26 18:26:29 2019...
++ fgrep CRITICAL vivado.log
+ test -z ''
+ /usr/local/google/home/tansell/github/SymbiFlow/prjxray/build/tools/bitread --part_file /usr/local/google/home/tansell/github/SymbiFlow/prjxray/database/zynq7/xc7z010clg400-1.yaml -F 0x00000000:0xffffffff -o design.bits -z -y design.bit
Bitstream size: 2083839 bytes
Config size: 520922 words
Number of configuration frames: 5144
DONE
+ python3 /usr/local/google/home/tansell/github/SymbiFlow/prjxray/utils/bit2fasm.py --verbose design.bit
+ python3 /usr/local/google/home/tansell/github/SymbiFlow/prjxray/utils/fasm2frames.py design.fasm design.frm
+ PYTHONPATH=/usr/local/google/home/tansell/github/SymbiFlow/prjxray:/usr/local/google/home/tansell/github/SymbiFlow/prjxray/third_party/fasm:/usr/local/google/home/tansell/github/SymbiFlow/prjxray:/usr/local/google/home/tansell/github/SymbiFlow/prjxray/third_party/fasm:/usr/local/google/home/tansell/github/SymbiFlow/prjxray:/usr/local/google/home/tansell/github/SymbiFlow/prjxray/third_party/fasm:/usr/local/google/home/tansell/github/SymbiFlow/prjxray:/usr/local/google/home/tansell/github/SymbiFlow/prjxray/third_party/fasm:/usr/local/buildtools/current/sitecustomize:/usr/local/google/home/tansell/github/SymbiFlow/prjxray/utils
+ python3 ../create_design_json.py --design_info_txt design_info.txt --design_txt design.txt --pad_wires design_pad_wires.txt --design_fasm design.fasm
Traceback (most recent call last):
  File "../create_design_json.py", line 114, in <module>
    main()
  File "../create_design_json.py", line 92, in main
    unknown_base_address)
AssertionError: Found unknown bit in base address 0x00001380
Makefile:14: recipe for target 'build/design.bit' failed
make: *** [build/design.bit] Error 1
```

litghost, 2019-03-27:
@mithro Did you update the zync databases after https://github.com/SymbiFlow/prjxray/pull/727 was merged?

litghost, 2019-03-27:
```
{ unknown_bit = "0000139a_0_8", unknown_segment = "0x00001380", unknown_segbit = "26_08" }
{ unknown_bit = "0000139a_42_25", unknown_segment = "0x00001380", unknown_segbit = "26_1369" }
{ unknown_bit = "0000139a_42_28", unknown_segment = "0x00001380", unknown_segbit = "26_1372" }
{ unknown_bit = "0000139b_24_12", unknown_segment = "0x00001380", unknown_segbit = "27_780" }
{ unknown_bit = "0000139b_0_14", unknown_segment = "0x00001380", unknown_segbit = "27_14" }
{ unknown_bit = "00001b1b_49_25", unknown_segment = "0x00001b00", unknown_segbit = "27_1593" }
{ unknown_bit = "00001b1b_49_23", unknown_segment = "0x00001b00", unknown_segbit = "27_1591" }
{ unknown_bit = "00001b1b_50_25", unknown_segment = "0x00001b00", unknown_segbit = "27_1625" }
{ unknown_bit = "00001b1b_49_29", unknown_segment = "0x00001b00", unknown_segbit = "27_1597" }
{ unknown_bit = "00001b1b_49_31", unknown_segment = "0x00001b00", unknown_segbit = "27_1599" }
{ unknown_bit = "00001ba6_0_22", unknown_segment = "0x00001b80", unknown_segbit = "38_22" }
{ unknown_bit = "00001ba6_1_22", unknown_segment = "0x00001b80", unknown_segbit = "38_54" }
{ unknown_bit = "00001ba6_53_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1718" }
{ unknown_bit = "00001ba6_61_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1974" }
{ unknown_bit = "00001ba6_54_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1750" }
{ unknown_bit = "00001ba6_62_22", unknown_segment = "0x00001b80", unknown_segbit = "38_2006" }
{ unknown_bit = "00001ba7_53_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1719" }
{ unknown_bit = "00001ba7_61_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1975" }
{ unknown_bit = "00001ba7_54_17", unknown_segment = "0x00001b80", unknown_segbit = "39_1745" }
{ unknown_bit = "00001ba7_62_17", unknown_segment = "0x00001b80", unknown_segbit = "39_2001" }
{ unknown_bit = "00001ba7_0_21", unknown_segment = "0x00001b80", unknown_segbit = "39_21" }
{ unknown_bit = "00001ba7_0_23", unknown_segment = "0x00001b80", unknown_segbit = "39_23" }
{ unknown_bit = "00001ba7_54_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1751" }
{ unknown_bit = "00001ba7_62_23", unknown_segment = "0x00001b80", unknown_segbit = "39_2007" }
{ unknown_bit = "00001ba7_54_29", unknown_segment = "0x00001b80", unknown_segbit = "39_1757" }
{ unknown_bit = "00001ba7_62_29", unknown_segment = "0x00001b80", unknown_segbit = "39_2013" }
{ unknown_bit = "00001ba7_1_17", unknown_segment = "0x00001b80", unknown_segbit = "39_49" }
{ unknown_bit = "00001ba7_53_21", unknown_segment = "0x00001b80", unknown_segbit = "39_1717" }
{ unknown_bit = "00001ba7_61_21", unknown_segment = "0x00001b80", unknown_segbit = "39_1973" }
{ unknown_bit = "00001ba7_1_23", unknown_segment = "0x00001b80", unknown_segbit = "39_55" }
{ unknown_bit = "00001ba7_1_29", unknown_segment = "0x00001b80", unknown_segbit = "39_61" }
{ unknown_bit = "0040139d_73_0", unknown_segment = "0x00401380", unknown_segbit = "29_2336" }
{ unknown_bit = "00401ba8_89_30", unknown_segment = "0x00401b80", unknown_segbit = "40_2878" }
{ unknown_bit = "00401ba8_90_22", unknown_segment = "0x00401b80", unknown_segbit = "40_2902" }
{ unknown_bit = "00401ba8_89_22", unknown_segment = "0x00401b80", unknown_segbit = "40_2870" }
{ unknown_bit = "00401ba9_90_29", unknown_segment = "0x00401b80", unknown_segbit = "41_2909" }
{ unknown_bit = "00401ba9_89_23", unknown_segment = "0x00401b80", unknown_segbit = "41_2871" }
{ unknown_bit = "00401ba9_90_17", unknown_segment = "0x00401b80", unknown_segbit = "41_2897" }
{ unknown_bit = "00401ba9_89_21", unknown_segment = "0x00401b80", unknown_segbit = "41_2869" }
{ unknown_bit = "00401ba9_90_23", unknown_segment = "0x00401b80", unknown_segbit = "41_2903" }
{ unknown_bit = "00401ba9_89_29", unknown_segment = "0x00401b80", unknown_segbit = "41_2877" }
{ unknown_bit = "0040111d_68_29", unknown_segment = "0x00401100", unknown_segbit = "29_2205" }
```

Looks like a base address problem.

mithro, 2019-03-27:
@litghost Yes, I updated the database based on c7d7e9d7ade70bd47f969f9713d9a6e4defbd53a

tmichalak, 2019-03-28:
I did a small crosscheck of the missing bits and the current tilegrid.json:

<pre>
# In frame 0x0000139b 1 bits were not converted.
{ unknown_bit = "0000139b_24_12", unknown_segment = "0x00001380", unknown_segbit = "27_780" }
</pre>

Missing bit: CLK_BUFG_REBUF_X82Y65 27_12

<pre> 
# In frame 0x0040139d 1 bits were not converted.
{ unknown_bit = "0040139d_73_0", unknown_segment = "0x00401380", unknown_segbit = "29_2336" }
</pre>

Missing bit: CLK_BUFG_REBUF_X82Y38 29_0

<pre>
# In frame 0x00001ba6 6 bits were not converted.
{ unknown_bit = "00001ba6_0_22", unknown_segment = "0x00001b80", unknown_segbit = "38_22" }
{ unknown_bit = "00001ba6_1_22", unknown_segment = "0x00001b80", unknown_segbit = "38_54" }
</pre>

Missing bits:
RIOB33_SING_X31Y50 38_22
RIOB33_SING_X31Y50 38_54

<pre>
{ unknown_bit = "00001ba6_53_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1718" }
{ unknown_bit = "00001ba6_61_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1974" }
</pre>

Missing bits:
RIOB33_X31Y75 38_86
RIOB33_X31Y79 38_86

<pre>
{ unknown_bit = "00001ba6_54_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1750" }
{ unknown_bit = "00001ba6_62_22", unknown_segment = "0x00001b80", unknown_segbit = "38_2006" }
</pre>

Missing bits:
RIOB33_X31Y75 38_118
RIOB33_X31Y79 38_118

<pre>
# In frame 0x00001ba7 15 bits were not converted.
{ unknown_bit = "00001ba7_53_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1719" }
{ unknown_bit = "00001ba7_61_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1975" }
{ unknown_bit = "00001ba7_54_17", unknown_segment = "0x00001b80", unknown_segbit = "39_1745" }
{ unknown_bit = "00001ba7_62_17", unknown_segment = "0x00001b80", unknown_segbit = "39_2001" }
{ unknown_bit = "00001ba7_0_21", unknown_segment = "0x00001b80", unknown_segbit = "39_21" }
{ unknown_bit = "00001ba7_0_23", unknown_segment = "0x00001b80", unknown_segbit = "39_23" }
{ unknown_bit = "00001ba7_54_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1751" }
{ unknown_bit = "00001ba7_62_23", unknown_segment = "0x00001b80", unknown_segbit = "39_2007" }
{ unknown_bit = "00001ba7_54_29", unknown_segment = "0x00001b80", unknown_segbit = "39_1757" }
{ unknown_bit = "00001ba7_62_29", unknown_segment = "0x00001b80", unknown_segbit = "39_2013" }
{ unknown_bit = "00001ba7_1_17", unknown_segment = "0x00001b80", unknown_segbit = "39_49" }
{ unknown_bit = "00001ba7_53_21", unknown_segment = "0x00001b80", unknown_segbit = "39_1717" }
{ unknown_bit = "00001ba7_61_21", unknown_segment = "0x00001b80", unknown_segbit = "39_1973" }
{ unknown_bit = "00001ba7_1_23", unknown_segment = "0x00001b80", unknown_segbit = "39_55" }
{ unknown_bit = "00001ba7_1_29", unknown_segment = "0x00001b80", unknown_segbit = "39_61" }
</pre> 

Other RIOB tile related bits are missing

<pre>
# In frame 0x00401ba8 3 bits were not converted.
{ unknown_bit = "00401ba8_89_30", unknown_segment = "0x00401b80", unknown_segbit = "40_2878" }
{ unknown_bit = "00401ba8_90_22", unknown_segment = "0x00401b80", unknown_segbit = "40_2902" }
{ unknown_bit = "00401ba8_89_22", unknown_segment = "0x00401b80", unknown_segbit = "40_2870" }
</pre>

RIOB tiles in the southern half

<pre>
# In frame 0x0040111d 1 bits were not converted.
{ unknown_bit = "0040111d_68_29", unknown_segment = "0x00401100", unknown_segbit = "29_2205" }
</pre>

Couldn't find any tile in tilegrid.json that would have more than 28 frames with this baseaddress.

<pre>
# In frame 0x00401ba9 6 bits were not converted.
{ unknown_bit = "00401ba9_90_29", unknown_segment = "0x00401b80", unknown_segbit = "41_2909" }
{ unknown_bit = "00401ba9_89_23", unknown_segment = "0x00401b80", unknown_segbit = "41_2871" }
{ unknown_bit = "00401ba9_90_17", unknown_segment = "0x00401b80", unknown_segbit = "41_2897" }
{ unknown_bit = "00401ba9_89_21", unknown_segment = "0x00401b80", unknown_segbit = "41_2869" }
{ unknown_bit = "00401ba9_90_23", unknown_segment = "0x00401b80", unknown_segbit = "41_2903" }
{ unknown_bit = "00401ba9_89_29", unknown_segment = "0x00401b80", unknown_segbit = "41_2877" }
</pre>

Again RIOB tiles in the southern half

Summarizing:
We are missing few bits for CLK_BUFG_REBUF, RIOB33 and RIOB33_SINGLE.
There is also one bit with minor frame address above the frame count for any known segment in tilegrid.json:
<pre>
# In frame 0x0040111d 1 bits were not converted.
{ unknown_bit = "0040111d_68_29", unknown_segment = "0x00401100", unknown_segbit = "29_2205" 
</pre>

tmichalak, 2019-03-28:
I guess that for the RIOBs we could modify the harness design to use the features we already have solved the bits for.

litghost, 2019-03-28:
@tmichalak I would start by figuring out that bit with no frame address, as that likely needs to be solved first.  The maybe figure out what is going on in the clock column.  I'll warn you that it may look like a CLK_BUFG_REBUF, but my confidence in the address/word ranges for the clock column is not the highest.

litghost, 2019-03-28:
Also note that bits in the IOB may also belong to the IOI, as we are currently have merged those sections for expedience.

tmichalak, 2019-03-28:
IOI? Could you say a bit more about it?

litghost, 2019-03-28:
The tile next to IOB, containing the ILOGIC and OLOGIC sites.

tmichalak, 2019-03-28:
Yepp, I suspected you might have meant those. Wanted to be sure.

tmichalak, 2019-04-04:
So for the mystical unknown bit:
<pre>
# In frame 0x0040111d 1 bits were not converted.
{ unknown_bit = "0040111d_68_29", unknown_segment = "0x00401100", unknown_segbit = "29_2205" 
</pre>
It turns out when the BITSTREAM.GENERAL.PERFRAMECRC property is set to YES becomes:
<pre>
# In frame 0x0040111b 1 bits were not converted. 
{ unknown_bit = "0040111b_68_29", unknown_segment = "0x00401100", unknown_segbit = "27_2205" }
</pre>
And this bit can be found in each and every single bitstream in our zynq fuzzers.
This address falls into INT_L region, but there are no bits for it because I guess it's always on.

tmichalak, 2019-04-15:
<pre>
{ unknown_bit = "00001ba6_0_22", unknown_segment = "0x00001b80", unknown_segbit = "38_22" }
{ unknown_bit = "00001ba6_1_22", unknown_segment = "0x00001b80", unknown_segbit = "38_54" }
{ unknown_bit = "00001ba6_53_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1718" }
{ unknown_bit = "00001ba6_61_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1974" }
{ unknown_bit = "00001ba6_54_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1750" }
{ unknown_bit = "00001ba6_62_22", unknown_segment = "0x00001b80", unknown_segbit = "38_2006" }
{ unknown_bit = "00001ba7_53_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1719" }
{ unknown_bit = "00001ba7_61_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1975" }
{ unknown_bit = "00001ba7_54_17", unknown_segment = "0x00001b80", unknown_segbit = "39_1745" }
{ unknown_bit = "00001ba7_62_17", unknown_segment = "0x00001b80", unknown_segbit = "39_2001" }
{ unknown_bit = "00001ba7_0_21", unknown_segment = "0x00001b80", unknown_segbit = "39_21" }
{ unknown_bit = "00001ba7_0_23", unknown_segment = "0x00001b80", unknown_segbit = "39_23" }
{ unknown_bit = "00001ba7_54_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1751" }
{ unknown_bit = "00001ba7_62_23", unknown_segment = "0x00001b80", unknown_segbit = "39_2007" }
{ unknown_bit = "00001ba7_54_29", unknown_segment = "0x00001b80", unknown_segbit = "39_1757" }
{ unknown_bit = "00001ba7_62_29", unknown_segment = "0x00001b80", unknown_segbit = "39_2013" }
{ unknown_bit = "00001ba7_1_17", unknown_segment = "0x00001b80", unknown_segbit = "39_49" }
{ unknown_bit = "00001ba7_53_21", unknown_segment = "0x00001b80", unknown_segbit = "39_1717" }
{ unknown_bit = "00001ba7_61_21", unknown_segment = "0x00001b80", unknown_segbit = "39_1973" }
{ unknown_bit = "00001ba7_1_23", unknown_segment = "0x00001b80", unknown_segbit = "39_55" }
{ unknown_bit = "00001ba7_1_29", unknown_segment = "0x00001b80", unknown_segbit = "39_61" }
</pre>
These bits have been solved for artix:
<pre>
RIOB33.IOB_Y0.IN_ONLY 38_118 39_119 39_125
RIOB33.IOB_Y0.LVCMOS12_LVCMOS15_LVCMOS18.IN 38_86 39_85 !39_87 39_113
RIOB33.IOB_Y0.LVCMOS25_LVCMOS33_LVTTL.IN 38_86 39_85 39_87 39_113
</pre>
What remains is analyzing why they are not computed for zynq.

tmichalak, 2019-04-16:
So all of the unknown bits are known for artix except the one related to the   BITSTREAM.GENERAL.PERFRAMECRC setting:
<pre>
# In frame 0x0040111d 1 bits were not converted.
{ unknown_bit = "0040111d_68_29", unknown_segment = "0x00401100", unknown_segbit = "29_2205" 
</pre>
This bit seems to be Zynq specific, as I didn't notice it in Artix on Kintex.

To summarize the rest of the unknown bits I had with the latest roi_harness run for zynq7-swbut:
<pre>
{ unknown_bit = "00001ba6_0_22", unknown_segment = "0x00001b80", unknown_segbit = "38_22" }
{ unknown_bit = "00001ba6_1_22", unknown_segment = "0x00001b80", unknown_segbit = "38_54" }
{ unknown_bit = "00001ba6_53_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1718" }
{ unknown_bit = "00001ba6_61_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1974" }
{ unknown_bit = "00001ba6_54_22", unknown_segment = "0x00001b80", unknown_segbit = "38_1750" }
{ unknown_bit = "00001ba6_62_22", unknown_segment = "0x00001b80", unknown_segbit = "38_2006" }
{ unknown_bit = "00001ba7_53_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1719" }
{ unknown_bit = "00001ba7_61_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1975" }
{ unknown_bit = "00001ba7_54_17", unknown_segment = "0x00001b80", unknown_segbit = "39_1745" }
{ unknown_bit = "00001ba7_62_17", unknown_segment = "0x00001b80", unknown_segbit = "39_2001" }
{ unknown_bit = "00001ba7_0_21", unknown_segment = "0x00001b80", unknown_segbit = "39_21" }
{ unknown_bit = "00001ba7_0_23", unknown_segment = "0x00001b80", unknown_segbit = "39_23" }
{ unknown_bit = "00001ba7_54_23", unknown_segment = "0x00001b80", unknown_segbit = "39_1751" }
{ unknown_bit = "00001ba7_62_23", unknown_segment = "0x00001b80", unknown_segbit = "39_2007" }
{ unknown_bit = "00001ba7_54_29", unknown_segment = "0x00001b80", unknown_segbit = "39_1757" }
{ unknown_bit = "00001ba7_62_29", unknown_segment = "0x00001b80", unknown_segbit = "39_2013" }
{ unknown_bit = "00001ba7_1_17", unknown_segment = "0x00001b80", unknown_segbit = "39_49" }
{ unknown_bit = "00001ba7_53_21", unknown_segment = "0x00001b80", unknown_segbit = "39_1717" }
{ unknown_bit = "00001ba7_61_21", unknown_segment = "0x00001b80", unknown_segbit = "39_1973" }
{ unknown_bit = "00001ba7_1_23", unknown_segment = "0x00001b80", unknown_segbit = "39_55" }
{ unknown_bit = "00001ba7_1_29", unknown_segment = "0x00001b80", unknown_segbit = "39_61" }
</pre>

Solved for Artix:
<pre>
RIOB33.IOB_Y0.IN_ONLY 38_118 39_119 39_125
RIOB33.IOB_Y0.LVCMOS25_LVCMOS33_LVTTL.IN 38_86 39_85 39_87 39_113
</pre>

<pre>
 In frame 0x0000139a 2 bits were not converted.
{ unknown_bit = "0000139a_47_12", unknown_segment = "0x00001380", unknown_segbit = "26_1516" }
{ unknown_bit = "0000139a_50_23", unknown_segment = "0x00001380", unknown_segbit = "26_1623" }
{ unknown_bit = "0000139b_47_10", unknown_segment = "0x00001380", unknown_segbit = "27_1514" }
{ unknown_bit = "0000139c_50_18", unknown_segment = "0x00001380", unknown_segbit = "28_1618" }
</pre>

Solved for Artix:
<pre>
CLK_HROW_TOP_R.CLK_HROW_CK_MUX_OUT_R0.CLK_HROW_R_CK_GCLK16 26_172  27_170
CLK_HROW_TOP_R.CLK_HROW_R_CK_GCLK16_ACTIVE 26_279
CLK_HROW_TOP_R.CLK_HROW_CK_IN_R0_ACTIVE 28_274
</pre>

<pre>
# In frame 0x0000139b 2 bits were not converted.
{ unknown_bit = "0000139b_24_12", unknown_segment = "0x00001380", unknown_segbit = "27_780" }
</pre>

Solved for Artix:
<pre>
CLK_BUFG_REBUF.CLK_BUFG_REBUF_R_CK_GCLK16_TOP.CLK_BUFG_REBUF_R_CK_GCLK16_BOT 27_12
</pre>

I need to investigate why 030-iob, 041-clk-hrow-pips and 043-clk-rebuf-pips don't solve the bits for zynq while solving it for artix.

tmichalak, 2019-05-07:
The unknown bits calculated with 030-iob(#820) will be solved after #823 is merged.
We are still missing few BUFG_REBUF PIPs calcualted in ``043-clk-rebuf-pips``.
To be precise these four PIPs are calculated for Artix, but are not for Zynq:
<pre>
CLK_BUFG_REBUF.CLK_BUFG_REBUF_R_CK_GCLK0_TOP.CLK_BUFG_REBUF_R_CK_GCLK0_BOT 27_13
CLK_BUFG_REBUF.CLK_BUFG_REBUF_R_CK_GCLK16_TOP.CLK_BUFG_REBUF_R_CK_GCLK16_BOT 27_12
CLK_BUFG_REBUF.CLK_BUFG_REBUF_R_CK_GCLK19_TOP.CLK_BUFG_REBUF_R_CK_GCLK19_BOT 27_60
CLK_BUFG_REBUF.CLK_BUFG_REBUF_R_CK_GCLK21_BOT.CLK_BUFG_REBUF_R_CK_GCLK21_TOP 27_94
</pre>

Once these are addressed we will have to fix  ``041-clk-hrow-pips`` to get the few missing bits and then we should be good.

litghost, 2019-05-24:
> All bits, except some Zynq specific which are enabled in all Zynq's fuzzer bitstreams, are known.
> The always on bits are:
> 
> # In frame 0x0040111b 1 bits were not converted.
> { unknown_bit = "0040111b_68_29", unknown_segment = "0x00401100", unknown_segbit = "27_2205" }
> # In frame 0x0040111a 2 bits were not converted.
> { unknown_bit = "0040111a_68_30", unknown_segment = "0x00401100", unknown_segbit = "26_2206" }
> { unknown_bit = "0040111a_68_31", unknown_segment = "0x00401100", unknown_segbit = "26_2207" }

What tiles do you believe these bits belong too?

tmichalak, 2019-05-24:
Hard to tell. These bits were present in every single specimen generated for Zynq7. And one of them depended on the PERFRAMECRC setting as I described here: https://github.com/SymbiFlow/prjxray/issues/746#issuecomment-483615681

litghost, 2019-05-24:
If one of the bits is correlated with ``PERFRAMECRC``, simply write a fuzzer to document that?  There are several CMT tiles that are good candidates for fabric wide configuration, if I had to guess it lives there.

tmichalak, 2019-08-21:
I have created a minitest (https://github.com/antmicro/prjxray/tree/bitstream_properties) which shows that the missing bits are present in an almost empty bitstream.
Almost empty, because we need at least a LUT to generate a bitstream, but constraining the LUT to various places didn't affect the presence of the three bits.
My theory is that the bits belong to bitstream properties. For example when ``BITSTREAM.CONFIG.USERID`` which is by default set to 0xFFFFFFFF is changed to 0x0 we get the following bits:
<pre>
0040111a_068_00
0040111a_068_01
0040111a_068_02
0040111a_068_03
0040111a_068_04
0040111a_068_05
0040111a_068_06
0040111a_068_07
0040111a_068_08
0040111a_068_09
0040111a_068_10
0040111a_068_11
0040111a_068_12
0040111a_068_13
0040111a_068_14
0040111a_068_15
0040111a_068_30 << missing bit 1
0040111a_068_31 << missing bit 2
0040111b_068_00
0040111b_068_01
0040111b_068_02
0040111b_068_03
0040111b_068_04
0040111b_068_05
0040111b_068_06
0040111b_068_07
0040111b_068_08
0040111b_068_09
0040111b_068_10
0040111b_068_11
0040111b_068_12
0040111b_068_13
0040111b_068_14
0040111b_068_15
0040111b_068_29 << missing bit 3
</pre>

The pattern here is easy to spot - LSB in USERID set to 0 results in ``0040111a_068_00``, LSB+1 set to 0 results in ``0040111b_068_00``.
Unfortunately, USERID is 32bit so we only get to bit 15 in word 68 for a given frame.
I tried out all other bitstream properties and saw bits changing in the same frame address, but not where we need it. 
I compared it with Zynq 7020 to see if those bits are part related and as it turns out apart from the frame address the word and bits remained the same.
This must be something property related or part specific...

litghost, 2019-08-21:
@tmichalak Change your minitest to have no inputs or outputs or clocks.  I believe that toggles 2 of the unknown bits.

tmichalak, 2019-08-22:
@litghost: Having the pins or not doesn't influence the presence of these bits. Again, I don't think they rely on the design, but on the bitstream properties. 
I reran all fuzzers for zynq trying to find a case in which these 3 bits wouldn't show up in the bitstream and guess what - none. All bits file that were generated included the 3 bits.
I updated the minitest to be able to generate a bitstream with random properties and had the same result. Worth noting though is that with the following bitstream setting: ``set_property BITSTREAM.GENERAL.DISABLE_JTAG YES [current_design]`` the bitstream had bits very close to the ones we are looking for, namely:
<pre>
0040111a_068_26 << DISABLE_JTAG
0040111a_068_30 << missing
0040111a_068_31 << missing
0040111b_068_27 << DISABLE_JTAG
0040111b_068_28 << DISABLE_JTAG
0040111b_068_29 << missing
</pre>

Based on these finding I decided to start creating a fuzzer, the initial commit is on the branch.
The only difference in fuzzing I see compared to all other is that here we don't have an association to any tile as these features are bitstream related.

tmichalak, 2019-09-02:
@litghost 
I think that we could assume that the missing bits are in fact always on.
I experimented by artificially injecting bitstream config bits.
In order to not break the existing flow I added an artificial tile_type called bitstream_config:
`tile_type_BITSTREAM_CONFIG.json`:
<pre>
{
    "tile_type": "BITSTREAM_CONFIG"
}
</pre>
 and placed it on grid coordinates that do not exist on the fabric:
<pre>
    "BITSTREAM": {
        "bits": {
            "CLB_IO_CLK": {
                "baseaddr": "0x00401100",
                "frames": 28,
                "offset": 68,
                "words": 1
            }
        },
        "grid_x": 128,
        "grid_y": 0,
        "sites": {
        },
        "type": "BITSTREAM_CONFIG"
    }
</pre>
The tilegrid fuzzer will be responsible for calculating the base address and adding this to the tilegrid with the fake grid coordinates.

The next step will be to add the bitstream_config fuzzer that will inject the fake always on bits:
<pre>
BITSTREAM_CONFIG.ALWAYS_ON_1 26_30
BITSTREAM_CONFIG.ALWAYS_ON_2 26_31
BITSTREAM_CONFIG.ALWAYS_ON_3 27_29
</pre>
The initial version of the fuzzer is already on a branch: https://github.com/antmicro/prjxray/commits/bitstream_properties
However, it will have to be adjusted to the new fake tile type.

The bitstream produced from the fasm file with (as well as without) the bits asserted using fasm2frames and xcframes2bit works fine, at least for the counter test.

However I am not sure if the tile outside of the grid will not explode somewhere in symbiflow-arch-defs. That is something that I want to see next.

litghost, 2019-09-05:
> In order to not break the existing flow I added an artificial tile_type called bitstream_config:
> `tile_type_BITSTREAM_CONFIG.json`:

Why not just use `CFG_CENTER_MID` over a purely synthetic tile?  As far as I can tell `CFG_CENTER_MID` is the same.

litghost, 2019-09-05:
Like I said elsewhere, at least one of the bits is not always constant:

```
{ unknown_bit = "0040111b_68_29", unknown_segment = "0x00401100", unknown_segbit = "27_2205" }`
```

Appears to toggle.   I've attached an example verilog/dcp/bit/fasm example.
[toggle_ex.tgz.zip](https://github.com/SymbiFlow/prjxray/files/3579904/toggle_ex.tgz.zip)

tmichalak, 2019-09-06:
@litghost 
Hmm...I was always looking for the bits in the ``bits`` files. And in the archive you sent I also see these three bites across all of the ``bits`` files. However it's not always the case for fasm.
How come we have discrepancies between the bits file and their corresponding fasm?
Which ones should be trusted?

litghost, 2019-09-06:
> @litghost
> Hmm...I was always looking for the bits in the `bits` files. And in the archive you sent I also see these three bites across all of the `bits` files. However it's not always the case for [...]

Interesting observation.  `bits` is trustworthy, but the absence of `unknown` from the  `fasm` is a signal too.  It may be a bug in `bit2fasm`, but I suspect not.  Instead, the question becomes which FASM feature matched that bit?  It may provide a hint as to what is going on.  If it is in fact a bug in `bit2fasm`, we should fix it.

tmichalak, 2019-09-09:
@litghost @mithro I analyzed the problem and it turned out to be a problem in fasm (https://github.com/SymbiFlow/fasm/pull/15). I am not saying this fix is very clean, but it proves that the bits are in fact constant.
I will now focus on accommodating for that in the prjxray flow. I will go with using the ``CFG_CENTER_MID`` rather than the synthetic tile.

litghost, 2019-09-09:
> I analyzed the problem and it turned out to be a problem in fasm ([SymbiFlow/fasm#15](https://github.com/SymbiFlow/fasm/pull/15)).

Perfect, thanks for finding that.

tmichalak, 2019-09-20:
The remaining unknown bits are always set to on. In order for the harness to work those bits have been added to the database with the following PR #1046.

closed 2019-09-20 by tmichalak