f4pga/prjxray #753: Implement python sdf parser

kgugala, 2019-04-03:
We need this to parse timing files and include delay info in arch.xml

elmsfu, 2019-04-03:
@mithro Can you give details on why you thought a python parser is needed?
Also why not wrap or add python to an existing opensource parser?

mithro, 2019-04-03:
SDF is simple enough and small enough a pure Python library should be easy to write compared to trying to wrap the existing C++ implementation.

Pure python dependencies are better than dependencies which require compilation.

mithro, 2019-04-03:
The Python library is needed to architecture definitions to read the SDF files and pull the timing info into the XML files needed by VPR.

kgugala, 2019-04-03:
this parser should be easy to implement. With https://pypi.org/project/ply/ I can reuse the whole lexer and grammar definition from libsdfparse. In fact, I already have an initial version which defines all the required tokens and can handle empty SDF (header only) file. I just need to move the rest of the grammar definition. This should be very complicated.

mithro, 2019-04-03:
> I just need to move the rest of the grammar definition. This should be very complicated.

Hopefully you mean "This should **not** be very complicated." :-P

kgugala, 2019-04-03:
Yeah, it should not ;)

closed 2019-04-10 by kgugala