f4pga/prjxray #768: BRAM patching has an offset of 2

litghost, 2019-04-06:
Using xc7patch to write to frame address 0x00820000 to will result in data getting written to 0x00820002.  I haven't pinned down if this is a regression, or it has always been like this.  @tmichalak you were in here last, any ideas?

litghost, 2019-04-06:
This error appears in 2aa3c309762a1ac2e3c5989e6419f4825a24ac0e, before the https://github.com/SymbiFlow/prjxray/pull/705 .  This might be a long standing bug.

litghost, 2019-04-06:
Replication steps:
1. ``build/tools/xc7patch --part_file prjxray-db/artix7/xc7a35tcpg236-1.yaml  --bitstream_file prjxray-db/artix7/harness/basys3/swbut/design.bit --output_file test.bit --frm_file top.frames``

Where top.frames is just:
```
0x00820000 0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x11190000,0x07DF8AB3,0x04351098,0x09EE160E,0x09BE05A5,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000,0x00000000
```

2. ``build/tools/bitread  -o test.bits --part_file ../symbiflow-arch-defs/third_party/prjxray-db/artix7/xc7a35tcpg236-1.yaml test.bit``

See that data is written to the 0x00820002 frame not the 0x00820000 frame.

tmichalak, 2019-04-08:
@litghost : Thanks for this issue, thanks to it I found a bug in the xc7frames2bit tool that I created a PR for (#769).
Now, regarding the bug you found, after a short analysis I found what is causing the 2 frame shift, namely it's the empty frames in the Type2 configuration package creation: https://github.com/SymbiFlow/prjxray/blob/master/lib/xilinx/xc7series/utils.cc#L35
However, in my opinion it's not a bug in the xc7patch as it is more a problem of bitread

closed 2019-04-11 by tmichalak