f4pga/prjxray #774: Figure out why prjxray database builds have gone up to 4h

mithro, 2019-04-10:
The database builds seem to be taking a long time at the moment?
```
Start Time: Apr 5 2019, 4:52:50 PM UTC-7
Duration: 3 hr 58 min
```

The last quick build was on commit a7066bb246ab8c11d3c6313ac25317f2b46d17aa on Mar 21st at 18:41 (timezone unknown). It took less than 2.4 hours -- all runs after it take 4+ hours with a further jump to 6+ hours becoming common after commit commit a792c51cce761aff2e63ccc82e01e870d16228fc

mithro, 2019-04-10:
FYI `fuzzers/057-pip-bi` continues to be the cause of most failures (~50% of the failures). `fuzzers/052-pip-clkin` is also causing a lot of failures (~25%).

mithro, 2019-04-10:
I'm guessing https://github.com/SymbiFlow/prjxray/issues/661 is related.

mithro, 2019-06-12:
@tmichalak What is the status here?

tmichalak, 2019-06-13:
@mithro Currently our longest running fuzzer is 074 and the overall runtime is dictated by its runtime.
So the observations from the previous post in this issue no longer apply. 
However, we still have to restore the harness for Artix and Zynq which will add up some time.
Once we have 041 back on the running fuzzers list and the harnesses reinstated I will be able to close or update this issue.

closed 2019-07-30 by litghost