f4pga/prjxray #776: add tool for merging sdf files

kgugala, 2019-04-12:
the data generated by https://github.com/SymbiFlow/prjxray/pull/706 is divided by tiles.

It looks like sites within those tiles are identical - e.g all the slicel's BELs have always the same timing no matter if they are in CLBLL_[LR] or CLBLM_[LR]. We need a script which could merge those and produce one single timing file (e.g. slicel). This will ease the import to arch.xml in arch-defs

mithro, 2019-04-12:
@kgugala Does this make sense as part of https://github.com/SymbiFlow/python-sdf-timing?

kgugala, 2019-04-17:
I thought about something arch specific which will get different SDF's, check if all timing for a selected site are identical and produce sdf containing only relevant info. I started this here https://github.com/SymbiFlow/prjxray/blob/fd66eb7ccc25f007591f048f3006a1e8f29c2332/utils/sdfmerge.py 

I need to refactor sdf writer. The writer should be merged with python-sdf-timing lib

mithro, 2019-04-17:
SGTM!

mithro, 2019-06-12:
@kgugala What is the status here?