f4pga/prjxray #777: Terminate the looping fuzzers when TODO list stops changing

mithro, 2019-04-12:
Currently the looping fuzzers like the PIP fuzzers terminate when either;
 - [ ] There is nothing left in the TODO list.
 - [ ] There have been too many iterations (normally like >50).

Instead, it makes more sense to terminate the fuzzer when the TODO list is no longer getting smaller. If the list stays the same for >4 iterations it is probably never going to finish.

mithro, 2019-04-12:
The looping fuzzer should also dump the TODO list when the fuzzer fails to give us a better idea of which entries are causing issues.

mithro, 2019-04-12:
It seems like the int fuzzers already behave this way? See the following files;
 * https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/int_loop.sh
 * https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/int_loop_check.py
 * https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/int_loop.mk
 * https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/int_maketodo.py

mithro, 2019-04-12:
The PIP fuzzers do not?
 * https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/pip_list.mk
 * https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/pip_loop.mk

litghost, 2019-04-12:
> Currently the looping fuzzers like the PIP fuzzers terminate when either;
>
> * [ ]  There is nothing left in the TODO list.
> * [ ]  There have been too many iterations (normally like >50).
>
> Instead, it [...]

This is incorrect.  The default behavior of having 0 entries is the correct behavior for some fuzzers.  It is possible to enable stop at a "stable" point, but some fuzzers need to reach 0 entries for correctness.

litghost, 2019-04-12:
> The looping fuzzer should also dump the TODO list when the fuzzer fails to give us a better idea of which entries are causing issues.

It already does this in the build/todo folder?

mithro, 2019-04-12:
> This is incorrect. The default behavior of having 0 entries is the correct behavior for some fuzzers. It is possible to enable stop at a "stable" point, but some fuzzers need to reach 0 entries for correctness.

@litghost - I was suggesting terminating with a **failure** if the todo list stops getting smaller. Looking at the current failures on CI, the TODO list gets stuck at around 5-10 runs and then it runs for another ~40 before actually failing.

mithro, 2019-04-12:
> It already does this in the build/todo folder?

I was suggesting echoing it out to stdout on failure.

litghost, 2019-04-12:
> It seems like the int fuzzers already behave this way? See the following files;

Only ones that specify "--min_progress", however "--min_progress" is a success termination condition

> The PIP fuzzers do not?

All fuzzers (INT and PIP) use the same underlying scripts (maketodo and loop_check).  Fuzzers that specify --zero-entries will only terminate if 0 or timeout.  Fuzzers (like 050) that specific "--min_progress" will terminate once that level out.  Something akin to "--min_progress_failure" + "--min_progress_timeout" would satisfy the original idea.

litghost, 2019-04-12:
> > It already does this in the build/todo folder?
> 
> I was suggesting echoing it out to stdout on failure.

Probably a better plan to also capture the build/todo folder if we aren't already.

mithro, 2019-06-12:
@tmichalak Can you take a look at this and tell me if it is already done?

tmichalak, 2019-06-12:
@mithro: From what I see, not much changed in this matter, i.e. we only terminate with a termination in case of a timeout. --min-progress violation between two consecutive runs terminates the fuzzer, but not with a failure. So this still has to be implemented.