f4pga/prjxray #783: Fuzzer 052-pip-clkin fails a lot in CI

kgugala, 2019-04-16:
There is a lot of routing errors in the 052 fuzzer e.g:

```
052-pip-clkin- 30m53s: Rolling back route
052-pip-clkin- 30m53s: Command: route_design -unroute -nets mynet_8
052-pip-clkin- 30m53s: Attempting to get a license for feature 'Implementation' and/or device 'xc7k70t-fbg676'
052-pip-clkin- 30m53s: INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7k70t-fbg676'
052-pip-clkin- 30m53s: Command: report_drc (run_mandatory_drcs) for: incr_eco_checks
052-pip-clkin- 30m53s: INFO: [DRC 23-27] Running DRC with 8 threads
052-pip-clkin- 30m53s: report_drc (run_mandatory_drcs) completed successfully
052-pip-clkin- 30m53s: INFO: [Vivado_Tcl 4-198] DRC finished with 0 Errors
052-pip-clkin- 30m53s: INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
052-pip-clkin- 30m53s: INFO: [Route 35-333] 1 net was fully unrouted.
052-pip-clkin- 30m53s: 18 Infos, 0 Warnings, 0 Critical Warnings and 0 Errors encountered.
052-pip-clkin- 30m53s: route_design completed successfully
052-pip-clkin- 30m53s: WARNING: failed to route net after 3 tries
052-pip-clkin- 30m53s: wrong # args: should be "error message ?errorInfo? ?errorCode?"
052-pip-clkin- 30m53s:     while executing
052-pip-clkin- 30m53s: "error"
052-pip-clkin- 30m53s:     invoked from within
052-pip-clkin- 30m53s: "if {$tries >= 3} {
052-pip-clkin- 30m53s:             puts "WARNING: failed to route net after $tries tries"
052-pip-clkin- 30m53s:             error
052-pip-clkin- 30m53s:         }"    
052-pip-clkin- 30m53s:     ("while" body line 38)
052-pip-clkin- 30m53s:     invoked from within
052-pip-clkin- 30m53s: "while {1} {
052-pip-clkin- 30m53s:         set tile_idx [expr $tries + [expr $idx * 3]]
052-pip-clkin- 30m53s:         incr tries
052-pip-clkin- 30m53s: 
052-pip-clkin- 30m53s:         if {$tile_type == "INT_L"} {set tile [lindex $int_l_tiles..."
052-pip-clkin- 30m53s:     ("for" body line 16)
052-pip-clkin- 30m53s:     invoked from within
052-pip-clkin- 30m53s: "for {set idx 0} {$idx < [llength $todo_lines]} {incr idx} {^
052-pip-clkin- 30m53s:     set line [lindex $todo_lines $idx]
052-pip-clkin- 30m53s:     puts "== $idx: $line"
052-pip-clkin- 30m53s: 
052-pip-clkin- 30m53s:     set tile_type [linde..."
052-pip-clkin- 30m53s:     (file "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-kintex7/fuzzers/052-pip-clkin/generate.tcl" line 38)
```052-pip-clkin- 30m53s: Rolling back route
052-pip-clkin- 30m53s: Command: route_design -unroute -nets mynet_8
052-pip-clkin- 30m53s: Attempting to get a license for feature 'Implementation' and/or device 'xc7k70t-fbg676'
052-pip-clkin- 30m53s: INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xc7k70t-fbg676'
052-pip-clkin- 30m53s: Command: report_drc (run_mandatory_drcs) for: incr_eco_checks
052-pip-clkin- 30m53s: INFO: [DRC 23-27] Running DRC with 8 threads
052-pip-clkin- 30m53s: report_drc (run_mandatory_drcs) completed successfully
052-pip-clkin- 30m53s: INFO: [Vivado_Tcl 4-198] DRC finished with 0 Errors
052-pip-clkin- 30m53s: INFO: [Vivado_Tcl 4-199] Please refer to the DRC report (report_drc) for more information.
052-pip-clkin- 30m53s: INFO: [Route 35-333] 1 net was fully unrouted.
052-pip-clkin- 30m53s: 18 Infos, 0 Warnings, 0 Critical Warnings and 0 Errors encountered.
052-pip-clkin- 30m53s: route_design completed successfully
052-pip-clkin- 30m53s: WARNING: failed to route net after 3 tries
052-pip-clkin- 30m53s: wrong # args: should be "error message ?errorInfo? ?errorCode?"
052-pip-clkin- 30m53s:     while executing
052-pip-clkin- 30m53s: "error"
052-pip-clkin- 30m53s:     invoked from within
052-pip-clkin- 30m53s: "if {$tries >= 3} {
052-pip-clkin- 30m53s:             puts "WARNING: failed to route net after $tries tries"
052-pip-clkin- 30m53s:             error
052-pip-clkin- 30m53s:         }"    
052-pip-clkin- 30m53s:     ("while" body line 38)
052-pip-clkin- 30m53s:     invoked from within
052-pip-clkin- 30m53s: "while {1} {
052-pip-clkin- 30m53s:         set tile_idx [expr $tries + [expr $idx * 3]]
052-pip-clkin- 30m53s:         incr tries
052-pip-clkin- 30m53s: 
052-pip-clkin- 30m53s:         if {$tile_type == "INT_L"} {set tile [lindex $int_l_tiles..."
052-pip-clkin- 30m53s:     ("for" body line 16)
052-pip-clkin- 30m53s:     invoked from within
052-pip-clkin- 30m53s: "for {set idx 0} {$idx < [llength $todo_lines]} {incr idx} {^
052-pip-clkin- 30m53s:     set line [lindex $todo_lines $idx]
052-pip-clkin- 30m53s:     puts "== $idx: $line"
052-pip-clkin- 30m53s: 
052-pip-clkin- 30m53s:     set tile_type [linde..."
052-pip-clkin- 30m53s:     (file "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-kintex7/fuzzers/052-pip-clkin/generate.tcl" line 38)
```

this significantly impacts CI build time (see #774)

tmichalak, 2019-04-17:
One example for which we get this error is for PIP ``INT_L.CLK_L0.GCLK_L_B11_WEST`` and the following routing path:
<pre>
set route [find_routing_path -quiet -from CLBLM_R_X11Y74/CLBLM_L_A -to INT_L_X10Y75/GCLK_L_B11_WEST]
  CLBLM_R_X11Y74/CLBLM_L_A -> INT_L_X10Y75/GCLK_L_B11_WEST: CLBLM_R_X11Y74/CLBLM_L_A CLBLM_R_X11Y74/CLBLM_LOGIC_OUTS8 INT_R_X11Y74/NE2BEG0 INT_L_X12Y75/EE2BEG0 INT_L_X14Y75/EE4BEG0 INT_L_X24Y75/ER1BEG1 INT_R_X25Y75/CLK1 CLK_HROW_BOT_R_X67Y78/CLK_HROW_CK_INT_1_1 CLK_HROW_BOT_R_X67Y78/CLK_HROW_CK_MUX_OUT_L5 CLK_HROW_BOT_R_X67Y78/CLK_HROW_CK_HCLK_OUT_L5 CLK_HROW_BOT_R_X67Y78/CLK_HROW_CK_BUFHCLK_L5 HCLK_R_X32Y78/HCLK_CK_INOUT_R5 HCLK_L_X31Y78/HCLK_LEAF_CLK_B_TOPL5 INT_L_X10Y75/GCLK_L_B11_WEST
  set route [find_routing_path -quiet -from INT_L_X10Y75/GCLK_L_B11_WEST -to INT_L_X10Y75/CLK_L0]
  INT_L_X10Y75/GCLK_L_B11_WEST -> INT_L_X10Y75/CLK_L0: no route found - assuming direct PIP
  set route [find_routing_path -quiet -from INT_L_X10Y75/CLK_L0 -to CLBLM_L_X10Y75/CLBLM_L_CLK]
  INT_L_X10Y75/CLK_L0 -> CLBLM_L_X10Y75/CLBLM_L_CLK: no route found - assuming direct PIP
  Failed to route net mynet_0, status UNROUTED, route: CLBLM_R_X11Y74/CLBLM_L_A CLBLM_R_X11Y74/CLBLM_LOGIC_OUTS8 INT_R_X11Y74/NE2BEG0 INT_L_X12Y75/EE2BEG0 INT_L_X14Y75/EE4BEG0 INT_L_X24Y75/ER1BEG1 INT_R_X25Y75/CLK1 CLK_HROW_BOT_R_X67Y78/CLK_HROW_CK_INT_1_1 CLK_HROW_BOT_R_X67Y78/CLK_HROW_CK_MUX_OUT_L5 CLK_HROW_BOT_R_X67Y78/CLK_HROW_CK_HCLK_OUT_L5 CLK_HROW_BOT_R_X67Y78/CLK_HROW_CK_BUFHCLK_L5 HCLK_R_X32Y78/HCLK_CK_INOUT_R5 HCLK_L_X31Y78/HCLK_LEAF_CLK_B_TOPL5 INT_L_X10Y75/GCLK_L_B11_WEST INT_L_X10Y75/CLK_L0 CLBLM_L_X10Y75/CLBLM_L_CLK
</pre>

tmichalak, 2019-04-18:
It seems that the problem with successfully routing is not related to a specific PIP in a given interconnect tile. The same routing path that fails in one fuzzer run is routed successfully when done in isolation.
Here is an example from the failing test:
<pre>
== 7: INT_R CLK0 GCLK_B9
ffbel CFF
tile INT_R_X3Y95
Routing net mynet_7:
  set route [find_routing_path -quiet -from CLBLL_L_X2Y77/CLBLL_L_A -to INT_L_X2Y95/GCLK_L_B9_EAST]
  CLBLL_L_X2Y77/CLBLL_L_A -> INT_L_X2Y95/GCLK_L_B9_EAST: no route found - assuming direct PIP
  set route [find_routing_path -quiet -from INT_L_X2Y95/GCLK_L_B9_EAST -to INT_R_X3Y95/CLK0]
  INT_L_X2Y95/GCLK_L_B9_EAST -> INT_R_X3Y95/CLK0: no route found - assuming direct PIP
  set route [find_routing_path -quiet -from INT_R_X3Y95/CLK0 -to CLBLM_R_X3Y95/CLBLM_L_CLK]
  INT_R_X3Y95/CLK0 -> CLBLM_R_X3Y95/CLBLM_L_CLK: no route found - assuming direct PIP
  Failed to route net mynet_7, status UNROUTED, route: CLBLL_L_X2Y77/CLBLL_L_A INT_L_X2Y95/GCLK_L_B9_EAST INT_R_X3Y95/CLK0 CLBLM_R_X3Y95/CLBLM_L_CLK
WARNING: failed to route net
</pre>

and here when done in isloation, i.e. only one PIP and hardcoded driving and receiving tile:
<pre>
== 0: INT_R CLK0 GCLK_B9
ffbel CFF
tile INT_R_X3Y95
Routing net mynet_0:
  set route [find_routing_path -quiet -from CLBLL_L_X2Y77/CLBLL_L_A -to INT_L_X2Y95/GCLK_L_B9_EAST]
  CLBLL_L_X2Y77/CLBLL_L_A -> INT_L_X2Y95/GCLK_L_B9_EAST: CLBLL_L_X2Y77/CLBLL_L_A CLBLL_L_X2Y77/CLBLL_LOGIC_OUTS8 INT_L_X2Y77/SS2BEG0 INT_L_X2Y75/WW2BEG0 INT_L_X0Y75/BYP_ALT4 INT_L_X0Y75/BYP_L4 LIOI3_X0Y75/IOI_IMUX_RC1 HCLK_IOI3_X1Y78/HCLK_IOI_RCLK_BEFORE_DIV0 HCLK_IOI3_X1Y78/HCLK_IOI_RCLK_OUT0 HCLK_IOI3_X1Y78/HCLK_IOI_RCLK2RCLK0 HCLK_IOI3_X1Y78/HCLK_IOI_CK_BUFRCLK0 HCLK_L_X11Y78/HCLK_LEAF_CLK_B_TOPL3 INT_L_X2Y95/GCLK_L_B9_EAST
  set route [find_routing_path -quiet -from INT_L_X2Y95/GCLK_L_B9_EAST -to INT_R_X3Y95/CLK0]
  INT_L_X2Y95/GCLK_L_B9_EAST -> INT_R_X3Y95/CLK0: no route found - assuming direct PIP
  set route [find_routing_path -quiet -from INT_R_X3Y95/CLK0 -to CLBLM_R_X3Y95/CLBLM_L_CLK]
  INT_R_X3Y95/CLK0 -> CLBLM_R_X3Y95/CLBLM_L_CLK: no route found - assuming direct PIP

ROUTING DONE!
</pre>

This could be caused by some remnants of previous routing attempts that are being rolled back in the script.
I will try to lower the amount of PIPs that are being routed (and unrouted on failure) during one vivado run.

kgugala, 2019-04-18:
could it be that in the first case the path cannot be found, because some other net is occupying a PIP which has to be used in the currently considered path?

as @acomodi suggested this could be even a case where different path is using the PIP we want to route through

tmichalak, 2019-04-18:
Yes, this is exactly what I suspect. I guess I didn't state it clear enough in my comment.
Anyhow, by lowering the amount of pips routed during one vivado run we lower the chance of one route reusing the PIP of another.

acomodi, 2019-04-18:
I am thinking that it could be possible to modify the tcl script to keep track of the PIPs that have been already used as endpoints so that the next nets cannot include them (e.g. `GCLK_L_B9_EAST` cannot be chosen as an endpoint by another net).
If a PIP has not been used by another net,  I am pretty sure that a feasible route can be surely found.

This could apply to all the PIP fuzzers that use the same methodology.

mithro, 2019-04-18:
Could we just fail and try again with a new random set?

acomodi, 2019-04-18:
@mithro I think it is also doable as you say. The only thing is that I suppose that by adding a list of "already used" PIPs could be better from a performance point of view instead of letting it fail and try again.

tmichalak, 2019-04-18:
@acomodi I am not sure if we could use this information to change the behavior of the script. As you can see from the routing paths I pasted few comments before the difference is in the first part of the route. For finding the routing path, Vivado's ``find_routing_path`` command is used with only ``-from`` and ``-to`` arguments. I am not if and how the command would take the list of already visited PIPs into account.

In the meantime my test with the amount of PIPs routed during one Vivado run finished successfully.
This is what I changed:
<pre>
diff --git a/fuzzers/052-pip-clkin/Makefile b/fuzzers/052-pip-clkin/Makefile
index 00b113c..eaa4625 100644
--- a/fuzzers/052-pip-clkin/Makefile
+++ b/fuzzers/052-pip-clkin/Makefile
@@ -1,4 +1,4 @@
 MAKETODO_FLAGS=--re "^INT_[LR].CLK"
-TODO_N = 10
+TODO_N = 5
 include ../int_loop.mk
</pre>

And this is the script I used for testing:
<pre>

rm iteration_*.log
for i in {1..10}; do
echo "Iteration $i"
rm ${XRAY_DATABASE_DIR}/${XRAY_DATABASE}/segbits_int_*.db
script --return --command "make clean"
set -e
touch iteration_$i.log
script --return --command "make run -j$(nproc) > iteration_$i.log"
set +e
done
</pre>

tmichalak, 2019-04-25:
I have analyzed a number of situations in which the routing failed.
The problems were related to the fact that some parts of the routing paths done in the same vivado run for different PIPs were leftover and kept other specific PIPs from being routed properly despite 3 tries.
In PR #799 several precaution methods have been added to avoid specific routing attempts in the same vivado run:
1. Don't route a PIP with a destination or source wire already routed in this run
2. Don't route to nodes with the X coordinate already routed in this run

PR #799 (2019-04-26): 052-pip-clkin: Exclude potentially conflicting PIP + nodes from a run.
Add several precautions to avoid specific routing attempts in the same Vivado run:
 1. Don't route a PIP with a destination or source wire already routed in this run
 2. Don't route to nodes with the X coordinate already routed in this run

Fixes #783.

Some parts of the routing paths done in the same vivado run for different PIPs were leftover and kept other specific PIPs from being routed properly.
closed 2019-04-26 by mithro