f4pga/prjxray #784: Timeout on 041-clk-hrow-pips fuzzer

tmichalak, 2019-04-16:
The 041-clk-hrow-pips tends to timeout most of the times.
The objective is to get to the bottom of the timeout, but it would be beneficial in general for the run time of the entire database generation to analyze the root cause and shorten the run time of this fuzzer.

tmichalak, 2019-05-28:
The fuzzer times out because after few iterations it ends up with a todo list that it's unable to solve no matter how many further iterations will be performed:
One example of such list:
<pre>
CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO21.CLK_HROW_CK_IN_R8
CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO21.CLK_HROW_CK_IN_R9
CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO27.CLK_HROW_CK_IN_L9
CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_CK_IN_L8
CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO2.CLK_HROW_CK_IN_L9
CLK_HROW_BOT_R.CLK_HROW_BOT_R_CK_BUFG_CASCO3.CLK_HROW_CK_IN_R9
</pre>

litghost, 2019-05-28:
Ya, the R -> CASCO were very annoying to get right.  I'd try tuning the TODO routing portion and see if it is missing diversity.  I believe you mentioned that these bits were getting dropped due to bit conflicts?

tmichalak, 2019-05-28:
The bits we had bit conflicts with were: 
<pre>
CLK_HROW_TOP_R.CLK_HROW_TOP_R.CLK_HROW_CK_INT_0_1_ACTIVE
</pre>
Not sure if that is related.

Could you be more specific about what kind of diversity is a must have?

litghost, 2019-05-28:
Didn't we say that ``CLK_HROW_TOP_R.CLK_HROW_TOP_R.CLK_HROW_CK_INT_0_1_ACTIVE`` were fake bits?  So how is there a conflict?

tmichalak, 2019-05-28:
Yes, once we assumed they are fake we excluded them. They were conflicting before we had done this.
I was simply referring to what you wrote earlier:
>  I believe you mentioned that these bits were getting dropped due to bit conflicts?

and don't recall mentioning any other bit conflicts

closed 2019-06-26 by tmichalak