f4pga/prjxray #790: 045-hclk-cmt-pips unstable for zynq7

tmichalak, 2019-04-17:
In one of the CI runs I spotted the following fail in zynq:
<pre>
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s:     main()^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/utils/mergedb.py", line 59, in main^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s:     verbose=args.verbose)^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s:   File "/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/utils/mergedb.py", line 34, in run^M 
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s:     assert not strict, "strict: got duplicate bits"^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: AssertionError: strict: got duplicate bits^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: + finish^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: + echo 'Cleaning up temp files'^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: + rm -f ./tmp.FfV2omx8wK^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: + rm -f ./tmp.XUUNmaBdq8^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: make[2]: *** [database] Error 1^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: make[1]: *** [run] Error 2^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: ^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: --------------------------------------------------------------------------^M
2019-04-17T17:08:16.678024 - 045-hclk-cmt-pips        - 39m36s: Failed @ 2019-04-17T17:08:16.495855 with exit code: 2^M
2019-04-17T17:08:16.839452 - 045-hclk-cmt-pips        - 39m36s: WARNING: Fuzzer failed!^M
</pre>
We need to keep an eye on this fuzzer.

tmichalak, 2019-04-29:
The first thing I noticed when analyzing why the fuzzer fails for zynq7 is the fact that there are no HCLK_CMT pips, i.e. the piplist for HCLK_CMT is empty.

mithro, 2019-04-29:
@tmichalak Didn't you disable that fuzzer?

tmichalak, 2019-04-29:
@mithro: yes it has been disabled due to this issue. I am now working on fixing this so we can reenable the fuzzer.

tmichalak, 2019-04-30:
The fuzzer used to hang due to the missing HCLK_CMT tiles (following Makefile targes tried to use empty .rdb and .db files), hence I disabled the hclk_cmt specific targets for zynq7 and leaving only the hclk_cmt_l.

However, now I am getting the following error form segmaker:
<pre>
Loading bits from design.bits.
Loading tags from design.txt.
Compiling segment data.
Traceback (most recent call last):
  File "/home/tmichalak/Development/X-Ray/prjxray/fuzzers/045-hclk-cmt-pips/generate.py", line 105, in <module>
    main()
  File "/home/tmichalak/Development/X-Ray/prjxray/fuzzers/045-hclk-cmt-pips/generate.py", line 100, in main
    segmk.compile(bitfilter=bitfilter)
  File "/home/tmichalak/Development/X-Ray/prjxray/prjxray/segmaker.py", line 353, in compile
    tilename, this_tile_tags)
AssertionError: Tile HCLK_CMT_L_X119Y78 does not have bitstream info but 237 tags
/home/tmichalak/Development/X-Ray/prjxray/utils/top_generate.mk:25: recipe for target 'generate.ok' failed
make[2]: *** [generate.ok] Error 1
make[2]: Leaving directory '/home/tmichalak/Development/X-Ray/prjxray/fuzzers/045-hclk-cmt-pips/build/1/specimen_001'
../pip_loop.mk:32: recipe for target 'build/1/specimen_001/OK' failed
make[1]: *** [build/1/specimen_001/OK] Error 2
make[1]: Leaving directory '/home/tmichalak/Development/X-Ray/prjxray/fuzzers/045-hclk-cmt-pips'
../pip_loop.mk:63: recipe for target 'run' failed
</pre>

However, this error doesn't seem to be directly connected to the small change I made.

@litghost: Can be the fact that Zynq7 doesn't have the HCLK_CMT tiles, only the HCLK_CMT_L have destructive impact on the fuzzer?

litghost, 2019-04-30:
> @litghost: Can be the fact that Zynq7 doesn't have the HCLK_CMT tiles, only the HCLK_CMT_L have destructive impact on the fuzzer?

That looks like a problem with the `tilegrid.json` output from 005-tilegrid.  Did you look at `HCLK_CMT_L_X119Y78` in `tilegrid.json`?

tmichalak, 2019-04-30:
@litghost 

> That looks like a problem with the tilegrid.json output from 005-tilegrid. Did you look at HCLK_CMT_L_X119Y78 in tilegrid.json?

I didn't find anything special about this tile in tilegrid.json:
<pre>
    "HCLK_CMT_L_X119Y78": {
        "bits": {},
        "grid_x": 119,
        "grid_y": 26,
        "sites": {
            "BUFMRCE_X0Y2": "BUFMRCE",
            "BUFMRCE_X0Y3": "BUFMRCE"
        },
        "type": "HCLK_CMT_L"
    },
</pre>
 
What kind of problem do you have in mind?

litghost, 2019-04-30:
> @litghost
> 
> > That looks like a problem with the tilegrid.json output from 005-tilegrid. Did you look at HCLK_CMT_L_X119Y78 in tilegrid.json?
> 
> I didn't find anything special about this tile in tilegrid.json:
> 
>     "HCLK_CMT_L_X119Y78": {
>         "bits": {},
>         "grid_x": 119,
>         "grid_y": 26,
>         "sites": {
>             "BUFMRCE_X0Y2": "BUFMRCE",
>             "BUFMRCE_X0Y3": "BUFMRCE"
>         },
>         "type": "HCLK_CMT_L"
>     },
> What kind of problem do you have in mind?

There is a problem, there are no bits definition, just like the error described.

tmichalak, 2019-04-30:
You are right. Don't really know how I missed that. I will rerun the tilegrid fuzzer and see if the bits definitions appear. If not then there is a problem right there.

tmichalak, 2019-05-09:
The fuzzer fails due to the following problem:
<pre>
WARNING: got duplicate bits frozenset({'26_203', '26_209'})
  Orig line: HCLK_CMT_L.HCLK_CMT_MUX_CLK_PLL6_ACTIVE 26_203 26_209
  New line : HCLK_CMT_L.HCLK_CMT_CK_IN9.HCLK_CMT_MUX_CLK_PLL6 26_203 26_209
Traceback (most recent call last):
  File "/home/build/tmichalak/prjxray/utils/mergedb.py", line 63, in <module>
    main() 
  File "/home/build/tmichalak/prjxray/utils/mergedb.py", line 59, in main
    verbose=args.verbose)
  File "/home/build/tmichalak/prjxray/utils/mergedb.py", line 34, in run
    assert not strict, "strict: got duplicate bits"
AssertionError: strict: got duplicate bits
</pre>

The conflict is detected between iterations which perform pushdb during which the database is validated.
``HCLK_CMT_L.HCLK_CMT_CK_IN9.HCLK_CMT_MUX_CLK_PLL6 26_203 26_209`` is the correct result.
`HCLK_CMT_L.HCLK_CMT_MUX_CLK_PLL6_ACTIVE `` gets sometimes solved with one bit, sometimes with two (causing the conflict) or not at all.
 
@litghost 
The ``*_ACTIVE`` features are not PIPs and in the .rdb I see a lot of the  *_ACTIVE features with 0 candidates. In the end in the final db we only have a couple of the _ACTIVE features. Do we actually need these features?

litghost, 2019-05-09:
> The fuzzer fails due to the following problem:
> 
> WARNING: got duplicate bits frozenset({'26_203', '26_209'})
>   Orig line: HCLK_CMT_L.HCLK_CMT_MUX_CLK_PLL6_ACTIVE 26_203 26_209
>   New line : HCLK_CMT_L.HCLK_CMT_CK_IN9.HCLK_CMT_MUX_CLK_PLL6 26_203 26_209
> Traceback (most recent call last):
>   File "/home/build/tmichalak/prjxray/utils/mergedb.py", line 63, in 
>     main() 
>   File "/home/build/tmichalak/prjxray/utils/mergedb.py", line 59, in main
>     verbose=args.verbose)
>   File "/home/build/tmichalak/prjxray/utils/mergedb.py", line 34, in run
>     assert not strict, "strict: got duplicate bits"
> AssertionError: strict: got duplicate bits
> The conflict is detected between iterations which perform pushdb during which the database is validated.
> `HCLK_CMT_L.HCLK_CMT_CK_IN9.HCLK_CMT_MUX_CLK_PLL6 26_203 26_209` is the correct result.
> `HCLK_CMT_L.HCLK_CMT_MUX_CLK_PLL6_ACTIVE `` gets sometimes solved with one bit, sometimes with two (causing the conflict) or not at all.
> 
> @litghost
> The `*_ACTIVE` features are not PIPs and in the .rdb I see a lot of the *_ACTIVE features with 0 candidates. In the end in the final db we only have a couple of the _ACTIVE features. Do we actually need these features?

The _ACTIVE features are required, otherwise unknown bits will be present.  It might be possible to re-write the fuzzer to needing the _ACTIVE features, but that is unclear to me.

tmichalak, 2019-05-09:
>The _ACTIVE features are required, otherwise unknown bits will be present. 
In the ``segbits_hclk_cmt_l.db`` file for Artix (https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/segbits_hclk_cmt_l.db) we only have bits for 20 ports. The rest was either 0 candidates or 0 const (HCLK_CMT_L.HCLK_CMT_MUX_CLK_PLL6_ACTIVE among them).
How can be sure the database is complete when it comes to the *_ACTIVE features?

tmichalak, 2019-05-09:
One solution to avoid the fuzzer to fail for zynq is to increase the number of stimuli.
I just don't feel that we are calculating bits for all *_ACTIVE features.

litghost, 2019-05-09:
I believe only the HCLK and RCLK inputs have active lines, so with some trickery we could compute a correct todo.

tmichalak, 2019-05-15:
Increasing the number of specimen solved the stability problem for the cost of longer runtime.
Once it becomes a problem we will have to implement the extra todo mechanism shortly described in PR (#822)

closed 2019-05-15 by tmichalak