f4pga/prjxray #793: /bin/sh: 1: source: not found error on db extras generation

tmichalak, 2019-04-18:
When the CI is run on a Ubuntu machine where ``/bin/dash`` is used as the default shell, we get the following error on the harness generation step (the target is called ``db-extras-``:
<pre>
source minitests/roi_harness/basys3-swbut.sh && make -C fuzzers part_only
/bin/sh: 1: source: not found 
Makefile:108: recipe for target 'db-extras-artix7' failed
make: *** [db-extras-artix7] Error 127
</pre>

One way to overcome this is to change the shell used by the Makefile using the ``SHELL`` variable:
<pre>
SHELL = bash
</pre>

mithro, 2019-04-18:
You might want to check the other Makefiles....

mithro, 2019-06-12:
@tmichalak What is the status here? Has all the Makefiles been fixed?

tmichalak, 2019-06-13:
@mithro I have been running all of my stabilization tests on the same machine with Ubuntu that exposed the original problem and I haven't spotted any other similar issues. I guess this one can be closed. If anything similar comes up we will just open a new issue or reopen this one. Anyway this doesn't seem to have high priority.

mithro, 2019-06-13:
Great, closing.

closed 2019-06-13 by mithro