f4pga/prjxray #814: Figure out what to do with "empty" SDF files for tiles.

mithro, 2019-04-29:
The hash of fdbd15cd9a5b54d402980d250333c7ab8bdbff2f89eaffbc8053e1afde4dbfd5 is because of a large number of "empty" files like this;
```
            
            (DELAYFILE            
                (SDFVERSION "3.0")            
                (TIMESCALE 1ps)            
            
            )
```

These files are empty;
 * [`./artix7/timings/BRAM_INT_INTERFACE_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRAM_INT_INTERFACE_L.sdf)
 * [`./artix7/timings/BRAM_INT_INTERFACE_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRAM_INT_INTERFACE_R.sdf)
 * [`./artix7/timings/BRAM_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRAM_L.sdf)
 * [`./artix7/timings/BRAM_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRAM_R.sdf)
 * [`./artix7/timings/BRKH_BRAM.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_BRAM.sdf)
 * [`./artix7/timings/BRKH_B_TERM_INT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_B_TERM_INT.sdf)
 * [`./artix7/timings/BRKH_CLB.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_CLB.sdf)
 * [`./artix7/timings/BRKH_CLK.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_CLK.sdf)
 * [`./artix7/timings/BRKH_CMT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_CMT.sdf)
 * [`./artix7/timings/BRKH_DSP_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_DSP_L.sdf)
 * [`./artix7/timings/BRKH_DSP_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_DSP_R.sdf)
 * [`./artix7/timings/BRKH_GTX.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_GTX.sdf)
 * [`./artix7/timings/BRKH_INT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_INT.sdf)
 * [`./artix7/timings/BRKH_TERM_INT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/BRKH_TERM_INT.sdf)
 * [`./artix7/timings/B_TERM_INT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/B_TERM_INT.sdf)
 * [`./artix7/timings/CFG_CENTER_BOT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CFG_CENTER_BOT.sdf)
 * [`./artix7/timings/CLK_BUFG_BOT_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_BUFG_BOT_R.sdf)
 * [`./artix7/timings/CLK_BUFG_REBUF.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_BUFG_REBUF.sdf)
 * [`./artix7/timings/CLK_BUFG_TOP_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_BUFG_TOP_R.sdf)
 * [`./artix7/timings/CLK_FEED.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_FEED.sdf)
 * [`./artix7/timings/CLK_MTBF2.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_MTBF2.sdf)
 * [`./artix7/timings/CLK_PMV.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_PMV.sdf)
 * [`./artix7/timings/CLK_PMV2.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_PMV2.sdf)
 * [`./artix7/timings/CLK_PMV2_SVT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_PMV2_SVT.sdf)
 * [`./artix7/timings/CLK_PMVIOB.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_PMVIOB.sdf)
 * [`./artix7/timings/CLK_TERM.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CLK_TERM.sdf)
 * [`./artix7/timings/CMT_PMV.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CMT_PMV.sdf)
 * [`./artix7/timings/CMT_PMV_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/CMT_PMV_L.sdf)
 * [`./artix7/timings/GTP_INT_INTERFACE.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/GTP_INT_INTERFACE.sdf)
 * [`./artix7/timings/HCLK_BRAM.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_BRAM.sdf)
 * [`./artix7/timings/HCLK_CLB.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_CLB.sdf)
 * [`./artix7/timings/HCLK_DSP_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_DSP_L.sdf)
 * [`./artix7/timings/HCLK_DSP_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_DSP_R.sdf)
 * [`./artix7/timings/HCLK_FEEDTHRU_1.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_FEEDTHRU_1.sdf)
 * [`./artix7/timings/HCLK_FEEDTHRU_2.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_FEEDTHRU_2.sdf)
 * [`./artix7/timings/HCLK_FIFO_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_FIFO_L.sdf)
 * [`./artix7/timings/HCLK_GTX.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_GTX.sdf)
 * [`./artix7/timings/HCLK_INT_INTERFACE.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_INT_INTERFACE.sdf)
 * [`./artix7/timings/HCLK_IOB.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_IOB.sdf)
 * [`./artix7/timings/HCLK_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_L.sdf)
 * [`./artix7/timings/HCLK_L_BOT_UTURN.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_L_BOT_UTURN.sdf)
 * [`./artix7/timings/HCLK_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_R.sdf)
 * [`./artix7/timings/HCLK_R_BOT_UTURN.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_R_BOT_UTURN.sdf)
 * [`./artix7/timings/HCLK_TERM.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_TERM.sdf)
 * [`./artix7/timings/HCLK_TERM_GTX.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_TERM_GTX.sdf)
 * [`./artix7/timings/HCLK_VBRK.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_VBRK.sdf)
 * [`./artix7/timings/HCLK_VFRAME.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/HCLK_VFRAME.sdf)
 * [`./artix7/timings/INT_FEEDTHRU_1.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/INT_FEEDTHRU_1.sdf)
 * [`./artix7/timings/INT_FEEDTHRU_2.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/INT_FEEDTHRU_2.sdf)
 * [`./artix7/timings/INT_INTERFACE_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/INT_INTERFACE_L.sdf)
 * [`./artix7/timings/INT_INTERFACE_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/INT_INTERFACE_R.sdf)
 * [`./artix7/timings/INT_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/INT_L.sdf)
 * [`./artix7/timings/INT_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/INT_R.sdf)
 * [`./artix7/timings/IO_INT_INTERFACE_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/IO_INT_INTERFACE_L.sdf)
 * [`./artix7/timings/IO_INT_INTERFACE_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/IO_INT_INTERFACE_R.sdf)
 * [`./artix7/timings/L_TERM_INT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/L_TERM_INT.sdf)
 * [`./artix7/timings/MONITOR_MID.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/MONITOR_MID.sdf)
 * [`./artix7/timings/MONITOR_TOP.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/MONITOR_TOP.sdf)
 * [`./artix7/timings/PCIE_INT_INTERFACE_L.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/PCIE_INT_INTERFACE_L.sdf)
 * [`./artix7/timings/PCIE_INT_INTERFACE_R.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/PCIE_INT_INTERFACE_R.sdf)
 * [`./artix7/timings/PCIE_NULL.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/PCIE_NULL.sdf)
 * [`./artix7/timings/PCIE_TOP.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/PCIE_TOP.sdf)
 * [`./artix7/timings/R_TERM_INT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/R_TERM_INT.sdf)
 * [`./artix7/timings/R_TERM_INT_GTX.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/R_TERM_INT_GTX.sdf)
 * [`./artix7/timings/TERM_CMT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/TERM_CMT.sdf)
 * [`./artix7/timings/T_TERM_INT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/T_TERM_INT.sdf)
 * [`./artix7/timings/VBRK.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/VBRK.sdf)
 * [`./artix7/timings/VBRK_EXT.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/VBRK_EXT.sdf)
 * [`./artix7/timings/VFRAME.sdf`](https://github.com/SymbiFlow/prjxray-db/tree/master/artix7/timings/VFRAME.sdf)

mithro, 2019-06-12:
@kgugala - Thoughts?

kgugala, 2019-06-12:
Some of those empty files are caused by bugs in the 007 fuzzer (e.g. for BRAM solved in #857 )

I have to iterate through those empty files and figure out which are really empty and just do not emit them.

PR #882 (2019-06-16): 007-timing: Do not emit SDF files for sites with no timings.
fixes #814
closed 2019-06-16 by mithro