f4pga/prjxray #816: Implement a mechanism to track which fuzzer generated a specific solution

tmichalak, 2019-04-30:
The work related to stabilizing the results produced by all fuzzers shows that in many cases we have problems with duplicated tags detected during mergedb.
This is caused by the fact that for several features (mostly PIPs) we have different fuzzers calculating the solution for the same tag.
Maybe it would make sense to implement a mechanism in the mergedb.py script for example in which in parallel to the segbits_*.db file we generate a tag file which contains the information about which fuzzer created the tag. This file could be a temporary file that is used for reporting the conflicts by mergedb.py, and apart from the name of the tag we would have an extra information about which fuzzer produced the solution for the tag.

mithro, 2019-06-12:
@tmichalak This is done now that #842 has been merged, right?

closed 2019-06-12 by mithro

tmichalak, 2019-06-12:
@mithro: Yes, this has been merged.