f4pga/prjxray #817: Add feature for each input of a mux

HackerFoo, 2019-05-03:
Example:

```
CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI 00_00
CLBLM_L.SLICEM_X0.ALUT.DI1MUX.BDI1 !00_00
CLBLM_L.SLICEM_X0.BLUT.DI1MUX.BI 00_20
CLBLM_L.SLICEM_X0.BLUT.DI1MUX.DI !00_20
```

This makes it easier to read, and will allow detecting conflicts.

mithro, 2019-05-04:
A grep inside https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/segbits_clblm_l.db gives;
```
CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI 00_00
CLBLM_L.SLICEM_X0.BLUT.DI1MUX.BI 00_20
CLBLM_L.SLICEM_X0.CLUT.DI1MUX.CI 01_43
```

Should this really be;
```
CLBLM_L.SLICEM_X0.DI1MUX.AI 00_00
CLBLM_L.SLICEM_X0.DI1MUX.BI 00_20
CLBLM_L.SLICEM_X0.DI1MUX.CI 01_43
```

mithro, 2019-05-04:
This seems to be done by the https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/019-clb-ndi1mux fuzzer...

mithro, 2019-05-04:
https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/019-clb-ndi1mux/minitest has some more comments?

mithro, 2019-05-04:
BTW If it is the missing zero bits -- then bits.dbf file is what should be used I think?

See https://github.com/SymbiFlow/prjxray/blob/36e6290ec6638f88962d23582a3c5329e6d665d3/fuzzers/036-iob-ologic/bits.dbf#L1-L3 and https://github.com/SymbiFlow/prjxray/blob/36e6290ec6638f88962d23582a3c5329e6d665d3/fuzzers/036-iob-ologic/Makefile#L9-L11

litghost, 2019-05-08:
> A grep inside https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/segbits_clblm_l.db gives;
>
> ```
> CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI 00_00
> CLBLM_L.SLICEM_X0.BLUT.DI1MUX.BI 00_20 [...]

This is wrong, the current definition is correct, we are missing the inverted versions.

mithro, 2019-06-19:
So for the `ADI1MUX` both `BDI1` and the `BMC31` values are missing?

[image]

```
CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI     00_00
CLBLM_L.SLICEM_X0.ALUT.DI1MUX.BDI1  !00_00
CLBLM_L.SLICEM_X0.ALUT.DI1MUX.BMC31  ?????
```

https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/019-clb-ndi1mux/minitest

litghost, 2019-06-19:
> So for the `ADI1MUX` both `BDI1` and the `BMC31` values are missing?
> 
> [image]
> 
> ```
> CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI     00_00
> CLBLM_L.SLICEM_X0.ALUT.DI1MUX.BDI1  !00_00
> CLBLM_L.SLICEM_X0.ALUT.DI1MUX.BMC31  ?????
> ```
> 
> https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/019-clb-ndi1mux/minitest
> 
> > [image][image] GitHub**[SymbiFlow/prjxray](https://github.com/SymbiFlow/prjxray)**Documenting the Xilinx 7-series bit-stream format. - SymbiFlow/prjxray

No, as a said above, the database is correct.  There is no bit at all for the BMC31 or BDI1 port of the mux.  The likely table is:
 ``SRL`` -> ``ADI1MUX.BMC31``
``RAM`` -> ``ADI1MUX.BDI1``
``ADI1MUX.AI`` -> ``ADI1MUX.AI``

mithro, 2019-06-19:
So, this would be the real info?
```
CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI              00_00
# Slice configured as SRL == BCM31, RAM == BDI1
CLBLM_L.SLICEM_X0.ALUT.DI1MUX.BDI1_OR_BMC31  !00_00
```

litghost, 2019-06-19:
> So, this would be the real info?
> 
> ```
> CLBLM_L.SLICEM_X0.ALUT.DI1MUX.AI              00_00
> # Slice configured as SRL == BCM31, RAM == BDI1
> CLBLM_L.SLICEM_X0.ALUT.DI1MUX.BDI1_OR_BMC31  !00_00
> ```

Yes

mithro, 2019-07-08:
@litghost Is this done now with #900 or do we need a list of other muxes which need to be done?

litghost, 2019-07-08:
I only added bits for the ADI mux, I believe there are others.  For example, I know the CARRY [ABCD]CY mux needs this treatment.

mithro, 2019-07-10:
I opened #939 to get someone to go through and figure which other muxes need this treatment (and then hopefully fix them).

Closing this issue.

closed 2019-07-10 by mithro