f4pga/prjxray #867: Create a test design using LiteX, LiteEth and LiteDRAM and figure out what bits are still needed

mithro, 2019-06-04:
Create LiteX based SoC which includes LiteEth and LiteDRAM.

Enable Yosys for the frontend + Vivado for the backend.

Look at which bits are still needed in prjxray to make this work.

mithro, 2019-06-04:
An older version of this was kind of done here -> https://github.com/mithro/litex-buildenv-arty-yosys-xilinx

mithro, 2019-06-04:
https://github.com/enjoy-digital/linux-on-litex-vexriscv

mithro, 2019-06-04:
It is expected we will need;
 - [ ] Bits around CMT
 - [ ] Bits around SERDES
 - [ ] Bits around DELAY
 - [ ] Bits around clock buffer

mithro, 2019-06-04:
There are a bunch of issues with Yosys + Vivado issues with LiteX. See the closed issue @ https://github.com/YosysHQ/yosys/issues/711 (It wasn't necessarily fixed however...)

daveshah1, 2019-06-04:
FYI as of https://github.com/enjoy-digital/litex/pull/154 a basic SoC was working at least

mithro, 2019-06-04:
@daveshah1 - Florent never did reply and say it actually made things work? 

@enjoy-digital did that change actually fix things?

mithro, 2019-06-12:
@mkurc-ant Can you take charge on following up on this?

enjoy-digital, 2019-06-12:
@mithro: i remember testing but not sure it worked otherwise i would have answered. I can do another test tomorrow.

mkurc-ant, 2019-06-12:
@mithro Ok, I can take on this issue.

mkurc-ant, 2019-06-13:
https://github.com/SymbiFlow/prjxray/pull/883

litghost, 2019-07-19:
Number of unknown bits including #946: 2290, down from ~2.5k

litghost, 2019-07-20:
Number of unknown bits including  https://github.com/SymbiFlow/prjxray/pull/934: 1915

mithro, 2019-07-21:
Great work!

litghost, 2019-07-23:
Number of unknown bits after #970, #968, and #934: 1387

litghost, 2019-07-24:
Number of unknown bits after #970, #971, #968, and #934: 631

litghost, 2019-07-24:
Remaining bits to be documented (based on issues 955-966):

```
26_25
26_57
26_109
28_110
29_17
30_30
30_38
30_44
30_95
30_99
32_54
32_72
32_73
32_94
32_112
33_15
33_33
33_54
33_55
33_73
39_05                                                                                                                      
39_07
39_59
39_61
39_121
39_123
```

tmichalak, 2019-07-24:
Including the missing bits from issue #924 the list has 3 more entries for frame 27, so the list is:
<pre>
26_25
26_57
26_109
27_18
27_70
27_102
28_110
29_17
30_30
30_38
30_44
30_95
30_99
32_54
32_72
32_73
32_94
32_112
33_15
33_33
33_54
33_55
33_73
39_05                                                                                                                      
39_07
39_59
39_61
39_121
39_123
</pre>

tmichalak, 2019-07-26:
37 bits remaining:
<pre>
26_25
26_57
26_109

27_18
27_70
27_102

28_110

29_17

30_30
30_38
30_44
30_95
30_99

31_28
31_32
31_83
31_89
31_95
31_97

32_72
32_73
32_94

33_33
33_54
33_55

38_04
38_06
38_14
38_120
38_122

39_05                                                                                                                      
39_07
39_59
39_61
39_113
39_121
39_123
</pre>

litghost, 2019-07-26:
Number of unknown bits after #983, #981 and #934: 314

mithro, 2019-07-26:
@litghost How does your info verse @tmichalak's info compare? Is it 314 unknown bits in the image of which map to 37 unique bits in frames?

litghost, 2019-07-26:
> @litghost How does your info verse @tmichalak's info compare? Is it 314 unknown bits in the image of which map to 37 unique bits in frames?

I'm reporting the simple absolute count, e.g.:
```
grep unknown top.fasm | wc
```

I'm working up a little tool to generate a report similiar to @tmichalak's.

litghost, 2019-07-26:
Here is the output of the report:

```
Group HCLK_IOI (count = 25):
  26_23
  27_15
  27_16
  28_20
  28_23
  28_26
  28_29
  29_15
  29_17
  29_19
  29_21
  29_31
  30_15
  30_16
  30_19
  31_21
  32_14
  37_28
  37_29
  38_15
  38_26
  39_14
  39_15
  39_16
  39_22
Group IOI3 (count = 19):
  26_109
  26_25
  26_57
  27_102
  27_18
  27_70
  28_110
  29_17
  30_30
  30_38
  30_44
  30_95
  30_99
  31_28
  31_32
  31_83
  31_89
  31_95
  31_97
```

litghost, 2019-07-26:
Put up a PR for the report tool #984

litghost, 2019-07-26:
Updated report once I remembered to run 047:
```
Total unknown bits: 299
Group HCLK_IOI (count = 10):
  29_175 (count = 1)
  29_177 (count = 1)
  29_179 (count = 1)
  29_191 (count = 1)
  38_175 (count = 1)
  38_186 (count = 1)
  39_174 (count = 1)
  39_175 (count = 1)
  39_176 (count = 1)
  39_182 (count = 1)
Group IOI3 (count = 19):
  26_109 (count = 8)
  26_25 (count = 8)
  26_57 (count = 8)
  27_102 (count = 8)
  27_18 (count = 8)
  27_70 (count = 8)
  28_110 (count = 8)
  29_17 (count = 8)
  30_30 (count = 20)
  30_38 (count = 20)
  30_44 (count = 20)
  30_95 (count = 25)
  30_99 (count = 25)
  31_28 (count = 20)
  31_32 (count = 20)
  31_83 (count = 25)
  31_89 (count = 25)
  31_95 (count = 2)
  31_97 (count = 23)
```

mithro, 2019-07-26:
So, it seems like 29 unique bits to go?

litghost, 2019-07-27:
> So, it seems like 29 unique bits to go?

On that order, and I've just found another couple in the HCLK_IOI, and @mkurc-ant has an update to the ISERDES fuzzer that documents another 4 (https://github.com/SymbiFlow/prjxray/issues/955#issuecomment-515047949).   I'll post an updated breakdown once I've update the HCLK_IOI fuzzer.

litghost, 2019-07-27:
```
Total unknown bits: 295
Group HCLK_IOI (count = 6):
  38_175 (count = 1)
  38_186 (count = 1)
  39_174 (count = 1)
  39_175 (count = 1)
  39_176 (count = 1)
  39_182 (count = 1)
Group IOI3 (count = 19):
  26_109 (count = 8)
  26_25 (count = 8)
  26_57 (count = 8)
  27_102 (count = 8)
  27_18 (count = 8)
  27_70 (count = 8)
  28_110 (count = 8)
  29_17 (count = 8)
  30_30 (count = 20)
  30_38 (count = 20)
  30_44 (count = 20)
  30_95 (count = 25)
  30_99 (count = 25)
  31_28 (count = 20)
  31_32 (count = 20)
  31_83 (count = 25)
  31_89 (count = 25)
  31_95 (count = 2)
  31_97 (count = 23)
```

acomodi, 2019-07-29:
By running fuzzer 036 (from master) the ISERDES.IN_USE gives us the following additional bits that could be removed from the list:
```
27_102
27_70
28_110
26_25
26_57
```

acomodi, 2019-07-29:
I have created this updatable list of remaining bits to easily keep track of the remaining bits:

HCLK_IOI
- [ ] 38_175
- [ ] 39_174
- [ ] 39_175
- [ ] 39_176

IOI3
- [ ] 26_109
- [ ] 27_18
- [ ] 29_17
- [ ] 30_30
- [ ] 30_38
- [ ] 30_44
- [ ] 30_95
- [ ] 30_99
- [ ] 31_28
- [ ] 31_32
- [ ] 31_83
- [ ] 31_89
- [ ] 31_95
- [ ] 31_97

acomodi, 2019-07-30:
`29_17` is solved by 035 and is related to ISERDES.IN_USE

kgugala, 2019-07-30:
`26_109` and `27_18` are solved in #991 and are related to ISERDESE.DATA_WIDTH

litghost, 2019-07-30:
Remaining bits after current fuzzer updates are merged:
```
Total unknown bits: 229
Group HCLK_IOI (count = 4):
  38_175 (count = 1)
  39_174 (count = 1)
  39_175 (count = 1)
  39_176 (count = 1)
Group IOI3 (count = 11):
  30_30 (count = 20)
  30_38 (count = 20)
  30_44 (count = 20)
  30_95 (count = 25)
  30_99 (count = 25)
  31_28 (count = 20)
  31_32 (count = 20)
  31_83 (count = 25)
  31_89 (count = 25)
  31_95 (count = 2)
  31_97 (count = 23)
```

litghost, 2019-07-30:
Remaining bits after current fuzzer updates are merged:
```
Total unknown bits: 225
Group HCLK_IOI (count = 0):
Group IOI3 (count = 11):
  30_30 (count = 20)
  30_38 (count = 20)
  30_44 (count = 20)
  30_95 (count = 25)
  30_99 (count = 25)
  31_28 (count = 20)
  31_32 (count = 20)
  31_83 (count = 25)
  31_89 (count = 25)
  31_95 (count = 2)
  31_97 (count = 23)
```

mithro, 2019-07-30:
Looking good!

kgugala, 2019-07-31:
the missing bits seem to be clk PIP related. When I modified the 036 fuzzer to feed OSERDESE with CLK generated in PLL and routed through bufg (as Litex does) those bits appear in the fuzzed bitstreams (see #1001)

litghost, 2019-07-31:
With #1004:
```
Total unknown bits: 90
Group HCLK_IOI (count = 0):
Group IOI3 (count = 4):
  30_95 (count = 25)
  30_99 (count = 25)
  31_28 (count = 20)
  31_32 (count = 20)
```

litghost, 2019-07-31:
With #1005: No unknown bits!

mithro, 2019-07-31:
@litghost YAY!

closed 2019-08-01 by litghost

mithro, 2019-08-02:
Building at ae80db55ad33c56f61d6cabcd6e3cfc6d6625b79
```
Checking artix7 database
============================
Checking individual .db...
segbits_*.db: 33 okay
mask_*.db: 38 okay
Checking aggregate dir...
Checked 10239 tiles, 14584841 bits

Checking kintex7 database
============================
Checking individual .db...
segbits_*.db: 33 okay
mask_*.db: 38 okay
Checking aggregate dir...
Checked 13303 tiles, 20564482 bits

Checking zynq7 database
============================
Checking individual .db...
segbits_*.db: 32 okay
mask_*.db: 37 okay
Checking aggregate dir...
Checked 5675 tiles, 8840026 bits
```

mithro, 2019-08-02:
Latest push @ https://github.com/SymbiFlow/prjxray-db/commit/50b9a02db6b6151ac4bc85a1a4e4eb2a79adf365