f4pga/prjxray #874: Duplicate tag in 050-pip-seed

tmichalak, 2019-06-07:
**Symptom**

Duplicate tags.

 (https://source.cloud.google.com/results/invocations/b8eb941a-3779-459d-af46-71c376001497/targets)

XRAY_DATABASE_DIR=/tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/fuzzers/050-pip-seed/build/database bash /tmpfs/src/github/symbiflow-prjxray-presubmit-db-artix7/utils/mergedb.sh int_l build/segbits_int_l.db
WARNING: got duplicate tag INT_L.IMUX_L5.LOGIC_OUTS_L16
  Orig line: INT_L.IMUX_L5.LOGIC_OUTS_L16 !22_41 !23_41 !24_41 20_41 25_41 
  New line : INT_L.IMUX_L5.LOGIC_OUTS_L16 !22_41 !23_41 !24_41 01_03 20_41 25_41 

**Analysis**
PIPs withLOGIC_OUTS_ src wire are solved incorrectly

**Remedy**
These bits are solved in a separate fuzzer (051-pip-imuxlout-bypalts), so we just need to remove these bits from the list of solved bits in 050-pip-seed

mithro, 2019-06-12:
@tmichalak What is the status here?

tmichalak, 2019-06-13:
@mithro I have a fix on a branch https://github.com/antmicro/prjxray/tree/874_pip_seed.
I am testing it on #848 before a create a PR.

closed 2019-06-19 by tmichalak