f4pga/prjxray #875: Duplicate tag in 044-clk-bufg-pips

tmichalak, 2019-06-07:
**Symptom**
Duplicate tags.
(https://source.cloud.google.com/results/invocations/3a54b56c-a5c7-432d-954b-89b40c0cf2ed/targets )

<pre>
XRAY_DATABASE_DIR=/tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/fuzzers/044-clk-bufg-pips/build/database bash /tmpfs/src/github/symbiflow-prjxray-presubmit-db-zynq7/utils/mergedb.sh clk_bufg_bot_r build/segbits_clk_bufg_bot_r.db
WARNING: got duplicate tag CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_BOT_R_CK_MUXED20
  Orig line: CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_BOT_R_CK_MUXED20 !26_167 !26_23 !26_24 !27_166 26_168 27_22
  New line : CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_BOT_R_CK_MUXED20 !26_167 !27_166 26_168 
</pre>

**Analysis**
PIPs with CLK_BUFG_BOT_R_CK_MUXED src wire are solved incorrectly, if the segmatch constraint was -c 1 then it would have been solved correctly, however the rest of the PIPs calculated by this fuzzer require -c 2 

**Remedy**
Create a separate fuzzer for the clk-bufg-muxed PIPs

mithro, 2019-06-12:
@tmichalak What is the status here?

tmichalak, 2019-06-13:
@mithro I have a fix on a branch https://github.com/antmicro/prjxray/tree/875_44_clk_bufg_pips
I want to test it on #848 before creating a PR.

tmichalak, 2019-06-27:
PR has been merged. Closing.

closed 2019-06-27 by tmichalak