f4pga/prjxray #884: Instability in SDF output

mithro, 2019-06-14:
It looks like we are still not sorting everything in the SDF output which is causing diffs in the output, see the patch below;

```patch
diff --git a/artix7/timings/GTP_CHANNEL_2.sdf b/artix7/timings/GTP_CHANNEL_2.sdf
index d846ae0..7ebaf18 100644
--- a/artix7/timings/GTP_CHANNEL_2.sdf
+++ b/artix7/timings/GTP_CHANNEL_2.sdf
@@ -144,16 +144,16 @@
         (CELLTYPE "GTPE2_CHANNEL_GTPE2_CHANNELGTPE2_CHANNEL_RXCHBONDI_GEN_TRUE_16")
         (INSTANCE GTPE2_CHANNEL)
         (TIMINGCHECK
-            (HOLD RX_DATA_WIDTH (posedge RXUSRCLK2) (0.159::0.183))
-            (SETUP RX_DATA_WIDTH (posedge RXUSRCLK2) (0.328::0.378))
+            (HOLD RX_DATA_WIDTH (posedge RXUSRCLK) (0.159::0.183))
+            (SETUP RX_DATA_WIDTH (posedge RXUSRCLK) (0.328::0.378))
         )
     )
     (CELL
         (CELLTYPE "GTPE2_CHANNEL_GTPE2_CHANNELGTPE2_CHANNEL_RXCHBONDI_GEN_TRUE_20")
         (INSTANCE GTPE2_CHANNEL)
         (TIMINGCHECK
-            (HOLD RX_DATA_WIDTH (posedge RXUSRCLK2) (0.159::0.183))
-            (SETUP RX_DATA_WIDTH (posedge RXUSRCLK2) (0.328::0.378))
+            (HOLD RX_DATA_WIDTH (posedge RXUSRCLK) (0.159::0.183))
+            (SETUP RX_DATA_WIDTH (posedge RXUSRCLK) (0.328::0.378))
         )
     )
     (CELL
diff --git a/artix7/timings/GTP_CHANNEL_3.sdf b/artix7/timings/GTP_CHANNEL_3.sdf
index d846ae0..7ebaf18 100644
--- a/artix7/timings/GTP_CHANNEL_3.sdf
+++ b/artix7/timings/GTP_CHANNEL_3.sdf
@@ -144,16 +144,16 @@
         (CELLTYPE "GTPE2_CHANNEL_GTPE2_CHANNELGTPE2_CHANNEL_RXCHBONDI_GEN_TRUE_16")
         (INSTANCE GTPE2_CHANNEL)
         (TIMINGCHECK
-            (HOLD RX_DATA_WIDTH (posedge RXUSRCLK2) (0.159::0.183))
-            (SETUP RX_DATA_WIDTH (posedge RXUSRCLK2) (0.328::0.378))
+            (HOLD RX_DATA_WIDTH (posedge RXUSRCLK) (0.159::0.183))
+            (SETUP RX_DATA_WIDTH (posedge RXUSRCLK) (0.328::0.378))
         )
     )
     (CELL
         (CELLTYPE "GTPE2_CHANNEL_GTPE2_CHANNELGTPE2_CHANNEL_RXCHBONDI_GEN_TRUE_20")
         (INSTANCE GTPE2_CHANNEL)
         (TIMINGCHECK
-            (HOLD RX_DATA_WIDTH (posedge RXUSRCLK2) (0.159::0.183))
-            (SETUP RX_DATA_WIDTH (posedge RXUSRCLK2) (0.328::0.378))
+            (HOLD RX_DATA_WIDTH (posedge RXUSRCLK) (0.159::0.183))
+            (SETUP RX_DATA_WIDTH (posedge RXUSRCLK) (0.328::0.378))
         )
     )
     (CELL
```

Related #776 and #815.

mithro, 2019-06-14:
@kgugala - Can you track what is going wrong here?

kgugala, 2019-06-16:
@mithro I'm afraid this is not sorting issue, but rather a bug in the `tim2json` script. It seems it does not handle correctly very complicated timings definitions as we have in e.g. GTPE2_CHANNEL.

By complicated I mean a timing which includes number of properties. I think the above timing should in fact be:

```
(CELLTYPE "GTPE2_CHANNEL_GTPE2_CHANNELGTPE2_CHANNEL_RXCHBONDI_GEN_RXUSRCLK_TRUE_RX_DATA_WIDTH_16")
         (INSTANCE GTPE2_CHANNEL)
         (TIMINGCHECK
            (HOLD RXCHBONDI  (posedge RXUSRCLK2) (0.159::0.183))
            (SETUP RXCHBONDI (posedge RXUSRCLK2) (0.328::0.378))
         )
```
So from `RXCHBONDI` input to `RXUSRCLK2` clock. with `GEN_RXUSRCLK` property set to `true` and `RX_DATA_WIDTH` set to `16`
It looks like the script confused inputs with parameters

PR #901 (2019-06-25): BEL fuzzer stabilization
fixes #884
closed 2019-06-25 by litghost