f4pga/prjxray #893: IOB bits for OBUFDS, OBUFTDS and OBUFT

tmichalak, 2019-06-18:
Litex for Arty uses these primitives. We need to update the exising iob fuzzer with bits responsible for instantiating these primitives.

acomodi, 2019-07-01:
@tmichalak I have tried to add `OBUFT` to https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/030-iob/top.py (e.g. added it to the tile_types to instantiate the `OBUFT` primitive) and run the fuzzer.
I don't know if it is the correct way to go, but I have not discovered any new bits regarding IOBs.

I think that probably we have all the possible bits, it is just a matter of letting VPR understand what bits should be used when a different primitive is adopted (between OBUF, OBUFDS, OBUFTDS or OBUFT).

Some additional information about those primitives:
- OBUFDS: output buffer differential: as far as I understood from documentation and from inspecting a design, this is used to output two differntial signal by using two IOBs (IOB33M and IOB33S) 
- OBUFT: output buffer tristate: it uses another signal to enable/disable the buffer. 
- OBUFTDS: output buffer tristate differential: it is a combination of the previous two.

IMO the bits are already there, the thing to do is to update the architecture accordingly with the correct FASM features and add these three new primitives.

tmichalak, 2019-07-02:
@acomodi You may be right. You analysis alone is a vital piece of information. I agree that in case of the differential OBUFs we may already have all bits and it's a matter of making VPR understand it and emit them correctly. As for the OBUFT, could you create a minitest to double check that we have the bit responsible for enabling/disabling the buffer?

acomodi, 2019-07-04:
Got some interesting results by doing some minitests:

- OBUFT:
```
# In frame 0x00400026 1 bits were not converted.
{ unknown_bit = "00400026_50_15", unknown_segment = "0x00400000", unknown_segbit = "38_1615" }

# In frame 0x00400027 3 bits were not converted.
{ unknown_bit = "00400027_50_16", unknown_segment = "0x00400000", unknown_segbit = "39_1616" }
{ unknown_bit = "00400027_50_14", unknown_segment = "0x00400000", unknown_segbit = "39_1614" }
{ unknown_bit = "00400027_50_15", unknown_segment = "0x00400000", unknown_segbit = "39_1615" }
```

- OBUFDS:
```
# In frame 0x00400027 268 bits were not converted.
{ unknown_bit = "00400027_64_27", unknown_segment = "0x00400000", unknown_segbit = "39_2075" }
{ unknown_bit = "00400027_64_29", unknown_segment = "0x00400000", unknown_segbit = "39_2077" }
{ unknown_bit = "00400027_0_31", unknown_segment = "0x00400000", unknown_segbit = "39_31" }
{ unknown_bit = "00400027_64_31", unknown_segment = "0x00400000", unknown_segbit = "39_2079" }
{ unknown_bit = "00400027_65_1", unknown_segment = "0x00400000", unknown_segbit = "39_2081" }
{ unknown_bit = "00400027_65_31", unknown_segment = "0x00400000", unknown_segbit = "39_2111" }
{ unknown_bit = "00400027_2_1", unknown_segment = "0x00400000", unknown_segbit = "39_65" }
{ unknown_bit = "00400027_2_15", unknown_segment = "0x00400000", unknown_segbit = "39_79" }
{ unknown_bit = "00400027_66_17", unknown_segment = "0x00400000", unknown_segbit = "39_2129" }
{ unknown_bit = "00400027_66_21", unknown_segment = "0x00400000", unknown_segbit = "39_2133" }
{ unknown_bit = "00400027_66_23", unknown_segment = "0x00400000", unknown_segbit = "39_2135" }
{ unknown_bit = "00400027_66_31", unknown_segment = "0x00400000", unknown_segbit = "39_2143" }
{ unknown_bit = "00400027_67_1", unknown_segment = "0x00400000", unknown_segbit = "39_2145" }
{ unknown_bit = "00400027_67_15", unknown_segment = "0x00400000", unknown_segbit = "39_2159" }
{ unknown_bit = "00400027_3_27", unknown_segment = "0x00400000", unknown_segbit = "39_123" }
{ unknown_bit = "00400027_3_29", unknown_segment = "0x00400000", unknown_segbit = "39_125" }
{ unknown_bit = "00400027_3_31", unknown_segment = "0x00400000", unknown_segbit = "39_127" }
{ unknown_bit = "00400027_4_1", unknown_segment = "0x00400000", unknown_segbit = "39_129" }
{ unknown_bit = "00400027_68_27", unknown_segment = "0x00400000", unknown_segbit = "39_2203" }
{ unknown_bit = "00400027_68_29", unknown_segment = "0x00400000", unknown_segbit = "39_2205" }
{ unknown_bit = "00400027_4_31", unknown_segment = "0x00400000", unknown_segbit = "39_159" }
{ unknown_bit = "00400027_68_31", unknown_segment = "0x00400000", unknown_segbit = "39_2207" }
{ unknown_bit = "00400027_69_1", unknown_segment = "0x00400000", unknown_segbit = "39_2209" }
{ unknown_bit = "00400027_5_17", unknown_segment = "0x00400000", unknown_segbit = "39_177" }
{ unknown_bit = "00400027_5_21", unknown_segment = "0x00400000", unknown_segbit = "39_181" }
{ unknown_bit = "00400027_5_23", unknown_segment = "0x00400000", unknown_segbit = "39_183" }
{ unknown_bit = "00400027_5_31", unknown_segment = "0x00400000", unknown_segbit = "39_191" }
...

# In frame 0x0040001f 24 bits were not converted.
{ unknown_bit = "0040001f_36_22", unknown_segment = "0x00400000", unknown_segbit = "31_1174" }
{ unknown_bit = "0040001f_44_22", unknown_segment = "0x00400000", unknown_segbit = "31_1430" }
{ unknown_bit = "0040001f_48_22", unknown_segment = "0x00400000", unknown_segbit = "31_1558" }
{ unknown_bit = "0040001f_20_22", unknown_segment = "0x00400000", unknown_segbit = "31_662" }
{ unknown_bit = "0040001f_12_22", unknown_segment = "0x00400000", unknown_segbit = "31_406" }
{ unknown_bit = "0040001f_16_22", unknown_segment = "0x00400000", unknown_segbit = "31_534" }
{ unknown_bit = "0040001f_4_22", unknown_segment = "0x00400000", unknown_segbit = "31_150" }
{ unknown_bit = "0040001f_8_22", unknown_segment = "0x00400000", unknown_segbit = "31_278" }
{ unknown_bit = "0040001f_24_22", unknown_segment = "0x00400000", unknown_segbit = "31_790" }
{ unknown_bit = "0040001f_28_22", unknown_segment = "0x00400000", unknown_segbit = "31_918" }
{ unknown_bit = "0040001f_32_22", unknown_segment = "0x00400000", unknown_segbit = "31_1046" }
{ unknown_bit = "0040001f_40_22", unknown_segment = "0x00400000", unknown_segbit = "31_1302" }
{ unknown_bit = "0040001f_81_22", unknown_segment = "0x00400000", unknown_segbit = "31_2614" }
{ unknown_bit = "0040001f_53_22", unknown_segment = "0x00400000", unknown_segbit = "31_1718" }
{ unknown_bit = "0040001f_85_22", unknown_segment = "0x00400000", unknown_segbit = "31_2742" }
{ unknown_bit = "0040001f_57_22", unknown_segment = "0x00400000", unknown_segbit = "31_1846" }
{ unknown_bit = "0040001f_73_22", unknown_segment = "0x00400000", unknown_segbit = "31_2358" }
{ unknown_bit = "0040001f_61_22", unknown_segment = "0x00400000", unknown_segbit = "31_1974" }
{ unknown_bit = "0040001f_77_22", unknown_segment = "0x00400000", unknown_segbit = "31_2486" }
{ unknown_bit = "0040001f_65_22", unknown_segment = "0x00400000", unknown_segbit = "31_2102" }
{ unknown_bit = "0040001f_69_22", unknown_segment = "0x00400000", unknown_segbit = "31_2230" }
{ unknown_bit = "0040001f_89_22", unknown_segment = "0x00400000", unknown_segbit = "31_2870" }
{ unknown_bit = "0040001f_93_22", unknown_segment = "0x00400000", unknown_segbit = "31_2998" }
{ unknown_bit = "0040001f_97_22", unknown_segment = "0x00400000", unknown_segbit = "31_3126" }
[... 3 lines cut ...]
[... cut ...]
{ unknown_bit = "00400026_97_0", unknown_segment = "0x00400000", unknown_segbit = "38_3104" }
{ unknown_bit = "00400026_17_16", unknown_segment = "0x00400000", unknown_segbit = "38_560" }
{ unknown_bit = "00400026_33_16", unknown_segment = "0x00400000", unknown_segbit = "38_1072" }
...

```

Seems like there are quite a lot of bits related IMO to the `IOI` tiles which have not been documented yet.
The IOI tiles do include I/OLOGIC sites which, on their behalf, do contain I/OSERDES as well (not all of them, but they do alternate).

closed 2019-07-30 by litghost