f4pga/prjxray #898: Document HCLK_CMT config bits

mkurc-ant, 2019-06-19:
The HCLK_CMT tile contains:
- HCLK_CMT_MUX interconnect. Bits for this are required to break from the ROI (route clock from IOB to BUFHCE)
- 2xBUFMRCE. Those can route clock between clock regions to other CMT tiles.

Base address of this tile is already known. Having bits for the interconnect should be the priority.

tmichalak, 2019-06-26:
@litghost @mkurc-ant : Doesn't fuzzer 045-hclk-cmt-pips solve the interconnect bits already?

litghost, 2019-06-26:
I double checked this, and agree with @tmichalak that the HCLK CMT pips appear solved.  I do not believe at this time any additional bits are required for LiteX, but until LiteX reaches zero, it will be hard to tell.  In the mean time I've moved onto the PLL (which has a bunch of bits).  Hopefully once the PLL bits are documented, it will be clearer if this issue has remaining work.

closed 2019-07-30 by litghost