f4pga/prjxray #917: Should some pips bounce from 050-pip-seed and 056-pi-rem?

mithro, 2019-06-27:
The solver from some of the INT pips seems to be oscillating between `050-pip-seed` and `056-pi-rem`. The bit patterns seem to be the same but the origin is changing. Is this expected?

See example below;

```patch
--- a/artix7/segbits_int_r.origin_info.db
+++ b/artix7/segbits_int_r.origin_info.db
@@ -373,7 +373,7 @@ INT_R.EE4BEG1.SE6END1 origin:050-pip-seed 03_25 06_24
 INT_R.EE4BEG1.SS2END1 origin:050-pip-seed 03_24 05_27
 INT_R.EE4BEG1.SS6END1 origin:050-pip-seed 05_27 06_24
 INT_R.EE4BEG1.SW2END1 origin:050-pip-seed 02_25 05_27
-INT_R.EE4BEG1.SW6END1 origin:056-pip-rem 05_24 05_27
+INT_R.EE4BEG1.SW6END1 origin:050-pip-seed 05_24 05_27
 INT_R.EE4BEG2.EE2END2 origin:050-pip-seed 03_40 03_41
 INT_R.EE4BEG2.EE4END2 origin:050-pip-seed 03_41 05_40
 INT_R.EE4BEG2.LOGIC_OUTS10 origin:050-pip-seed 03_40 07_41
@@ -584,7 +584,7 @@ INT_R.FAN_ALT0.FAN_BOUNCE6 origin:050-pip-seed !23_00 20_00 22_00 24_00 25_00
 INT_R.FAN_ALT0.GFAN0 origin:054-pip-fan-alt !22_00 !23_00 !24_00 21_00 25_00
 INT_R.FAN_ALT0.LOGIC_OUTS0 origin:050-pip-seed !23_00 21_00 22_00 24_00 25_00
 INT_R.FAN_ALT0.LOGIC_OUTS12 origin:050-pip-seed !22_00 21_00 23_00 24_00 25_00
-INT_R.FAN_ALT0.LOGIC_OUTS22 origin:050-pip-seed !22_00 !23_00 !25_00 21_00 24_00
+INT_R.FAN_ALT0.LOGIC_OUTS22 origin:056-pip-rem !22_00 !23_00 !25_00 21_00 24_00
 INT_R.FAN_ALT0.NE2END0 origin:050-pip-seed !22_00 !23_00 !25_00 18_01 24_00
 INT_R.FAN_ALT0.NL1END0 origin:050-pip-seed !22_00 19_01 23_00 24_00 25_00
 INT_R.FAN_ALT0.NN2END0 origin:050-pip-seed !22_00 !23_00 !24_00 18_01 25_00
@@ -678,7 +678,7 @@ INT_R.FAN_ALT4.ER1END0 origin:050-pip-seed !23_08 17_08 22_08 24_08 25_08
 INT_R.FAN_ALT4.FAN_BOUNCE2 origin:050-pip-seed !23_08 20_08 22_08 24_08 25_08
 INT_R.FAN_ALT4.FAN_BOUNCE7 origin:056-pip-rem !22_08 20_08 23_08 24_08 25_08
 INT_R.FAN_ALT4.GFAN0 origin:054-pip-fan-alt !22_08 !23_08 !24_08 21_08 25_08
-INT_R.FAN_ALT4.LOGIC_OUTS18 origin:056-pip-rem !22_08 !23_08 !25_08 21_08 24_08
+INT_R.FAN_ALT4.LOGIC_OUTS18 origin:050-pip-seed !22_08 !23_08 !25_08 21_08 24_08
 INT_R.FAN_ALT4.LOGIC_OUTS4 origin:050-pip-seed !23_08 21_08 22_08 24_08 25_08
 INT_R.FAN_ALT4.LOGIC_OUTS8 origin:050-pip-seed !22_08 21_08 23_08 24_08 25_08
 INT_R.FAN_ALT4.NE2END0 origin:050-pip-seed !22_08 !23_08 !25_08 17_08 24_08
@@ -2191,9 +2191,9 @@ INT_R.NE6BEG3.NN6END3 origin:050-pip-seed 03_53 06_52
 INT_R.NE6BEG3.NW2END3 origin:050-pip-seed 02_53 04_53
 INT_R.NE6BEG3.NW6END3 origin:050-pip-seed 04_53 06_52
 INT_R.NE6BEG3.SE2END3 origin:050-pip-seed 02_53 05_55
-INT_R.NE6BEG3.SE6END3 origin:050-pip-seed 05_55 06_52
+INT_R.NE6BEG3.SE6END3 origin:056-pip-rem 05_55 06_52
 INT_R.NE6BEG3.WW2END2 origin:050-pip-seed 03_52 04_53
-INT_R.NE6BEG3.WW4END3 origin:050-pip-seed 04_53 05_52
+INT_R.NE6BEG3.WW4END3 origin:056-pip-rem 04_53 05_52
 INT_R.NL1BEG0.LOGIC_OUTS1 origin:050-pip-seed 07_16 14_17
 INT_R.NL1BEG0.LOGIC_OUTS13 origin:050-pip-seed 10_17 14_17
 INT_R.NL1BEG0.LOGIC_OUTS19 origin:050-pip-seed 08_17 14_17
@@ -3275,7 +3275,7 @@ INT_R.SW6BEG1.SW6END1 origin:050-pip-seed 03_29 05_28
 INT_R.SW6BEG1.WW2END1 origin:050-pip-seed 03_28 05_31
 INT_R.SW6BEG1.WW4END2 origin:050-pip-seed 05_28 05_31
 INT_R.SW6BEG2.EE2END2 origin:050-pip-seed 03_44 04_45
-INT_R.SW6BEG2.EE4END2 origin:050-pip-seed 04_45 05_44
+INT_R.SW6BEG2.EE4END2 origin:056-pip-rem 04_45 05_44
 INT_R.SW6BEG2.LOGIC_OUTS10 origin:050-pip-seed 03_44 04_46
 INT_R.SW6BEG2.LOGIC_OUTS14 origin:050-pip-seed 03_44 07_45
 INT_R.SW6BEG2.LOGIC_OUTS16 origin:050-pip-seed 04_46 06_44
@@ -3295,7 +3295,7 @@ INT_R.SW6BEG2.SW6END2 origin:050-pip-seed 03_45 05_44
 INT_R.SW6BEG2.WW2END2 origin:050-pip-seed 03_44 05_47
 INT_R.SW6BEG2.WW4END3 origin:050-pip-seed 05_44 05_47
 INT_R.SW6BEG3.EE2END3 origin:050-pip-seed 03_60 04_61
-INT_R.SW6BEG3.EE4END3 origin:050-pip-seed 04_61 05_60
+INT_R.SW6BEG3.EE4END3 origin:056-pip-rem 04_61 05_60
 INT_R.SW6BEG3.LH0 origin:056-pip-rem 04_62 05_60
 INT_R.SW6BEG3.LOGIC_OUTS11 origin:050-pip-seed 03_60 07_61
 INT_R.SW6BEG3.LOGIC_OUTS15 origin:050-pip-seed 03_60 04_62
@@ -3603,7 +3603,7 @@ INT_R.WW4BEG2.LOGIC_OUTS6 origin:050-pip-seed 02_33 07_33
 INT_R.WW4BEG2.LVB0 origin:056-pip-rem 04_34 05_32
 INT_R.WW4BEG2.LVB12 origin:056-pip-rem 05_32 07_33
 INT_R.WW4BEG2.NE2END2 origin:050-pip-seed 02_33 05_35
-INT_R.WW4BEG2.NE6END2 origin:056-pip-rem 05_32 05_35
+INT_R.WW4BEG2.NE6END2 origin:050-pip-seed 05_32 05_35
 INT_R.WW4BEG2.NN2END2 origin:050-pip-seed 03_32 05_35
 INT_R.WW4BEG2.NN6END2 origin:050-pip-seed 05_35 06_32
 INT_R.WW4BEG2.NW2END2 origin:050-pip-seed 02_33 03_33
```

tmichalak, 2019-06-28:
@mithro: yes, this is expected as 050-pip-seed is partially random and solves the PIPs from the todo list until some min progress mark, therefore there is always a non deterministic list of PIPs that don't get solved in 050. Hence theses unresolved fall into the 056-pip-rem.

mithro, 2019-06-28:
This makes me uncomfortable, but not enough that I think we should spend time on fixing it....