f4pga/prjxray #918: Should some pips bounce from 044-clk-bufg-pips and 046-clk-bufg-muxed-pips?

mithro, 2019-06-27:
The solver from some of the CLK_BUFG pips seems to be moving from `044-clk-bufg-pips` to `046-clk-bufg-muxed-pips`. The bit patterns seem to be the same but the origin is changing. 

Is this expected? If this is expected, I'll go ahead and update the database.

See example below;

```patch
diff --git a/artix7/segbits_clk_bufg_bot_r.origin_info.db b/artix7/segbits_clk_bufg_bot_r.origin_info.db
index 96b2ddd..c66984c 100644
--- a/artix7/segbits_clk_bufg_bot_r.origin_info.db
+++ b/artix7/segbits_clk_bufg_bot_r.origin_info.db
@@ -158,162 +158,162 @@ CLK_BUFG_BOT_R.BUFGCTRL.BUFGCTRL_X0Y9.ZINV_CE1 origin:042-clk-bufg-config 27_155
 CLK_BUFG_BOT_R.BUFGCTRL.BUFGCTRL_X0Y9.ZINV_S0 origin:042-clk-bufg-config 27_147
 CLK_BUFG_BOT_R.BUFGCTRL.BUFGCTRL_X0Y9.ZINV_S1 origin:042-clk-bufg-config 26_155
 CLK_BUFG_BOT_R.BUFGCTRL.BUFGCTRL_X0Y9.ZPRESELECT_I0 origin:042-clk-bufg-config 26_146
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I0.CLK_BUFG_BOT_R_CK_MUXED0 origin:044-clk-bufg-pips !26_07 !27_06 26_08
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I0.CLK_BUFG_BOT_R_CK_MUXED0 origin:046-clk-bufg-muxed-pips !26_07 !27_06 26_08
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I0.CLK_BUFG_IMUX24_0 origin:044-clk-bufg-pips !26_07 !26_08 27_06
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I0.CLK_BUFG_IMUX28_0 origin:044-clk-bufg-pips !26_07 !26_08 !27_06
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I0.CLK_BUFG_R_FBG_OUT1 origin:044-clk-bufg-pips !26_08 26_07 27_06
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I0.CLK_BUFG_R_FBG_OUT15 origin:044-clk-bufg-pips !26_08 !27_06 26_07
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I1.CLK_BUFG_BOT_R_CK_MUXED1 origin:044-clk-bufg-pips !26_05 !27_05 26_04
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I1.CLK_BUFG_BOT_R_CK_MUXED1 origin:046-clk-bufg-muxed-pips !26_05 !27_05 26_04
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I1.CLK_BUFG_IMUX24_0 origin:044-clk-bufg-pips !26_04 !27_05 26_05
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I1.CLK_BUFG_IMUX28_0 origin:044-clk-bufg-pips !26_04 !26_05 !27_05
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I1.CLK_BUFG_R_FBG_OUT1 origin:044-clk-bufg-pips !26_04 26_05 27_05
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL0_I1.CLK_BUFG_R_FBG_OUT15 origin:044-clk-bufg-pips !26_04 !26_05 27_05
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_BOT_R_CK_MUXED20 origin:044-clk-bufg-pips !26_167 !27_166 26_168
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_BOT_R_CK_MUXED20 origin:046-clk-bufg-muxed-pips !26_167 !27_166 26_168
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_IMUX26_2 origin:044-clk-bufg-pips !26_167 !26_168 27_166
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_IMUX30_2 origin:044-clk-bufg-pips !26_167 !26_168 !27_166
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_R_FBG_OUT11 origin:044-clk-bufg-pips !26_168 26_167 27_166
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I0.CLK_BUFG_R_FBG_OUT9 origin:044-clk-bufg-pips !26_168 !27_166 26_167
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I1.CLK_BUFG_BOT_R_CK_MUXED21 origin:044-clk-bufg-pips !26_165 !27_165 26_164
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I1.CLK_BUFG_BOT_R_CK_MUXED21 origin:046-clk-bufg-muxed-pips !26_165 !27_165 26_164
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I1.CLK_BUFG_IMUX26_2 origin:044-clk-bufg-pips !26_164 !27_165 26_165
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I1.CLK_BUFG_IMUX30_2 origin:044-clk-bufg-pips !26_164 !26_165 !27_165
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I1.CLK_BUFG_R_FBG_OUT11 origin:044-clk-bufg-pips !26_164 26_165 27_165
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL10_I1.CLK_BUFG_R_FBG_OUT9 origin:044-clk-bufg-pips !26_164 !26_165 27_165
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I0.CLK_BUFG_BOT_R_CK_MUXED22 origin:044-clk-bufg-pips !26_183 !27_182 26_184
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I0.CLK_BUFG_BOT_R_CK_MUXED22 origin:046-clk-bufg-muxed-pips !26_183 !27_182 26_184
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I0.CLK_BUFG_IMUX27_2 origin:044-clk-bufg-pips !26_183 !26_184 27_182
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I0.CLK_BUFG_IMUX31_2 origin:044-clk-bufg-pips !26_183 !26_184 !27_182
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I0.CLK_BUFG_R_FBG_OUT10 origin:044-clk-bufg-pips !26_184 !27_182 26_183
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I0.CLK_BUFG_R_FBG_OUT12 origin:044-clk-bufg-pips !26_184 26_183 27_182
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I1.CLK_BUFG_BOT_R_CK_MUXED23 origin:044-clk-bufg-pips !26_181 !27_181 26_180
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I1.CLK_BUFG_BOT_R_CK_MUXED23 origin:046-clk-bufg-muxed-pips !26_181 !27_181 26_180
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I1.CLK_BUFG_IMUX27_2 origin:044-clk-bufg-pips !26_180 !27_181 26_181
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I1.CLK_BUFG_IMUX31_2 origin:044-clk-bufg-pips !26_180 !26_181 !27_181
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I1.CLK_BUFG_R_FBG_OUT10 origin:044-clk-bufg-pips !26_180 !26_181 27_181
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL11_I1.CLK_BUFG_R_FBG_OUT12 origin:044-clk-bufg-pips !26_180 26_181 27_181
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I0.CLK_BUFG_BOT_R_CK_MUXED24 origin:044-clk-bufg-pips !26_199 !27_198 26_200
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I0.CLK_BUFG_BOT_R_CK_MUXED24 origin:046-clk-bufg-muxed-pips !26_199 !27_198 26_200
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I0.CLK_BUFG_IMUX24_3 origin:044-clk-bufg-pips !26_199 !26_200 27_198
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I0.CLK_BUFG_IMUX28_3 origin:044-clk-bufg-pips !26_199 !26_200 !27_198
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I0.CLK_BUFG_R_FBG_OUT11 origin:044-clk-bufg-pips !26_200 !27_198 26_199
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I0.CLK_BUFG_R_FBG_OUT13 origin:044-clk-bufg-pips !26_200 26_199 27_198
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I1.CLK_BUFG_BOT_R_CK_MUXED25 origin:044-clk-bufg-pips !26_197 !27_197 26_196
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I1.CLK_BUFG_BOT_R_CK_MUXED25 origin:046-clk-bufg-muxed-pips !26_197 !27_197 26_196
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I1.CLK_BUFG_IMUX24_3 origin:044-clk-bufg-pips !26_196 !27_197 26_197
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I1.CLK_BUFG_IMUX28_3 origin:044-clk-bufg-pips !26_196 !26_197 !27_197
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I1.CLK_BUFG_R_FBG_OUT11 origin:044-clk-bufg-pips !26_196 !26_197 27_197
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL12_I1.CLK_BUFG_R_FBG_OUT13 origin:044-clk-bufg-pips !26_196 26_197 27_197
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL13_I0.CLK_BUFG_BOT_R_CK_MUXED26 origin:044-clk-bufg-pips !26_215 !27_214 26_216
+CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL13_I0.CLK_BUFG_BOT_R_CK_MUXED26 origin:046-clk-bufg-muxed-pips !26_215 !27_214 26_216
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL13_I0.CLK_BUFG_IMUX25_3 origin:044-clk-bufg-pips !26_215 !26_216 27_214
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL13_I0.CLK_BUFG_IMUX29_3 origin:044-clk-bufg-pips !26_215 !26_216 !27_214
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL13_I0.CLK_BUFG_R_FBG_OUT12 origin:044-clk-bufg-pips !26_216 !27_214 26_215
 CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL13_I0.CLK_BUFG_R_FBG_OUT14 origin:044-clk-bufg-pips !26_216 26_215 27_214
-CLK_BUFG_BOT_R.CLK_BUFG_BUFGCTRL13_I1.CLK_BUFG_BOT_R_CK_MUXED27 origin:044-clk-bufg-pips !26_213 !27_213 26_212
[... 311 lines cut ...]
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I0.CLK_BUFG_R_FBG_OUT9 origin:044-clk-bufg-pips !26_136 26_135 27_134
-CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I0.CLK_BUFG_TOP_R_CK_MUXED16 origin:044-clk-bufg-pips !26_135 !27_134 26_136
+CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I0.CLK_BUFG_TOP_R_CK_MUXED16 origin:046-clk-bufg-muxed-pips !26_135 !27_134 26_136
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I1.CLK_BUFG_IMUX24_2 origin:044-clk-bufg-pips !26_132 !27_133 26_133
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I1.CLK_BUFG_IMUX28_2 origin:044-clk-bufg-pips !26_132 !26_133 !27_133
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I1.CLK_BUFG_R_FBG_OUT7 origin:044-clk-bufg-pips !26_132 !26_133 27_133
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I1.CLK_BUFG_R_FBG_OUT9 origin:044-clk-bufg-pips !26_132 26_133 27_133
-CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I1.CLK_BUFG_TOP_R_CK_MUXED17 origin:044-clk-bufg-pips !26_133 !27_133 26_132
+CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL8_I1.CLK_BUFG_TOP_R_CK_MUXED17 origin:046-clk-bufg-muxed-pips !26_133 !27_133 26_132
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL9_I0.CLK_BUFG_IMUX25_2 origin:044-clk-bufg-pips !26_151 !26_152 27_150
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL9_I0.CLK_BUFG_IMUX29_2 origin:044-clk-bufg-pips !26_151 !26_152 !27_150
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL9_I0.CLK_BUFG_R_FBG_OUT10 origin:044-clk-bufg-pips !26_152 26_151 27_150
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL9_I0.CLK_BUFG_R_FBG_OUT8 origin:044-clk-bufg-pips !26_152 !27_150 26_151
-CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL9_I0.CLK_BUFG_TOP_R_CK_MUXED18 origin:044-clk-bufg-pips !26_151 !27_150 26_152
[... cut ...]
 CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL9_I1.CLK_BUFG_R_FBG_OUT8 origin:044-clk-bufg-pips !26_148 !26_149 27_149
-CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL9_I1.CLK_BUFG_TOP_R_CK_MUXED19 origin:044-clk-bufg-pips !26_149 !27_149 26_148
+CLK_BUFG_TOP_R.CLK_BUFG_BUFGCTRL9_I1.CLK_BUFG_TOP_R_CK_MUXED19 origin:046-clk-bufg-muxed-pips !26_149 !27_149 26_148
 CLK_BUFG_TOP_R.CLK_BUFG_CK_GCLK16.CLK_BUFG_BUFGCTRL0_O origin:044-clk-bufg-pips 27_14
 CLK_BUFG_TOP_R.CLK_BUFG_CK_GCLK17.CLK_BUFG_BUFGCTRL1_O origin:044-clk-bufg-pips 27_30
 CLK_BUFG_TOP_R.CLK_BUFG_CK_GCLK18.CLK_BUFG_BUFGCTRL2_O origin:044-clk-bufg-pips 27_46
```

tmichalak, 2019-06-28:
@mithro: Yes. this is expected due to the fix in #875. The fix introduced a new fuzzer 046 dedicated for solving the CLK_BUFG_TOP_R_CK_MUXED[0-9] PIPs, so they are no longer solved by 044.

mithro, 2019-06-28:
@tmichalak Okay great!

closed 2019-06-28 by mithro