f4pga/prjxray #924: Verify remaining bits for ISERDES and OSERDES

litghost, 2019-07-02:
Litex uses the ISERDES and OSERDES sites, need to verify via minitest that we understand the relevant FASM features in the ISERDES and OSERDES.

litghost, 2019-07-03:
ISERDES is definitely not fully document:
```
{ unknown_bit = "0040001a_37_25", unknown_segment = "0x00400000", unknown_segbit = "26_1209" }
{ unknown_bit = "0040001a_37_11", unknown_segment = "0x00400000", unknown_segbit = "26_1195" }
{ unknown_bit = "0040001a_36_7", unknown_segment = "0x00400000", unknown_segbit = "26_1159" }
{ unknown_bit = "0040001b_37_6", unknown_segment = "0x00400000", unknown_segbit = "27_1190" }
{ unknown_bit = "0040001b_36_6", unknown_segment = "0x00400000", unknown_segbit = "27_1158" }
{ unknown_bit = "0040001b_37_14", unknown_segment = "0x00400000", unknown_segbit = "27_1198" }
{ unknown_bit = "0040001b_36_16", unknown_segment = "0x00400000", unknown_segbit = "27_1168" }
{ unknown_bit = "0040001b_37_18", unknown_segment = "0x00400000", unknown_segbit = "27_1202" }
{ unknown_bit = "0040001c_36_3", unknown_segment = "0x00400000", unknown_segbit = "28_1155" }
{ unknown_bit = "0040001c_36_13", unknown_segment = "0x00400000", unknown_segbit = "28_1165" }
{ unknown_bit = "0040001c_37_14", unknown_segment = "0x00400000", unknown_segbit = "28_1198" }
{ unknown_bit = "0040001d_37_2", unknown_segment = "0x00400000", unknown_segbit = "29_1186" }
{ unknown_bit = "0040001d_36_14", unknown_segment = "0x00400000", unknown_segbit = "29_1166" }
{ unknown_bit = "0040001e_36_14", unknown_segment = "0x00400000", unknown_segbit = "30_1166" }
{ unknown_bit = "0040001f_36_3", unknown_segment = "0x00400000", unknown_segbit = "31_1155" }
{ unknown_bit = "0040001f_36_13", unknown_segment = "0x00400000", unknown_segbit = "31_1165" }
```

tmichalak, 2019-07-24:
These Litex bits:
<pre>
# In frame 0x0040159b 24 bits were not converted.
{ unknown_bit = "0040159b_53_6", unknown_segment = "0x00401580", unknown_segbit = "27_1702" }
{ unknown_bit = "0040159b_57_6", unknown_segment = "0x00401580", unknown_segbit = "27_1830" }
{ unknown_bit = "0040159b_61_6", unknown_segment = "0x00401580", unknown_segbit = "27_1958" }
{ unknown_bit = "0040159b_69_6", unknown_segment = "0x00401580", unknown_segbit = "27_2214" }
{ unknown_bit = "0040159b_77_6", unknown_segment = "0x00401580", unknown_segbit = "27_2470" }
{ unknown_bit = "0040159b_81_6", unknown_segment = "0x00401580", unknown_segbit = "27_2598" }
{ unknown_bit = "0040159b_85_6", unknown_segment = "0x00401580", unknown_segbit = "27_2726" }
{ unknown_bit = "0040159b_93_6", unknown_segment = "0x00401580", unknown_segbit = "27_2982" }
{ unknown_bit = "0040159b_54_6", unknown_segment = "0x00401580", unknown_segbit = "27_1734" }
{ unknown_bit = "0040159b_58_6", unknown_segment = "0x00401580", unknown_segbit = "27_1862" }
{ unknown_bit = "0040159b_62_6", unknown_segment = "0x00401580", unknown_segbit = "27_1990" }
{ unknown_bit = "0040159b_70_6", unknown_segment = "0x00401580", unknown_segbit = "27_2246" }
{ unknown_bit = "0040159b_78_6", unknown_segment = "0x00401580", unknown_segbit = "27_2502" }
{ unknown_bit = "0040159b_82_6", unknown_segment = "0x00401580", unknown_segbit = "27_2630" }
{ unknown_bit = "0040159b_86_6", unknown_segment = "0x00401580", unknown_segbit = "27_2758" }
{ unknown_bit = "0040159b_94_6", unknown_segment = "0x00401580", unknown_segbit = "27_3014" }
{ unknown_bit = "0040159b_83_18", unknown_segment = "0x00401580", unknown_segbit = "27_2674" }
{ unknown_bit = "0040159b_91_18", unknown_segment = "0x00401580", unknown_segbit = "27_2930" }
{ unknown_bit = "0040159b_95_18", unknown_segment = "0x00401580", unknown_segbit = "27_3058" }
{ unknown_bit = "0040159b_55_18", unknown_segment = "0x00401580", unknown_segbit = "27_1778" }
{ unknown_bit = "0040159b_59_18", unknown_segment = "0x00401580", unknown_segbit = "27_1906" }
{ unknown_bit = "0040159b_67_18", unknown_segment = "0x00401580", unknown_segbit = "27_2162" }
{ unknown_bit = "0040159b_71_18", unknown_segment = "0x00401580", unknown_segbit = "27_2290" }
{ unknown_bit = "0040159b_79_18", unknown_segment = "0x00401580", unknown_segbit = "27_2546" }
</pre>

are solved by 
<pre>
RIOB33.IOB_Y0.ISERDES.27_18 27_18
RIOB33.IOB_Y0.ISERDES.27_70 27_70
RIOB33.IOB_Y0.ISERDES.27_102 27_102
</pre>

litghost, 2019-07-26:
This issue is covered by the new "Bits for base address 0x401580, frame XX" issues.

closed 2019-07-26 by litghost

tmichalak, 2019-07-26:
Well, not exactly, but now it is #982