f4pga/prjxray #949: 045 instability?

litghost, 2019-07-19:
https://source.cloud.google.com/results/invocations/5deb49e3-29f8-4db5-b1e7-c03e607985ff/targets

I don't think this is a new issue.

litghost, 2019-07-19:
```
2019-07-15T18:17:25.590417 - 045-hclk-cmt-pips        - 29m16s: + echo '`define SEED 32'\''h3930e631'
2019-07-15T18:17:25.590417 - 045-hclk-cmt-pips        - 29m16s: + make -f /tmpfs/src/github/symbiflow-prjxray-continuous-db-artix7/utils/top_generate.mk
2019-07-15T18:17:25.590417 - 045-hclk-cmt-pips        - 29m16s: /opt/Xilinx/Vivado/2017.2/bin/loader: line 179: 11253 Segmentation fault      "$RDI_PROG" "$@"
2019-07-15T18:17:25.590417 - 045-hclk-cmt-pips        - 29m16s: make[3]: *** [vivado.ok] Error 139
2019-07-15T18:17:25.590417 - 045-hclk-cmt-pips        - 29m16s: make[2]: *** [build/1/specimen_085/OK] Error 2
2019-07-15T18:17:25.590417 - 045-hclk-cmt-pips        - 29m16s: make[2]: *** Waiting for unfinished jobs....
2019-07-15T18:17:25.590417 - 045-hclk-cmt-pips        - 29m16s: make[1]: *** [run] Error 2
```

mkurc-ant, 2019-12-11:
I encountered that while the CI was running for #1169
```
...
pure virtual method called
terminate called without an active exception
make[3]: *** [vivado.ok] Error 134
make[2]: *** [build/7/specimen_015/OK] Error 2
make[2]: *** Waiting for unfinished jobs....
Parent process (pid 62143) has died. This helper process will now exit
make[1]: *** [run] Error 2
```

tmichalak, 2021-02-02:
@mkurc-ant @litghost I haven't seen this for a while, have you?

mkurc-ant, 2021-02-02:
@tmichalak Me neither.

tmichalak, 2021-03-03:
Didn't show up for quite some time. Closing for now.

closed 2021-03-03 by tmichalak